litex-hub/litex-boards #8: recommended max length for platform/target?

AnttiLukats, 2019-08-05:
https://github.com/timvideos/litex-buildenv/blob/master/gateware/info/platform.py

there info for platform and target is set to fixed lenght of 8 characters, would it make sense for new targets and plaftorms to keep max_len 8, or is it ok to use longer names?

enjoy-digital, 2019-08-05:
For now this is specific to platforms/targets defined in litex-buildenv, llitex-boards target are not yet using it. We still need to discuss what's the best, there is already an identifier module that can also be used for that. For now i suggest not providing it, we'll update later.

AnttiLukats, 2019-08-06:
this did not answer the question - is there a recommended NAME length limit? if later the ident core has limit on name length, then platoforms with too long name would have issue :(

enjoy-digital, 2019-08-06:
The maximum length of the identifier module is 255 characters: https://github.com/enjoy-digital/litex/blob/master/litex/soc/cores/identifier.py#L11

AnttiLukats, 2019-08-06:
yes, I know, but timvideos has introduced "platform" and "target" info modules, both 8 characters max,, thats why I asked, I'd also better stick with single identified space of 255 characters..

enjoy-digital, 2019-08-06:
I found 8 characters too restrictive and managing a single identifier is less restrictive/more flexible, so for boards i litex-boards i recommend just using the LiteX identifier.

closed 2019-08-06 by enjoy-digital