litex-hub/litex-boards #9: EthernetSoC for Intel FPGA?

AnttiLukats, 2019-08-07:
there seem to be no ethernet enabled target for any Intel platform? What is the best way to create one? doing a copy of EthernetSoC from arty to C10LP platform does compile but ethernet seems to be dead, at least no packets are transmitted.

enjoy-digital, 2019-08-08:
Ethernet has not been tested yet in an Intel platform. The design your created seems fine:
- https://github.com/micro-FPGA/litex-boards/blob/master/litex_boards/partner/platforms/c10lprefkit.py
- https://github.com/micro-FPGA/litex-boards/blob/master/litex_boards/partner/targets/c10lprefkit.py

Now since this is a configuration that hasn't been validated yet, there can be some small things to figure out.

While testing your design, it seems there is a first issue with AsyncResetSynchronizer on Intel platforms, since eth_tx/eth_rx clock domain are not toggling, for now you can workaround this by replacing https://github.com/enjoy-digital/liteeth/blob/master/liteeth/phy/mii.py#L86-L89 with:
````python
        self.comb += [
            self.cd_eth_tx.rst.eq(reset),
            self.cd_eth_rx.rst.eq(reset)
        ]
````
This allow having eth_tx/eth_rx toggling and fix the netboot freeze.

To see what is transmitted by the CPU and what is received, you can un-comment:
https://github.com/enjoy-digital/litex/blob/master/litex/soc/software/libnet/microudp.c#L21-L22

RX packets seem corrects (i pinged the board while it was trying to netboot), but TX packets are not seen by Wireshark.

AnttiLukats, 2019-08-08:
I try that, I also did see incoming data with SignalTap but TX_EN was stuck, I try the workaround now.. to see what happens

enjoy-digital, 2019-08-08:
The issue with the AsyncResetSynchronizer was probably the polarity, it should be fixed with https://github.com/enjoy-digital/litex/commit/0ba9ab92b4fc9a6e9830843d11ef94d0a5df5b89.

AnttiLukats, 2019-08-08:
ok, I see TX_EN being activated then also TX_DATA for the transmit packet, but still nothing in wireshort, weird..

AnttiLukats, 2019-08-10:
tested both eth1 and eth2 with LiteX soc, no tx data ever seen, for some reason also not with phy loopback. Tested the PHY working using hamsterz arty ethernet transmit code, this worked instantly and transmit packet (no PHY registers accessed..) so the PHY works and can transmit

enjoy-digital, 2019-08-10:
It's probably related to the edge where tx_data/tx_en is set. Can you do a test with by replacing:
https://github.com/enjoy-digital/liteeth/blob/master/liteeth/phy/mii.py#L76 with: 
`self.comb += self.cd_eth_tx.clk.eq(~clock_pads.tx)`

AnttiLukats, 2019-08-11:
I had tried already to insert "falling edge" flops to txdata and txen with no success inverting the tx_clk did also not make it work. whats even more strange changing main clock input from 12MHz to 25MHz input (in different bank) made the rx path to fail also? 25mhz clock is sync to the PHY reference clocks same source, while 12Mhz is async to the phy clocks. weird, doing more signaltap analyze

AnttiLukats, 2019-08-11:
so simple, all solved. the tx_clk inversion is not needed. the problem was in SDC file generated it does try to timing constraints to PORTS while using internal clock net names, this causes those constraints to be ignored by Quartus, fix
in top.sdc

create_clock -name eth_rx_clk -period 80.0 [get_ports {eth1_clocks_rx}]
create_clock -name eth_tx_clk -period 80.0 [get_ports {eth1_clocks_tx}]

after that it all works, now I only need to figure out how to-do in python ;)

closed 2019-08-11 by AnttiLukats

enjoy-digital, 2019-08-11:
OK thanks, i just opened https://github.com/enjoy-digital/litex/issues/239 to be sure fix that.

enjoy-digital, 2019-08-12:
The best for now is to use this in your target file:
````python
        platform.add_period_constraint(platform.lookup_request("eth_clocks").tx, 1e9/12.5e6)
        platform.add_period_constraint(platform.lookup_request("eth_clocks").rx, 1e9/12.5e6)
        platform.add_false_path_constraints(
            platform.lookup_request("clk12"),
            platform.lookup_request("eth_clocks").tx,
            platform.lookup_request("eth_clocks").rx
        )