litex-hub/litex-boards #15: `litex_boards.partner.targets` depends on liteeth

xobs, 2019-08-28:
When importing something from `litex_boards.partner.targets`, all boards are imported.  This adds a dependency on `liteeth` where one is not necessary.

```
xobs@nas /d/x/f/litex> ./workshop.py  --board pvt
lxbuildenv: v2019.8.19.1 (run ./workshop.py --lx-help for help)
lxbuildenv: Skipping git configuration because "skip-git" was found in LX_CONFIGURATION
lxbuildenv: To fetch from git, run ./workshop.py --board pvt --lx-check-git
Traceback (most recent call last):
  File "./workshop.py", line 17, in <module>
    from litex_boards.partner.targets.fomu import _CRG
  File "/disk/xobs-data/fomu-workshop/litex/deps/litex_boards/litex_boards/partner/targets/__init__.py", line 1, in <module>
    from litex_boards.partner.targets import netv2
  File "/disk/xobs-data/fomu-workshop/litex/deps/litex_boards/litex_boards/partner/targets/netv2.py", line 19, in <module>
    from liteeth.phy.rmii import LiteEthPHYRMII
ModuleNotFoundError: No module named 'liteeth'
xobs@nas /d/x/f/litex>
```

enjoy-digital, 2019-08-28:
Thanks for the feedback. The ideas was to keep the files organized as it is with official/partner/community directory but allow just importing any platform/target from `litex_boards` directly. (to ease use/maintenance and simplify changing one platform/target location).

I need to think about a better way to do this and avoid adding dependency. I'll think of it but ideas are welcome :)

xobs, 2019-08-28:
I would have thought that this would work:

```python
from litex_boards.partner.targets.fomu import _CRG
```

But it seems that implicitly does an import of `litex_boards.partner.targets.__init__`, which is where the problem is coming from.  Is there another way I should be importing things?

enjoy-digital, 2019-08-31:
Sorry, i haven't been able to look at that yet. I need to have a closer look at how import work in Python. I'd like to be able to do:
```python
from litex_boards.targets.fomu import _CRG
```
instead of:
```python
from litex_boards.partner.targets.fomu import _CRG
```
to be able to move more easily platforms/targets from an location to another in the future and simplify imports.

xobs, 2019-09-02:
Alright, let me know what you'd like me to do -- I have a BaseSoC implementation of a Fomu target that might be nice to merge, but it doesn't work because of the import issue.

enjoy-digital, 2019-09-02:
@xobs: i need to look at that, to avoid blocking you i just commited https://github.com/litex-hub/litex-boards/commit/e704014b3643129e39e2b88fadc02a2581c2696b, this should allow your to integrate your code and give me time to do what i initially wanted to do.

xobs, 2019-09-02:
@enjoy-digital thanks for that, I just submitted a PR to add Fomu back in, now with a `BaseSoC` in addition to the `_CRG`.

enjoy-digital, 2019-09-03:
The imports are fixed with https://github.com/litex-hub/litex-boards/commit/b4eefa6c33ca6a8047da7cafc777d4caa254a34e.

closed 2019-09-03 by enjoy-digital