litex-hub/litex-boards #23: de10nano support

rob-ng15, 2019-11-27:
I have started trying to modify the de0nano for the de10nano, using the 128mb SDRAM expansion board for MiSTer, the retro console/computer/arcade emulator.

The problem I ma having is that when I use, from the documentation, the correct UART serial_rx (PIN A22)  and serial_tx (PIN B21) in platforms/de10nano

    ("serial", 0,
        Subsignal("tx", Pins("B21"), IOStandard("3.3-V LVTTL")),
        Subsignal("rx", Pins("A22"), IOStandard("3.3-V LVTTL"))
    ),

I get the following errors:
Error (14566): The Fitter cannot place 2 periphery component(s) due to conflicts with existing constraints (2 pin(s)). Fix the errors described in the submessages, and then rerun the Fitter. The Intel FPGA Knowledge Database may also contain articles with information on how to resolve this periphery placement failure. Review the errors and then visit the Knowledge Database at https://www.altera.com/support/support-resources/knowledge-base/search.html and search for this specific error message number.
    Error (175020): The Fitter cannot place logic pin in region (79, 81) to (79, 81), to which it is constrained, because there are no valid locations in the region for logic of this type.
        Info (14596): Information about the failing component(s):
            Info (175028): The pin name(s): serial_rx
        Error (16234): No legal location could be found out of 1 considered location(s).  Reasons why each location could not be used are summarized below:
            Error (184016): There were not enough single-ended input pin locations available (1 location affected)
                Info (175029): A22
            Info (175015): The I/O pad serial_rx is constrained to the location PIN_A22 due to: User Location Constraints (PIN_A22) File: /home/rob/Development/Quartus/Projects/linux-on-litex-vexriscv/build/de10nano/gateware/top.v Line: 6
                Info (14709): The constrained I/O pad is contained within this pin
    Error (175020): The Fitter cannot place logic pin in region (79, 81) to (79, 81), to which it is constrained, because there are no valid locations in the region for logic of this type.
        Info (14596): Information about the failing component(s):
            Info (175028): The pin name(s): serial_tx
        Error (16234): No legal location could be found out of 1 considered location(s).  Reasons why each location could not be used are summarized below:
            Error (184016): There were not enough single-ended output pin locations available (1 location affected)
                Info (175029): B21
            Info (175015): The I/O pad serial_tx is constrained to the location PIN_B21 due to: User Location Constraints (PIN_B21) File: /home/rob/Development/Quartus/Projects/linux-on-litex-vexriscv/build/de10nano/gateware/top.v Line: 5
                Info (14709): The constrained I/O pad is contained within this pin
Error (12289): An error occurred while applying the periphery constraints. Review the offending constraints and rerun the Fitter.
Info (11798): Fitter preparation operations ending: elapsed time is 00:00:15
Error (11802): Can't fit design in device. Modify your design to reduce resources, or choose a larger device. The Intel FPGA Knowledge Database contains many articles with specific details on how to resolve this error. Visit the Knowledge Database at https://www.altera.com/support/support-resources/knowledge-base/search.html and search for this specific error message number.
Error: Quartus Prime Fitter was unsuccessful. 9 errors, 4 warnings
    Error: Peak virtual memory: 1342 megabytes
    Error: Processing ended: Wed Nov 27 20:03:50 2019
    Error: Elapsed time: 00:00:22
    Error: Total CPU time (on all processors): 00:00:22

I can change these pins to general GPIO pins and get a successful compilation. Any ideas?

enjoy-digital, 2019-12-06:
Hi @rob-ng15,

sorry for the delay. Can you share the platform file you created so that i do some tests and try to understand? It seems there is a conflict in the constraint file.

rob-ng15, 2019-12-06:
Many thanks.

On the DE10nano the UART is connected to the ARM processor (HPS in the FPGA
terms) and is not accessible to the FPGA.

I'll dig the file out of my archives, as in the setup that I have it had
128mb sdram, so could be useful.

Some peripherals will need to be via the HPS system.

rob-ng15, 2019-12-07:
The PLATFORM file (this won't compile due to the B21/A22 error)
`# This file is Copyright (c) 2014-2019 Florent Kermarrec <florent@enjoy-digital.fr>
# License: BSD

from litex.build.generic_platform import *
from litex.build.altera import AlteraPlatform
from litex.build.altera.programmer import USBBlaster

# IOs ----------------------------------------------------------------------------------------------

_io = [
    ("clk50", 0, Pins("V11"), IOStandard("3.3-V LVTTL")),

    ("user_led", 0, Pins("W15"), IOStandard("3.3-V LVTTL")),
    ("user_led", 1, Pins("AA24"), IOStandard("3.3-V LVTTL")),
    ("user_led", 2, Pins("V16"), IOStandard("3.3-V LVTTL")),
    ("user_led", 3, Pins("V15"), IOStandard("3.3-V LVTTL")),
    ("user_led", 4, Pins("AF26"), IOStandard("3.3-V LVTTL")),
    ("user_led", 5, Pins("AE26"), IOStandard("3.3-V LVTTL")),
    ("user_led", 6, Pins("Y`6"), IOStandard("3.3-V LVTTL")),
    ("user_led", 7, Pins("AA23"), IOStandard("3.3-V LVTTL")),

    ("key", 0, Pins("AH17"), IOStandard("3.3-V LVTTL")),
    ("key", 1, Pins("AH16"), IOStandard("3.3-V LVTTL")),

    ("sw", 0, Pins("Y24"), IOStandard("3.3-V LVTTL")),
    ("sw", 1, Pins("W24"), IOStandard("3.3-V LVTTL")),
    ("sw", 2, Pins("W21"), IOStandard("3.3-V LVTTL")),
    ("sw", 3, Pins("W20"), IOStandard("3.3-V LVTTL")),

    ("serial", 0,
        Subsignal("tx", Pins("B21"), IOStandard("3.3-V LVTTL")),                                            # UART pins in documentation B21 (HPS_GPIO49) general GPIO pin Y15 
        Subsignal("rx", Pins("A22"), IOStandard("3.3-V LVTTL"))                                            # UART pins in documentation A22 (HPS_GPIO50) general GPIO pin AC24
    ),

    ("sdram_clock", 0, Pins("AD20"), IOStandard("3.3-V LVTTL")),                                            # ALLOW_SYNCH_CTRL_USAGE OFF -to *|SDRAM_*
    ("sdram", 0,
        Subsignal("a", Pins("Y11 AA26 AA13 AA11 W11 Y19 AB23 AC23 AC22 C12 AB26 AD17 D12")),                # FAST_OUTPUT_REGISTER ON -to SDRAM_A*
        Subsignal("ba", Pins("Y17 AB25")),                                                                  # FAST_OUTPUT_REGISTER ON -to SDRAM_BA*
        Subsignal("cs_n", Pins("Y18")),                                                                     # FAST_OUTPUT_REGISTER ON -to SDRAM_n*
        Subsignal("cke", Pins("AG10")),                                                                     # FAST_OUTPUT_REGISTER ON -to SDRAM_n*
        Subsignal("ras_n", Pins("W14")),                                                                    # FAST_OUTPUT_REGISTER ON -to SDRAM_n*
        Subsignal("cas_n", Pins("AA18")),                                                                   # FAST_OUTPUT_REGISTER ON -to SDRAM_n*
        Subsignal("we_n", Pins("AA19")),                                                                    # FAST_OUTPUT_REGISTER ON -to SDRAM_n*
        Subsignal("dq", Pins("E8 V12 D11 W12 AH13 D8 AH14 AF7 AE24 AD23 AE6 AE23 AG14 AD5 AF4 AH3")),       # FAST_OUTPUT_REGISTER ON -to SDRAM_DQ[*] FAST_OUTPUT_ENABLE_REGISTER ON -to SDRAM_DQ[*]
        Subsignal("dm", Pins("AG13 AF13")),                                                                 # FAST_OUTPUT_REGISTER ON -to SDRAM_DQM*
        IOStandard("3.3-V LVTTL")
    ),

#    ("epcs", 0,
#        Subsignal("data0", Pins("H2")),
#        Subsignal("dclk", Pins("H1")),
#        Subsignal("ncs0", Pins("D2")),
#        Subsignal("asd0", Pins("C1")),
#        IOStandard("3.3-V LVTTL")
#    ),

#    ("i2c", 0,
#        Subsignal("sclk", Pins("F2")),
#        Subsignal("sdat", Pins("F1")),
#        IOStandard("3.3-V LVTTL")
#    ),

#    ("g_sensor", 0,
#        Subsignal("cs_n", Pins("G5")),
#        Subsignal("int", Pins("M2")),
#        IOStandard("3.3-V LVTTL")
#    ),

#    ("adc", 0,
#        Subsignal("cs_n", Pins("A10")),
#        Subsignal("saddr", Pins("B10")),
#        Subsignal("sclk", Pins("B14")),
#        Subsignal("sdat", Pins("A9")),
#        IOStandard("3.3-V LVTTL")
#    ),

    ("gpio_0", 0,
        Pins("Y15 AC24 AA15 AD26 AG28  AF28 AE25 AF27 AG26 AH27 AG25 AH26 AH24 AF25 AG23 AF23 AG24 AH22 AH21 AG21 AH23 AA20 AF22 AE22 AG20 AF21 AG19 AH19 AG18 AH18 AF18 AF20 AG15 AE20 AE19 AE17"),
        IOStandard("3.3-V LVTTL")
    ),
#    ("gpio_1", 0,
#        Pins("AG13 AF13 AG10 AG9 U14 U13 AG8 AH8 AF17 AE15 AF15 AG16 AH11 AH12 AH9 AG11"),
#        IOStandard("3.3-V LVTTL")
#    ),
]

# Platform -----------------------------------------------------------------------------------------

class Platform(AlteraPlatform):
    default_clk_name = "clk50"
    default_clk_period = 1e9/50e6

    def __init__(self):
        AlteraPlatform.__init__(self, "5CSEBA6U23I7", _io)

    def create_programmer(self):
        return USBBlaster()
`

rob-ng15, 2019-12-07:
The PLATFORMS file
[de10nano.pdf](https://github.com/litex-hub/litex-boards/files/3935026/de10nano.pdf)

The TARGETS file
[de10nano.pdf](https://github.com/litex-hub/litex-boards/files/3935027/de10nano.pdf)

These are just modified de0 nano files, with the SDRAM from the MiSTer project linked. These WON'T compile due to the problem with the UART pins, probably due to them being on the ARM processor side.

AnttiLukats, 2019-12-19:
you can use jtag_uart that what we do

enjoy-digital, 2019-12-22:
@rob-ng15: as @AnttiLukats suggests, you can use the JTAG UART by adding `--uart-name=jtag_atlantic`when building your target file.

rob-ng15, 2019-12-23:
Many thanks for the feedback. I believe I have it built for the de10 nano + MiSTer SDRAM and I/O board. Will test when the FTDI lead with pins arrives.

Can I ask where do I put the --uart-name=jtag_atlantic ?

enjoy-digital, 2019-12-23:
When building your target, instead if just executing it, do: `./target.py --uart-name=jtag_atlantic`

rob-ng15, 2019-12-23:
Many thanks for that. I have it running, kind of. I have a system that boots to the BIOS, but due to having to use nios2-terminal to communicate with the board, no way of using lxterm to load and boot the linux image. That may be doable once the breakout cables arrive, fingers cross.

I've attached my files, included are a couple of screenshots. The de10nano is configured for MiSTer, https://github.com/MiSTer-devel/Main_MiSTer/wiki , and has all of the expansion boards, 128MB SDRAM, USB-HUB and I/O board (the USB3 a like port will be used for serial once cables arrive).

To go further with the de10nano, I guess it will be necessary to use the HPS system, to allow USB, ethernet, which is a shame, as it is a great board.

Anyway, for now, here it all is!

[de10nanomister.zip](https://github.com/litex-hub/litex-boards/files/3995528/de10nanomister.zip)

rob-ng15, 2019-12-23:
[image]

[image]

[image]

enjoy-digital, 2019-12-23:
Nice, thanks for sharing your results! Happy to merge it to litex-boards if you create a pull request for it. (could be located in community). I'm closing this since the problem is now identified and things are working.

closed 2019-12-23 by enjoy-digital

rob-ng15, 2019-12-24:
Hello.

I've been unable to create a branch/pull request for this code. Can this be done?

rob-ng15, 2020-02-05:
I've uploaded my files, so far, to https://github.com/rob-ng15/litex-mister