litex-hub/litex-boards #25: Aller, Nereid, and Tagus targets need updating to new CSR APIs

dkozel, 2019-12-14:
Trying to build the default Aller target with the latest LiteX and LiteX-boards code results in the following error:
```
Traceback (most recent call last):
  File "./aller.py", line 201, in <module>
    main()
  File "./aller.py", line 197, in main
    soc.generate_software_header("../software/kernel/csr.h")
  File "./aller.py", line 182, in generate_software_header
    csr_header = get_csr_header(self.get_csr_regions(),
  File "/home/dkozel/src/litex/litex/soc/integration/soc_core.py", line 530, in __getattr__
    return Module.__getattr__(self, name)
  File "/home/dkozel/src/migen/migen/fhdl/module.py", line 136, in __getattr__
    raise AttributeError("'"+self.__class__.__name__+"' object has no attribute '"+name+"'")
AttributeError: 'AllerSoC' object has no attribute 'get_csr_regions'
```
Commit https://github.com/enjoy-digital/litex/commit/8be5824e258b84df240d34636aaa539124b92c65 simplified the handling of csr regions and constants. These target files need to be updated to the new API.

They also still use the `with_shadow_base` which was removed/deprecated in https://github.com/enjoy-digital/litex/commit/a4ef9b29b9781f576caba316c7920c902a8d8c8f. I think the new default behavior of `get_csr_header` is the same as the old behavior with `with_shadow_base=False` and `shadow_base` left to the default 0x80000000 value.

However the LitePCIeWishboneBridge from https://github.com/enjoy-digital/litepcie is also using a `shadow_base` parameter with a value of 0x8000000. The default `shadow_base` parameter in the LitePCIeWishboneBridge is 0x00000000 so it does need to still be supplied.
https://github.com/enjoy-digital/litepcie/blob/ee78f4f7c0338f6b81e70a2ab5ecc94040d808ba/litepcie/frontend/wishbone.py#L11-L13

So, I think I've fixed both issues correctly in https://github.com/litex-hub/litex-boards/pull/24 but don't have any hardware to test with. I'm looking at getting an Aller soon and found this while doing a test build.

mithro, 2019-12-14:
@enjoy-digital - FYI

enjoy-digital, 2019-12-14:
This is now fixed with https://github.com/litex-hub/litex-boards/pull/24 and https://github.com/litex-hub/litex-boards/commit/7184032555ddb3aafa34f7b76fe065b1a43fe10c. Thanks @dkozel.

closed 2019-12-14 by enjoy-digital