litex-hub/litex-boards #34: hadbadge: make.py build: ValueError (Invalid speed grade c)

pdp7, 2020-01-11:
@enjoy-digital I am creating a new issue rather than continuing to comment on the closed PR:
https://github.com/litex-hub/litex-boards/pull/31#issuecomment-573301387

I still get this error during build for the hadbadge:
```
pdp7@x1:~/dev/upstream/linux-on-litex-vexriscv$ python3 make.py --board=hadbadge --build
make: Entering directory '/home/pdp7/dev/upstream/linux-on-litex-vexriscv/build/hadbadge/software/libcompiler_rt'
make: Nothing to be done for 'all'.
make: Leaving directory '/home/pdp7/dev/upstream/linux-on-litex-vexriscv/build/hadbadge/software/libcompiler_rt'
make: Entering directory '/home/pdp7/dev/upstream/linux-on-litex-vexriscv/build/hadbadge/software/libbase'
 CC       exception.o
 CC       system.o
 CC       id.o
 CC       uart.o
 CC       time.o
 CC       spiflash.o
 CC       mdio.o
 AR       libbase.a
 AR       libbase-nofloat.a
make: Leaving directory '/home/pdp7/dev/upstream/linux-on-litex-vexriscv/build/hadbadge/software/libbase'
make: Entering directory '/home/pdp7/dev/upstream/linux-on-litex-vexriscv/build/hadbadge/software/libnet'
 CC       microudp.o
 AR       libnet.a
make: Leaving directory '/home/pdp7/dev/upstream/linux-on-litex-vexriscv/build/hadbadge/software/libnet'
make: Entering directory '/home/pdp7/dev/upstream/linux-on-litex-vexriscv/build/hadbadge/software/bios'
 CC       isr.o
 CC       sdram.o
/home/pdp7/dev/upstream/litex/litex/soc/software/bios/sdram.c: In function 'sdrrderr':
/home/pdp7/dev/upstream/litex/litex/soc/software/bios/sdram.c:225:49: warning: division by zero [-Wdiv-by-zero]
    printf("%2x", DFII_PIX_DATA_BYTES/2 - 1 - (i % (DFII_PIX_DATA_BYTES/2)));
                                                 ^
 CC       main.o
 CC       boot.o
 LD       bios.elf
chmod -x bios.elf
 OBJCOPY  bios.bin
chmod -x bios.bin
python3 -m litex.soc.software.mkmscimg bios.bin --little
make: Leaving directory '/home/pdp7/dev/upstream/linux-on-litex-vexriscv/build/hadbadge/software/bios'
Traceback (most recent call last):
  File "make.py", line 318, in <module>
    main()
  File "make.py", line 305, in main
    builder.build()
  File "/home/pdp7/dev/upstream/litex/litex/soc/integration/builder.py", line 182, in build
    toolchain_path=toolchain_path, **kwargs)
  File "/home/pdp7/dev/upstream/litex/litex/soc/integration/soc_core.py", line 461, in build
    return self.platform.build(self, *args, **kwargs)
  File "/home/pdp7/dev/upstream/litex/litex/build/lattice/platform.py", line 34, in build
    return self.toolchain.build(self, *args, **kwargs)
  File "/home/pdp7/dev/upstream/litex/litex/build/lattice/trellis.py", line 213, in build
    (family, size, speed_grade, package) = nextpnr_ecp5_parse_device(platform.device)
  File "/home/pdp7/dev/upstream/litex/litex/build/lattice/trellis.py", line 83, in nextpnr_ecp5_parse_device
    raise ValueError("Invalid speed grade {}".format(speed_grade))
ValueError: Invalid speed grade c
````

pdp7, 2020-01-11:
I have run:
```
pdp7@x1:~/dev/upstream$ sudo ./litex_setup.py update
[updating migen]...
Already up to date.
[updating litex]...
Already up to date.
[updating liteeth]...
Already up to date.
[updating litedram]...
Already up to date.
[updating litepcie]...
Already up to date.
[updating litesata]...
Already up to date.
[updating litesdcard]...
Already up to date.
[updating liteiclink]...
Already up to date.
[updating litevideo]...
Already up to date.
[updating litescope]...
Already up to date.
[updating litejesd204b]...
Already up to date.
[updating litex-boards]...
Already up to date.
````

https://gist.github.com/pdp7/a288af12bd97685da2e90d051d63b6b9

pdp7, 2020-01-11:
I think I am up-to-date:
~/dev/upstream/./litesdcard
```
commit 652f9c24c4aa244bf0efdba98b69a491280a8504 (HEAD -> master, origin/master, origin/HEAD)
Author: Florent Kermarrec <florent@enjoy-digital.fr>
Date:   Thu Jan 9 12:28:41 2020 +0100

    example: switch cpu to VexRiscv
```
~/dev/upstream/./liteeth
```
commit f2b3f7eeb1a68a4db0d9a3315994374c15e46019 (HEAD -> master, origin/master, origin/HEAD)
Author: Florent Kermarrec <florent@enjoy-digital.fr>
Date:   Mon Nov 25 11:43:16 2019 +0100

    test: update test_etherbone, use litex.gen.sim for all tests
```
~/dev/upstream/./litedram
```
commit 34e6c24d7296aeed6fb71ddc009b357cfc5a82ca (HEAD -> master, origin/master, origin/HEAD)
Author: Florent Kermarrec <florent@enjoy-digital.fr>
Date:   Fri Jan 10 14:27:05 2020 +0100

    frontend/wishbone: add write data buffer to avoid stalling wishbone while waiting for wdata.ready
```
~/dev/upstream/./litescope
```
commit daf10e9473fb70b3034e0331ef89005661ac04e0 (HEAD -> master, origin/master, origin/HEAD)
Author: Florent Kermarrec <florent@enjoy-digital.fr>
Date:   Mon Dec 2 13:03:02 2019 +0100

    examples/make: based on Migen & LiteX
```
~/dev/upstream/./migen
```
commit 677ca961b7c0a8273ba4f0f14070180bf54771fa (HEAD -> master, origin/master, origin/HEAD)
Author: Sebastien Bourdeauducq <sb@m-labs.hk>
Date:   Mon Jan 6 16:54:51 2020 +0800

    fix previous commit
```
~/dev/upstream/./litepcie
```
commit 5fa8d1a263ebd250fdd9b61d66836a80c91d957d (HEAD -> master, origin/master, origin/HEAD)
Author: Florent Kermarrec <florent@enjoy-digital.fr>
Date:   Thu Jan 9 19:12:53 2020 +0100

    gen: allow sys_clk_freq definition
```
~/dev/upstream/./linux-on-litex-vexriscv-prebuilt
```
commit 484d4931acdb76f811ccb33f7f7ddc7b6104f3f7 (HEAD -> master, origin/master, origin/HEAD)
Merge: 29d8761 5587286
Author: enjoy-digital <florent@enjoy-digital.fr>
Date:   Mon Jan 6 10:52:36 2020 +0100

    Merge pull request #3 from zakgi/master
    
    Added Support for Saanlima's Pipistrello LX45
```
~/dev/upstream/./litejesd204b
```
commit 7e2f1736332d02218e5d0a79057a6e0533eadece (HEAD -> master, origin/master, origin/HEAD)
Author: Florent Kermarrec <florent@enjoy-digital.fr>
Date:   Mon Nov 18 23:26:34 2019 +0100

    link/Aligner: simplify
```
~/dev/upstream/./litevideo
```
commit 146d4a73a242f6449682b58de2428cd351297c8d (HEAD -> master, origin/master, origin/HEAD)
Merge: 7328d27 7c24d18
Author: enjoy-digital <florent@enjoy-digital.fr>
Date:   Sun Dec 1 12:22:35 2019 +0100

    Merge pull request #26 from FrankBuss/master
    
    minor comments and example code cleanup
```
~/dev/upstream/./litesata
```
commit 1e3573b07d382eac50ef764fd839009bf90cb8ce (HEAD -> master, origin/master, origin/HEAD)
Author: Florent Kermarrec <florent@enjoy-digital.fr>
Date:   Tue Jan 7 16:33:16 2020 +0100

    phy/a7sataphy: use same OOB clocking than USB3 PIPE
```
~/dev/upstream/./litex-boards
```
commit beccf670e510dca481cabfe6d6a13737e89e4a82 (HEAD -> master, origin/master, origin/HEAD)
Author: Florent Kermarrec <florent@enjoy-digital.fr>
Date:   Sat Jan 11 10:46:23 2020 +0100

    hadbadge: fix _CRG
```
~/dev/upstream/./linux-on-litex-vexriscv
```
commit 8ecfdcd855d42f5b373e28ecc106f8d67fdfd4be (HEAD -> master, origin/master, origin/HEAD)
Author: Florent Kermarrec <florent@enjoy-digital.fr>
Date:   Fri Jan 10 09:42:54 2020 +0100

    make.py/de0nano: increase l2_size to 2048, update comment
```
~/dev/upstream/./liteiclink
```
commit 51fb491ed29132b275ccff4b9ab1acb1490166b3 (HEAD -> master, origin/master, origin/HEAD)
Author: Florent Kermarrec <florent@enjoy-digital.fr>
Date:   Fri Dec 13 16:01:46 2019 +0100

    transceiver/serdes_ecp5: finish removing clock_aligner
```
~/dev/upstream/./litex
```
commit f70dd48279cdb54e5da715d4d5f7a743ee2c370d (HEAD -> master, origin/master, origin/HEAD)
Author: Florent Kermarrec <florent@enjoy-digital.fr>
Date:   Fri Jan 10 14:25:46 2020 +0100

    bios/sdram: add memspeed
```

enjoy-digital, 2020-01-11:
It seems you have another version of litex-boards installed somewhere and this is the one that is used, not the one with the updates. Here it's failing, because you don't have https://github.com/litex-hub/litex-boards/commit/82601ff7004fbef5dd6ddb593ff28392c8beaf90.

Can you try to corrupt the hadbadge platform or target and make sure it fails? (to be sure you it's really using this one?). If it's not failing, you have too be sure to remove the old litex-boards repository, or reinstall the curent one. If it's failing with the changes, then please add a print(platform.device) the BaseSoC, then we'll see if https://github.com/litex-hub/litex-boards/commit/82601ff7004fbef5dd6ddb593ff28392c8beaf90 is really applied.

pdp7, 2020-01-11:
@enjoy-digital I wiped everything out and started over fresh from upstream.
However, I now get:
```
pdp7@x1:~/dev/enjoy/linux-on-litex-vexriscv$ python3 make.py --board=hadbadge --build
make: Entering directory '/home/pdp7/dev/enjoy/linux-on-litex-vexriscv/build/hadbadge/software/libcompiler_rt'
make: *** No rule to make target '/home/pdp7/dev/upstream/litex/litex/soc/software/compiler_rt/lib/builtins/umodsi3.c', needed by 'umodsi3.o'.  Stop.
make: Leaving directory '/home/pdp7/dev/enjoy/linux-on-litex-vexriscv/build/hadbadge/software/libcompiler_rt'
Traceback (most recent call last):
  File "make.py", line 318, in <module>
    main()
  File "make.py", line 305, in main
    builder.build()
  File "/home/pdp7/dev/enjoy/litex/litex/soc/integration/builder.py", line 169, in build
    self._generate_software(not self.soc.integrated_rom_initialized)
  File "/home/pdp7/dev/enjoy/litex/litex/soc/integration/builder.py", line 152, in _generate_software
    subprocess.check_call(["make", "-C", dst_dir, "-f", makefile])
  File "/usr/lib/python3.7/subprocess.py", line 347, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['make', '-C', '/home/pdp7/dev/enjoy/linux-on-litex-vexriscv/build/hadbadge/software/libcompiler_rt', '-f', '/home/pdp7/dev/enjoy/litex/litex/soc/software/libcompiler_rt/Makefile']' returned non-zero exit status 2.
```
Maybe I missed a step?

I can run the sim ok:
```
pdp7@x1:~/dev/enjoy/linux-on-litex-vexriscv$ ./sim.py
```
It boots okay.

pdp7, 2020-01-11:
I can build ulx3s okay, but orangecrab also fails with:
```
terminate called after throwing an instance of 'std::out_of_range'
  what():  _Map_base::at
build_top.sh: line 4: 25401 Aborted                 (core dumped) nextpnr-ecp5 --json top.json --lpf top.lpf --textcfg top.config --25k --package CSFBGA285 --speed 8 --timing-allow-fail
Traceback (most recent call last):
  File "make.py", line 318, in <module>
    main()
  File "make.py", line 305, in main
    builder.build()
  File "/home/pdp7/dev/enjoy/litex/litex/soc/integration/builder.py", line 182, in build
    toolchain_path=toolchain_path, **kwargs)
  File "/home/pdp7/dev/enjoy/litex/litex/soc/integration/soc_core.py", line 461, in build
    return self.platform.build(self, *args, **kwargs)
  File "/home/pdp7/dev/enjoy/litex/litex/build/lattice/platform.py", line 34, in build
    return self.toolchain.build(self, *args, **kwargs)
  File "/home/pdp7/dev/enjoy/litex/litex/build/lattice/trellis.py", line 222, in build
    _run_script(script)
  File "/home/pdp7/dev/enjoy/litex/litex/build/lattice/trellis.py", line 152, in _run_script
    raise OSError("Subprocess failed")
OSError: Subprocess failed
```

pdp7, 2020-01-11:
Not sure exactly what my issue was, probably that I have too many different clones of litex in my homedir.  I cleaned everything out again.  And it finally builds again ok.  Sorry for the confusion.

pdp7, 2020-01-11:
I can now build and boot the hadbadge OK.

closed 2020-01-11 by pdp7

mithro, 2020-01-12:
@pdp7 - To prevent this type of problem is exactly why [`litex-buildenv`](https://github.com/timvideos/litex-buildenv) was written.