litex-hub/litex-boards #38: nexys4ddr - NameError: name 'LiteEthPHYMII' is not defined

mithro, 2020-01-17:
```
0.30s$ ./make.py --board=$BOARD
Traceback (most recent call last):
  File "./make.py", line 332, in <module>
    main()
  File "./make.py", line 283, in main
    soc = SoCLinux(board.soc_cls, **soc_kwargs)
  File "/home/travis/build/mithro/linux-on-litex-vexriscv/soc_linux.py", line 162, in SoCLinux
    return _SoCLinux(**kwargs)
  File "/home/travis/build/mithro/linux-on-litex-vexriscv/soc_linux.py", line 55, in __init__
    soc_cls.__init__(self, cpu_type="vexriscv", cpu_variant=cpu_variant, uart_baudrate=1e6, **kwargs)
  File "/home/travis/build/mithro/linux-on-litex-vexriscv/src/litex-boards/litex_boards/official/targets/nexys4ddr.py", line 82, in __init__
    self.submodules.ethphy = LiteEthPHYMII(
NameError: name 'LiteEthPHYMII' is not defined
The command "./make.py --board=$BOARD" exited with 1.
```

enjoy-digital, 2020-01-17:
I'm looking at that.

enjoy-digital, 2020-01-17:
https://github.com/litex-hub/litex-boards/commit/908539d49f14a966cd4f751edb6728a49034806c should do it.

closed 2020-01-17 by enjoy-digital