litex-hub/litex-boards #40: Colorlight-5a-75b: Optimize IP/UDP/Etherbone timings, add SDRAM

enjoy-digital, 2020-01-29:
- [ ] Use 32-bit datapath for the IP/UDP Stack + Etherbone (will be posssible with current work in https://github.com/enjoy-digital/liteeth/pull/21).
- [ ] Reduce `sys_clk_freq`.
- [ ] Integrate SDRAM.

rowanG077, 2021-04-20:
Is there any update on this? I have been eyeing this issue for a year and a half now with no progress :/.

enjoy-digital, 2021-04-20:
I have a version that works yes but still haven't been able to find the time to integrate it properly. As you can imagine, the Colorlight support is more a hobby/low cost project and does not really drive LiteX development so we had to wait to have another opportunity to do these improvements, which has been possible while working on 10G Ethernet.  I hope being able to integrate this in not too long (maybe in the next weeks), it's a bit difficult to help on this issue but if you are willing to contribute to the project, helping on other issues could free up some time to look at this one.

enjoy-digital, 2022-04-25:
This is now possible thanks to @david-sawatzke and @lschuermann's work. I just finished the integration on this with:
- https://github.com/enjoy-digital/liteeth/commit/28c8871624ab71c4b8280167b244a8d6cad73c36
- https://github.com/enjoy-digital/liteeth/commit/d10cda84b6367129f7d839b1f9b90d3b4522f45e
- https://github.com/enjoy-digital/litex/commit/bdbb6c0b3f12c2e6280addab75de1948cc22f034
- https://github.com/enjoy-digital/litex/commit/63cda6c7b90a0bd8b1b724ca217a2059499e85d1

We can now just build the colorlight design with a 32-bit Ethernet Datapath and JTAG-UART with:
```
python3 colorlight_5a_75x.py --uart-name=jtag_uart --cpu-type=vexriscv --with-ethernet --csr-csv=csr.csv --build --load
litex_term jtag --jtag-config=openocd_butterstick.cfg
```

```
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2022 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Apr 25 2022 18:39:49
 BIOS CRC passed (d1a436a2)

 LiteX git sha1: bdbb6c0b

--=============== SoC ==================--
CPU:		VexRiscv @ 60MHz
BUS:		WISHBONE 32-bit @ 4GiB
CSR:		32-bit data
ROM:		128KiB
SRAM:		8KiB
L2:		8KiB
SDRAM:		4096KiB 32-bit @ 60MT/s (CL-2 CWL-2)

--========== Initialization ============--
Ethernet init...
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2.0MiB)...
  Write: 0x40000000-0x40200000 2.0MiB     
   Read: 0x40000000-0x40200000 2.0MiB     
Memtest OK
Memspeed at 0x40000000 (Sequential, 2.0MiB)...
  Write speed: 22.2MiB/s
   Read speed: 30.2MiB/s

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
Timeout
Booting from network...
Local IP: 192.168.1.50
Remote IP: 192.168.1.100
Booting from boot.json...
Booting from boot.bin...
Copying boot.bin to 0x40000000... (7428 bytes)
Executing booted program at 0x40000000

--============= Liftoff! ===============--

LiteX minimal demo app built Apr 25 2022 18:28:30

Available commands:
help               - Show this command
reboot             - Reboot CPU
led                - Led demo
donut              - Spinning Donut demo
helloc             - Hello C
litex-demo-app>
```

The timings results are the following:
```
Info: Max frequency for clock             '$glbnet$main_crg_clkout0': 63.17 MHz (PASS at 60.00 MHz)
Warning: Max frequency for clock '$glbnet$eth_clocks0_rx$TRELLIS_IO_IN': 114.01 MHz (FAIL at 125.00 MHz)
Info: Max frequency for clock                     '$glbnet$jtag_clk': 156.62 MHz (PASS at 12.00 MHz)
```
and the timings violations on eth_clocks0_rx are false paths on the Gray counter of the CDC FIFO so can be ignored (not sure it's now possible to specify false paths with NextPnr).

closed 2022-04-25 by enjoy-digital