litex-hub/litex-boards #43: Colorlight 5A-75B: Connection Timed out by RemoteClient

r4d10n, 2020-02-22:
Hello folks... 

Just trying out litex picorv32 etherbone demo code on Colorlight 5a-75b, following the steps mentioned here: https://github.com/litex-hub/litex-boards/commit/dcc65b347df4c3e3334fce2a6723a793f60a69a6#diff-4c555bb99628703ce903d7c79bc090fa

The gateware seems to get jtagged just fine through my FTDI2232 module. I'm getting the LED blink working fine. I'm not sure whether i'm loading the BIOS/picorv32 binaries along with it. 

After setting up lxserver --udp, RemoteClient fails to connect to etherbone and says "connection timeout". I tried both Eth ports without luck (please also let me know which physical port is phy0 and which is phy1).

```
~/exp/cl5a/litex-boards/litex_boards/targets$ ./remote-colorlight_5a_75b.py 
Traceback (most recent call last):
  File "./remote-colorlight_5a_75b.py", line 21, in <module>
    if wb.regs.uart_xover_rxempty.read() == 0:
  File "/home/pjp/exp/cl5a/litex/litex/tools/remote/csr_builder.py", line 37, in read
    datas = self.readfn(self.addr, length=self.length)
  File "/home/pjp/exp/cl5a/litex/litex/tools/litex_client.py", line 49, in read
    packet = EtherbonePacket(self.receive_packet(self.socket))
  File "/home/pjp/exp/cl5a/litex/litex/tools/remote/etherbone.py", line 365, in receive_packet
    chunk = socket.recv(header_length - len(packet))
socket.timeout: timed out
```
Should i do anything to flash the bios / "software"  code separately ?

enjoy-digital, 2020-02-22:
Hi @r4d10n,

as described in https://github.com/litex-hub/litex-boards/issues/40, there are still some timing optimization to do. But on next monday i could provide you a bitstream that is working on my hardware to see if you are able to reproduce the results with it.

enjoy-digital, 2020-02-24:
@r4d10n: Please find attached a bitstream i just generated and tested on my ColorLight 5A-75B V7.0. Can you verify it also work on your hardware? To test it, connect the Ethernet PHY0 (the one near the power connector) to your network and try to ping `192.168.1.50`.

Note that there are large timings violations that need to be fixed as described in my previous answer, but i haven't seen issues while doing the hardware tests. If it's not working, try to be sure to be connected to a 1Gbps port of a switch/host.

[colorlight_5a_75b_v7.0_2020_02_24_etherbone_phy0.zip](https://github.com/litex-hub/litex-boards/files/4243484/colorlight_5a_75b_v7.0_2020_02_24_etherbone_phy0.zip)

r4d10n, 2020-02-25:
Hello Florent..

Many thanks for the bitstream and the pointers... Seems like the Ethernet in my laptop had driver issues and was not negotiating at 1Gbps. Tried it on my desktop and managed to get the Litex Bios prompt!

Having said that, I'm not able to interact with the terminal -it just hangs. Also it is non-responsive when I break and restart the RemoteClient code. I have to flash the svf again, restart lxserver and RemoteClient once again to get the prompt - still non-interactive. 

I also noticed the timing errors during the nextpnr-ecp5 task: 
```
Warning: Max frequency for clock              '$glbnet$main_crg_clkout': 40.69 MHz (FAIL at 125.00 MHz)
Warning: Max frequency for clock '$glbnet$eth_clocks0_rx$TRELLIS_IO_IN': 96.79 MHz (FAIL at 125.00 MHz)
```
I understand something needs to be done with datawidth extensions. Are there any tasks at hand with regard to that, probably which you can guide through ?

Finally, I would like to know your feedback on running Linux (even if not full-fledged, some microvariant) on the board. I read around about the limitations with regarding to SDRAM, but nothing concrete. Is it worth opening a separate issue here or on the linux-on-litex repos ?

Cheers

enjoy-digital, 2020-02-25:
Thanks @r4d10n for the feedback. The issue you see are expected since the script was only reading data from the UART. If you want to be able to interact with the BIOS, you can use wishbone-tool. I just added instructions to use it: https://github.com/litex-hub/litex-boards/commit/4a84e9b08a5aa7767667ec8d13614cf5511c12bd.

For work to support 32-bit datawidth is related to the work in progress to support 10Gbps: https://github.com/enjoy-digital/liteeth/pull/21 I'm planing to spend some time soon on that so that we can finish and merge the work, that's maybe a bit complicated to start with. If you're familar with FPGA development and would like to work on that i could try to give you directions.

Otherwise, for Linux, the minimum we are currently supporting is 32MB and the Colorlight board has only 4MB. So that would be fine to run smaller OSes (like Zephyr), but Linux would need footprint optimization and i'm not sure would be really useful since the FPGA is not very big.

r4d10n, 2020-02-25:
Once again thanks for the quick response! My bad, I'd noticed in the passing that RemoteClient was doing read-only ;( and now it's all looking good !! 

I'll keep playing with this for a while to familiarize with the LiteX SoC modules and see where it can be taken forward... 

Planning to do the mods for making IO bidirectional... [Bidirectional IO Hack](https://twitter.com/Claude1079/status/1231194849350647808) ... Probably then we can mod LiteScope for doing some fast logic analysis over GigE :) 

Cheers

closed 2020-02-25 by r4d10n