litex-hub/litex-boards #48: Ultrascale+ board DDR initialization failed

fei-g, 2020-02-25:
```
Initializing SDRAM...
SDRAM now under software control
Write leveling:
m0: |11111111111111111111111| delay: 00
m1: |11111111111111111111111| delay: 00
m2: |11111111111111111111111| delay: 00
m3: |11111111111111111111111| delay: 00
m4: |11111111111111111111111| delay: 00
m5: |11111111111111111111111| delay: 00
m6: |11111111111111111111111| delay: 00
m7: |11111111111111111111111| delay: 00
Read leveling:
m0, b0: |00000000000000000000000000000000| delays: -
m0, b1: |00000000000000000000000000000000| delays: -
m0, b2: |00000000000000000000000000000000| delays: -
m0, b3: |00000000000000000000000000000000| delays: -
m0, b4: |00000000000000000000000000000000| delays: -
m0, b5: |00000000000000000000000000000000| delays: -
m0, b6: |00000000000000000000000000000000| delays: -
m0, b7: |11111111111111111111111111111111| delays: 256+-256
best: m0, b7 delays: 256+-256
m1, b0: |00000000000000000000000000000000| delays: -
m1, b1: |00000000000000000000000000000000| delays: -
m1, b2: |00000000000000000000000000000000| delays: -
m1, b3: |00000000000000000000000000000000| delays: -
m1, b4: |00000000000000000000000000000000| delays: -
m1, b5: |00000000000000000000000000000000| delays: -
m1, b6: |00000000000000000000000000000000| delays: -
m1, b7: |11111111111111111111111111111111| delays: 43+-43
best: m1, b7 delays: 106+-106
m2, b0: |00000000000000000000000000000000| delays: -
m2, b1: |00000000000000000000000000000000| delays: -
m2, b2: |00000000000000000000000000000000| delays: -
m2, b3: |00000000000000000000000000000000| delays: -
m2, b4: |00000000000000000000000000000000| delays: -
m2, b5: |00000000000000000000000000000000| delays: -
m2, b6: |00000000000000000000000000000000| delays: -
m2, b7: |11110111111101111011111111111111| delays: 08+-08
best: m2, b7 delays: 16+-15
m3, b0: |00000000000000000000000000000000| delays: -
m3, b1: |00000000000000000000000000000000| delays: -
m3, b2: |00000000000000000000000000000000| delays: -
m3, b3: |00000000000000000000000000000000| delays: -
m3, b4: |00000000000000000000000000000000| delays: -
m3, b5: |00000000000000000000000000000000| delays: -
m3, b6: |00000000000000000000000000000000| delays: -
m3, b7: |11111111111111111111111111111111| delays: 256+-256
best: m3, b7 delays: 256+-256
m4, b0: |00000000000000000000000000000000| delays: -
m4, b1: |00000000000000000000000000000000| delays: -
m4, b2: |00000000000000000000000000000000| delays: -
m4, b3: |00000000000000000000000000000000| delays: -
m4, b4: |00000000000000000000000000000000| delays: -
m4, b5: |00000000000000000000000000000000| delays: -
m4, b6: |00000000000000000000000000000000| delays: -
m4, b7: |11111111111111111111111111111111| delays: 256+-256
best: m4, b7 delays: 256+-256
m5, b0: |00000000000000000000000000000000| delays: -
m5, b1: |00000000000000000000000000000000| delays: -
m5, b2: |00000000000000000000000000000000| delays: -
m5, b3: |00000000000000000000000000000000| delays: -
m5, b4: |00000000000000000000000000000000| delays: -
m5, b5: |00000000000000000000000000000000| delays: -
m5, b6: |00000000000000000000000000000000| delays: -
m5, b7: |11111111111111111111111111111111| delays: 256+-256
best: m5, b7 delays: 256+-256
m6, b0: |00000000000000000000000000000000| delays: -
m6, b1: |00000000000000000000000000000000| delays: -
m6, b2: |00000000000000000000000000000000| delays: -
m6, b3: |00000000000000000000000000000000| delays: -
m6, b4: |00000000000000000000000000000000| delays: -
m6, b5: |00000000000000000000000000000000| delays: -
m6, b6: |00000000000000000000000000000000| delays: -
m6, b7: |11111111111111111111111111111111| delays: 256+-256
best: m6, b7 delays: 256+-256
m7, b0: |00000000000000000000000000000000| delays: -
m7, b1: |00000000000000000000000000000000| delays: -
m7, b2: |00000000000000000000000000000000| delays: -
m7, b3: |00000000000000000000000000000000| delays: -
m7, b4: |00000000000000000000000000000000| delays: -
m7, b5: |00000000000000000000000000000000| delays: -
m7, b6: |00000000000000000000000000000000| delays: -
m7, b7: |00000000000000000000000000000000| delays: 12+-08
best: m7, b7 delays: 17+-08
SDRAM now under hardware control
Memtest bus failed: 256/256 errors
Memtest data failed: 524288/524288 errors
Memtest addr failed: 8191/8192 errors
Memory initialization failed
```

enjoy-digital, 2020-02-25:
@fei-g: can you do a test with https://github.com/litex-hub/litex-boards/commit/83d2c7109911951a4b756a1441ba5245ce2d4b78?

fei-g, 2020-02-25:
> @fei-g: can you do a test with [83d2c71](https://github.com/litex-hub/litex-boards/commit/83d2c7109911951a4b756a1441ba5245ce2d4b78)?

I also add the `"SLEW=FAST"` constraints to the DDR pins.
Still the similar result, except the write leveling shows all zeros
```
Write leveling:
m0: |00000000000000000000000| delay: 00
m1: |00000000000000000000000| delay: 00
m2: |00000000000000000000000| delay: 00
m3: |00000000000000000000000| delay: 00
m4: |00000000000000000000000| delay: 00
m5: |00000000000000000000000| delay: 00
m6: |00000000000000000000000| delay: 00
m7: |00000000000000000000000| delay: 00
```

daveshah1, 2020-02-25:
Not sure if it makes a difference for the speeds LiteX uses, but back when I was using litedram on the ZCU104 (a few months ago) I used POD12_DCI/DIFF_POD12_DCI rather than POD12 for the data/mask/strobe pins.

fei-g, 2020-02-25:
> Not sure if it makes a difference for the speeds LiteX uses, but back when I was using litedram on the ZCU104 (a few months ago) I used POD12_DCI/DIFF_POD12_DCI rather than POD12 for the data/mask/strobe pins.

Thanks for the suggestion. Tried that, but it didn't make a difference. For the IO standard constraints, it's weird that the master xdc file(suggest POD12 for data pins) does not match the board user guide(suggest POD12_DCI).  Something I don't understand, in the document UG571: 

> IMPORTANT: In banks where the input I/O standard has an input reference voltage requirement and uses an internally generated VREF (INTERNAL_VREF or VREF scan), connect the dedicated VREF pin to GND with a 500Ω or 1KΩ resistor.

How can I connect Vref pin to GND with a resistor? Is it the reason that it didn't work?

daveshah1, 2020-02-25:
Looking at the schematic, it appears as if Vref is totally disconnected so adding the resistor wouldn't be possible (not using internal Vref, and consequently using external Vref isn't an option either)
[image]
I can only presume that UG571 is out of date and the resistor isn't actually required, or DDR4 would be totally unusable on the VCU118.

OTOH, the 240 ohm resistor between VRP and ground is present, so using the DCI IO type is definitely an option.

fei-g, 2020-02-25:
I see. Really appreciate the explanation!

enjoy-digital, 2020-02-26:
@fei-g: https://github.com/litex-hub/litex-boards/commit/e516ff345267be095688e34a5ea61d1545459672 changes the IO settings to be as close as possible to the ones used by the Xilinx's MIG. I'm not able to test on VCU118, but I validated these settings on the KCU105.

fei-g, 2020-02-26:
> @fei-g: [e516ff3](https://github.com/litex-hub/litex-boards/commit/e516ff345267be095688e34a5ea61d1545459672) changes the IO settings to be as close as possible to the ones used by the Xilinx's MIG. I'm not able to test on VCU118, but I validated these settings on the KCU105.

I tried this version. Before bitstream_gen, Vivado(2017.1) gave me these errors on drama_dm pins:
`ERROR: [DRC PORTPROP-14] Attribute value compatibility: Ports cannot have IOSTANDARD=POD12_DCI, BANK_TYPE=HP, EQUALIZATION=EQ_LEVEL2, and DIRECTION=OUT. Port(s): ddram_dm[0].  For High Performance banks, Non-default EQUALIZATION attribute values are not allowed when IO_TYPE is set to OUTPUT`

It could generat the bitstream after I removed the EQUALIZATION constraints for drama_dm. But the memory initialization still failed.

enjoy-digital, 2020-03-10:
Hi @fei-g,

would you mind doing a test on the VCU118 with the current code (just be sure to also update LiteX/LiteDRAM) and if not working, doing another test with `cmd_latency` set to 1? If you do so, can you post the logs?

Thanks

fei-g, 2020-03-11:
Hi @enjoy-digital ,
```
--========== Initialization ============--
Initializing SDRAM...
SDRAM now under software control
Write leveling:
m0: |111111111| delay: 00
m1: |111111111| delay: 00
m2: |111111111| delay: 00
m3: |111111111| delay: 00
m4: |111111111| delay: 00
m5: |111111111| delay: 00
m6: |111111111| delay: 00
m7: |111111111| delay: 00
Read leveling:
m0, b0: |00000000000000000001111111111111| delays: 401+-110
m0, b1: |00000000000000000000000000000000| delays: -
m0, b2: |00000000000000000000000000000000| delays: -
m0, b3: |00000000000000000000000000000000| delays: -
m0, b4: |00000000000000000000000000000000| delays: -
m0, b5: |00000000000000000000000000000000| delays: -
m0, b6: |00000000000000000000000000000000| delays: -
m0, b7: |11111111111111111000000000000000| delays: 128+-128
best: m0, b7 delays: 129+-129
m1, b0: |00000000000000000001111111111111| delays: 400+-112
m1, b1: |00000000000000000000000000000000| delays: -
m1, b2: |00000000000000000000000000000000| delays: -
m1, b3: |00000000000000000000000000000000| delays: -
m1, b4: |00000000000000000000000000000000| delays: -
m1, b5: |00000000000000000000000000000000| delays: -
m1, b6: |00000000000000000000000000000000| delays: -
m1, b7: |11111111111111110000000000000000| delays: 124+-124
best: m1, b7 delays: 125+-125
m2, b0: |00000000000000000001111111111111| delays: 400+-112
m2, b1: |00000000000000000000000000000000| delays: -
m2, b2: |00000000000000000000000000000000| delays: -
m2, b3: |00000000000000000000000000000000| delays: -
m2, b4: |00000000000000000000000000000000| delays: -
m2, b5: |00000000000000000000000000000000| delays: -
m2, b6: |00000000000000000000000000000000| delays: -
m2, b7: |11111111111111110000000000000000| delays: 123+-123
best: m2, b7 delays: 123+-123
m3, b0: |00000000000000000011111111111111| delays: 400+-112
m3, b1: |00000000000000000000000000000000| delays: -
m3, b2: |00000000000000000000000000000000| delays: -
m3, b3: |00000000000000000000000000000000| delays: -
m3, b4: |00000000000000000000000000000000| delays: -
m3, b5: |00000000000000000000000000000000| delays: -
m3, b6: |00000000000000000000000000000000| delays: -
m3, b7: |11111111111111110000000000000000| delays: 121+-121
best: m3, b7 delays: 124+-124
m4, b0: |00000000000000000011111111111111| delays: 397+-114
m4, b1: |00000000000000000000000000000000| delays: -
m4, b2: |00000000000000000000000000000000| delays: -
m4, b3: |00000000000000000000000000000000| delays: -
m4, b4: |00000000000000000000000000000000| delays: -
m4, b5: |00000000000000000000000000000000| delays: -
m4, b6: |00000000000000000000000000000000| delays: -
m4, b7: |11111111111111110000000000000000| delays: 122+-122
best: m4, b7 delays: 122+-122
m5, b0: |00000000000000000011111111111111| delays: 399+-112
m5, b1: |00000000000000000000000000000000| delays: -
m5, b2: |00000000000000000000000000000000| delays: -
m5, b3: |00000000000000000000000000000000| delays: -
m5, b4: |00000000000000000000000000000000| delays: -
m5, b5: |00000000000000000000000000000000| delays: -
m5, b6: |00000000000000000000000000000000| delays: -
m5, b7: |11111111111111111000000000000000| delays: 127+-127
best: m5, b7 delays: 127+-127
m6, b0: |00000000000000000001111111111111| delays: 401+-111
m6, b1: |00000000000000000000000000000000| delays: -
m6, b2: |00000000000000000000000000000000| delays: -
m6, b3: |00000000000000000000000000000000| delays: -
m6, b4: |00000000000000000000000000000000| delays: -
m6, b5: |00000000000000000000000000000000| delays: -
m6, b6: |00000000000000000000000000000000| delays: -
m6, b7: |11111111111111110000000000000000| delays: 125+-125
best: m6, b7 delays: 126+-126
m7, b0: |00000000000000000001111111111111| delays: 405+-106
m7, b1: |00000000000000000000000000000000| delays: -
m7, b2: |00000000000000000000000000000000| delays: -
m7, b3: |00000000000000000000000000000000| delays: -
m7, b4: |00000000000000000000000000000000| delays: -
m7, b5: |00000000000000000000000000000000| delays: -
m7, b6: |00000000000000000000000000000000| delays: -
m7, b7: |01111111111111111000000000000000| delays: 128+-127
best: m7, b7 delays: 128+-128
SDRAM now under hardware control
Memtest bus failed: 256/256 errors
Memtest data failed: 524288/524288 errors
Memtest addr failed: 8191/8192 errors
Memory initialization failed
```
This is the log with `cmd_latency=1`, and it has no difference with `cmd_latency=0`.

enjoy-digital, 2020-03-11:
Thanks @fei-g, we start seeing valid patterns for the read leveling which is good. The write leveling is not done correctly, are you sure USDDRPHY is correctly defined with `self.add_constant("USDDRPHY", None)`?  (and not `"USPDDRPHY"`) It should iterate on more taps as you can see here: https://github.com/litex-hub/litex-boards/issues/54#issuecomment-597130869.

enjoy-digital, 2020-03-11:
If you still have a short write leveling window, can you share the generated software files?

fei-g, 2020-03-13:
> If you still have a short write leveling window, can you share the generated software files?

Thanks for your updates. I double checked the code I used, it's `self.add_constant("USDDRPHY", None)`. However, it's weird that the write leveling is still short of taps. I attached the generated software here. Thanks for checking that.
[software.zip](https://github.com/litex-hub/litex-boards/files/4330127/software.zip)

enjoy-digital, 2020-03-13:
Thanks, sorry i've not yet been able to look at it but will do it soon.

fei-g, 2020-04-30:
The automatic scan of the command delay added in [this commit](https://github.com/enjoy-digital/litex/commit/ab92e81e3103d3de6a643949570adaab70d2d981) fixed the write leveling! 

Together with setting `cmd_latency = 1`, part of the DRAMs could pass the memory test.
In this test, I was using the channel2 DDR on VCU118:
```
--========== Initialization ============--
Initializing SDRAM...
SDRAM now under software control
Write leveling:
Command/Clk scan:
|00000011|000111111|000111111|011111111| best: 361
Data scan:
m0: |0111111111111111110000| delay: 01
m1: |0011111111111111111100| delay: 27
m2: |0000001111111111111111| delay: 95
m3: |0000111111111111111111| delay: 59
m4: |0000000111111111111111| delay: 108
m5: |0000000011111111111111| delay: 121
m6: |0000011111111111111111| delay: 76
m7: |0000000111111111111111| delay: 109
Read leveling:
m0, b0: |00000000000000000000000000000000| delays: -
m0, b1: |00000000000000000000000000000000| delays: -
m0, b2: |00000000000000000000000000000000| delays: -
m0, b3: |11111111111000000000000000000000| delays: 85+-85
m0, b4: |00000000000000111111111111111000| delays: 328+-122
m0, b5: |00000000000000000000000000000000| delays: 505+-08
m0, b6: |00000000000000000000000000000000| delays: -
m0, b7: |00000000000000000000000000000000| delays: -
best: m0, b4 delays: 328+-124
m1, b0: |00000000000000000000000000000000| delays: -
m1, b1: |00000000000000000000000000000000| delays: -
m1, b2: |00000000000000000000000000000000| delays: -
m1, b3: |11111111111100000000000000000000| delays: 91+-91
m1, b4: |00000000000000111111111111111100| delays: 345+-123
m1, b5: |00000000000000000000000000000000| delays: 516+-08
m1, b6: |00000000000000000000000000000000| delays: -
m1, b7: |00000000000000000000000000000000| delays: -
best: m1, b4 delays: 339+-124
m2, b0: |00000000000000000000000000000000| delays: -
m2, b1: |00000000000000000000000000000000| delays: -
m2, b2: |00000000000000000000000000000000| delays: -
m2, b3: |11111111111000000000000000000000| delays: 84+-84
m2, b4: |00000000000000111111111111111000| delays: 331+-121
m2, b5: |00000000000000000000000000000001| delays: 503+-08
m2, b6: |00000000000000000000000000000000| delays: -
m2, b7: |00000000000000000000000000000000| delays: -
best: m2, b4 delays: 329+-121
m3, b0: |00000000000000000000000000000000| delays: -
m3, b1: |00000000000000000000000000000000| delays: -
m3, b2: |00000000000000000000000000000000| delays: -
m3, b3: |11111111100000000000000000000000| delays: 68+-68
m3, b4: |00000000000011111111111111000000| delays: 296+-118
m3, b5: |00000000000000000000000000000011| delays: 491+-20
m3, b6: |00000000000000000000000000000000| delays: -
m3, b7: |00000000000000000000000000000000| delays: -
best: m3, b4 delays: 297+-118
m4, b0: |00000000000000000000000000000000| delays: -
m4, b1: |00000000000000000000000000000000| delays: -
m4, b2: |00000000000000000000000000000000| delays: -
m4, b3: |00000000000000000000000000000000| delays: -
m4, b4: |00000000000000000000000000000000| delays: -
m4, b5: |00000000000000000000000000000000| delays: -
m4, b6: |00000000000000000000000000000000| delays: -
m4, b7: |00000000000000000000000000000000| delays: -
best: m4, b0 delays: -
m5, b0: |00000000000000000000000000000000| delays: -
m5, b1: |00000000000000000000000000000000| delays: -
m5, b2: |00000000000000000000000000000000| delays: -
m5, b3: |00000000000000000000000000000000| delays: -
m5, b4: |00000000000000000000000000000000| delays: -
m5, b5: |00000000000000000000000000000000| delays: -
m5, b6: |00000000000000000000000000000000| delays: -
m5, b7: |00000000000000000000000000000000| delays: -
best: m5, b0 delays: -
m6, b0: |00000000000000000000000000000000| delays: -
m6, b1: |00000000000000000000000000000000| delays: -
m6, b2: |00000000000000000000000000000000| delays: -
m6, b3: |11100000000000000000000000000000| delays: 18+-18
m6, b4: |00000000000000000000000000000000| delays: -
m6, b5: |00000000000000000000000000000000| delays: -
m6, b6: |00000000000000000000000000000000| delays: -
m6, b7: |00000000000000000000000000000000| delays: -
best: m6, b3 delays: 19+-19
m7, b0: |00000000000000000000000000000000| delays: -
m7, b1: |00000000000000000000000000000000| delays: -
m7, b2: |00000000000000000000000000000000| delays: -
m7, b3: |11110000000000000000000000000000| delays: 25+-25
m7, b4: |00000011111111111111100000000000| delays: 208+-122
m7, b5: |00000000000000000000000011111111| delays: 445+-67
m7, b6: |00000000000000000000000000000000| delays: -
m7, b7: |00000000000000000000000000000000| delays: -
best: m7, b4 delays: 208+-122
SDRAM now under hardware control
Memtest bus failed: 48/256 errors
Memtest data failed: 262144/524288 errors
Memtest addr failed: 4096/8192 errors
Memory initialization failed
```

at least, we can just use the first 32 bits of channel 2 as a workaround. It looks similar to the problem in #49.

enjoy-digital, 2020-04-30:
@fei-g: Thanks for the test and feedback, good we have some progress! To only use the first 4 modules as a workaround, you could use the new `PHYPadsReducer` in your target:

````python3
from litedram.common import PHYPadsReducer
ddram_pads = platform.request("ddram")
ddram_pads = PHYPadsReducer(ddram_pads, modules=[0, 1, 2, 3])
self.submodules.ddrphy = usddrphy.USPDDRPHY(ddram_pads,
[...]
````

Could you do a test with that?

fei-g, 2020-04-30:
> @fei-g: Thanks for the test and feedback, good we have some progress! To only use the first 4 modules as a workaround, you could use the new `PHYPadsReducer` in your target:
> 
> ```python
> from litedram.common import PHYPadsReducer
> ddram_pads = platform.request("ddram")
> ddram_pads = PHYPadsReducer(ddram_pads, modules=[0, 1, 2, 3])
> self.submodules.ddrphy = usddrphy.USPDDRPHY(ddram_pads,
> [...]
> ```
> 
> Could you do a test with that?

That works. Memtest OK.

closed 2020-06-24 by fei-g