litex-hub/litex-boards #53: icebreaker: target features

enjoy-digital, 2020-03-08:
The icebreaker target was inspired from the Fomu design. I started doing some simplifications with the idea to have something closer to the others targets and just provide a SoC with similar parameters than others SoCs that would run the LiteX BIOS from the SPI Flash. The others SoCs are able to embed the integrated ROM so are not using the SPI Flash for the BIOS so this would provide a good example for this use case.

But, maybe it's not going in the direction you initially wanted @esden, so let's discuss things here, so that i continue or revert my changes :) Are you ok with the direction i'm describing? Do you need specific debug/vexriscv_debug from Fomu in the target or just be able to select the cpu/variant as on others targets is fine?

enjoy-digital, 2020-03-10:
@esden: i've been playing a bit with the iCEBreaker yesterday, that's really a nice board, easy to use and it will help improving iCE40 support in LiteX, i should have tried it before! (FYI i have the original one, not the cloned version from china :))

I created a target in LiteX directly (to avoid breaking yours) with some changes from your original target https://github.com/enjoy-digital/litex/blob/master/litex/boards/targets/icebreaker.py:
- more similar to others targets (most of the parameters are passed from the common line, like cpu-type, cpu-variant, etc...), ex: ./icebreaker.py --cpu-type=vexriscv or ./icebreaker.py --cpu-type=picorv32, etc... 
- the only hardcoded parameters are the ROM and SRAM parameters to use the SPI Flash and SPRAM.
- the Jump module with fake ROM has been removed since no longer necessary.
- the Yosys/NextPnr settings have also been removed, since are no longer needed for the iCEBreaker. It will still probably be required for Fomu that packs more logic in the FPGA, but from the test i did yesterday, the "-relut -abc2 -dffe_min_ce_use 4 -relut" config was not improving things that much on the iCEBreaker with this simple design.
- A PLL has been integrated to run the design at 24MHz.

So maybe some things could be useful for your use case and the target in litex-boards.

esden, 2020-03-10:
Awesome! That looks great! I will integrate your changes into my target. I can update the target in litex_boards if you want to be similar to the one I have in the icebreaker example repository. I think that will be the last time (except bug fixes in the future) because the example in my repository is growing and getting custom local RTL, I don't think it makes sense to add things like that in litex_boards.

enjoy-digital, 2020-03-10:
Or do you want the one in litex-boards to be similar to the one i did yesterday with a link to your example design in the header to do more things with the board? (it's just that it's easier for me to maintain things if targets are not too customized and have some similarities between each others :))

esden, 2020-03-13:
Ok, I have now updated my example script to include all your improvements too. I think the best solution at the moment would be to either remove the target from litex_boards or synchronize it with the one you have in main litex repository. I don't think it makes sense to maintain three versions of the script in three repositories. 

Whatever you decide to do, it is probably a good idea to include a link to the [icebreaker-litex-examples](https://github.com/icebreaker-fpga/icebreaker-litex-examples) repository. The script in that repository contains known to work defaults, autogenerates and updates documentation and headers for C and Rust as well as READMEs with instructions on how to use it and which options are worth exploring, as well as example applications to run on the resulting SOC. 

I think the target scripts in the official repository should be easy for you to maintain, and be the most generic and simplified solution. As long as we tell people that there is additional material to be found we should be good.

Let me know what you think. :)

enjoy-digital, 2020-03-13:
Thanks for the feedback. https://github.com/litex-hub/litex-boards/commit/a99d25841135d40430bf5749ff86c2c218b4751b integrates the minimal version in litex_boards (that i'll be able to maintain easily) and adds a link to the Crowsupply campaign and to the more complete example you are building. Does it seem fine for you?

esden, 2020-03-13:
That looks great! Thank you! I think with that we can now close this issue. :)

enjoy-digital, 2020-03-13:
Great. The iCEBreaker is very convenient to improve iCE40 support in LiteX, i'll try to play more with it :)

closed 2020-03-13 by enjoy-digital