litex-hub/litex-boards #54: Mercury XU5: DDR initialization failed

enjoy-digital, 2020-03-09:
The write leveling succeeds but read leveling fails:
```
--========== Initialization ============--
Initializing SDRAM...
SDRAM now under software control
Write leveling:
m0: |000000001111111111111| delay: 127
m1: |000000000111111111111| delay: 131
Read leveling:
m0, b0: |00000000000000000000000000000000| delays: -
m0, b1: |00000000000000000000000000000000| delays: -
m0, b2: |00000000000000000000000000000000| delays: -
m0, b3: |00000000000000000000000000000000| delays: -
m0, b4: |00000000000000000000000000000000| delays: -
m0, b5: |00000000000000000000000000000000| delays: -
m0, b6: |00000000000000000000000000000000| delays: -
m0, b7: |00000000000000000000000000000000| delays: -
best: m0, b0 delays: -
m1, b0: |00000000000000000000000000000000| delays: -
m1, b1: |00000000000000000000000000000000| delays: -
m1, b2: |00000000000000000000000000000000| delays: -
m1, b3: |00000000000000000000000000000000| delays: -
m1, b4: |00000000000000000000000000000000| delays: -
m1, b5: |00000000000000000000000000000000| delays: -
m1, b6: |00000000000000000000000000000000| delays: -
m1, b7: |00000000000000000000000000000000| delays: -
best: m1, b0 delays: -
SDRAM now under hardware control
Memtest bus failed: 128/256 errors
Memtest data failed: 524288/524288 errors
Memtest addr failed: 8192/8192 errors
Memory initialization failed

--============= Console ================--
litex>
```

enjoy-digital, 2020-03-09:
@jedrzejboczar i've reviewed the Mercury XU5 platform support for the DDR4 and don't see obvious reasons it would fail. To go further in the investigation, i would recommend  using similar settings than on the KCU105: 125MHz sys_clk, use similar IOs constraints. If things are not improving, you could add some traces around to see the written/read test pattern and try to understand what is happening:
https://github.com/enjoy-digital/litex/blob/master/litex/soc/software/bios/sdram.c#L513
https://github.com/enjoy-digital/litex/blob/master/litex/soc/software/bios/sdram.c#L540

enjoy-digital, 2020-03-09:
Some debug functions have been added to the BIOS with https://github.com/enjoy-digital/litex/commit/72af1b39ebcbac93be8a0db15f1ccab681cda405. To enable it, you will just need to add the `USDDRPHY_DEBUG `constant to your design: `self.add_constant("USDDRPHY_DEBUG", None)`

This will allow you to reconfigure the Mode Registers of the DDR4 (with `sdram_mrwr`) and redo just the calibration (with `sdram_cal`). That could be useful to test different electrical settings than the default ones. The script attached could be helpful to re-generate the register values.

````
#!/usr/bin/env python3

# default electrical settings (point to point)
rtt_nom = "40ohm"
rtt_wr  = "120ohm"
ron     = "34ohm"

def format_mr1(dll_enable, ron, rtt_nom):
    mr1 = dll_enable
    mr1 |= ((ron >> 0) & 0b1) << 1
    mr1 |= ((ron >> 1) & 0b1) << 2
    mr1 |= ((rtt_nom >> 0) & 0b1) << 8
    mr1 |= ((rtt_nom >> 1) & 0b1) << 9
    mr1 |= ((rtt_nom >> 2) & 0b1) << 10
    return mr1

def format_mr2(cwl, rtt_wr):
    cwl_to_mr2 = {
       9: 0b000,
      10: 0b001,
      11: 0b010,
      12: 0b011,
      14: 0b100,
      16: 0b101,
      18: 0b110,
      20: 0b111
    }
    mr2 = cwl_to_mr2[cwl] << 3
    mr2 |= rtt_wr << 9
    return mr2

z_to_rtt_nom = {
    "disabled" : 0b000,
    "60ohm"    : 0b001,
    "120ohm"   : 0b010,
    "40ohm"    : 0b011,
    "240ohm"   : 0b100,
    "48ohm"    : 0b101,
    "80ohm"    : 0b110,
    "34ohm"    : 0b111
}

z_to_rtt_wr = {
    "disabled" : 0b000,
    "120ohm"   : 0b001,
    "240ohm"   : 0b010,
    "high-z"   : 0b011,
    "80ohm"    : 0b100,
}

z_to_ron = {
    "34ohm" : 0b00,
    "48ohm" : 0b01,
}

print("sdram_mrwr 1 {:d}".format(format_mr1(1, z_to_ron[ron], z_to_rtt_nom[rtt_nom])))
print("sdram_mrwr 2 {:d}".format(format_mr2(9, z_to_rtt_wr[rtt_wr])))
````

jedrzejboczar, 2020-03-09:
Thanks for the debug functions, they'll be useful. 

Until now I've been able to find out that the problem is only with phase 4 reads, all other phases seem to give correct results.

For reference I'm attaching a log ([rd-level-scan.bin.txt.zip](https://github.com/litex-hub/litex-boards/files/4306882/rd-level-scan.bin.txt.zip)) with results of adding [here](https://github.com/enjoy-digital/litex/blob/72af1b39ebcbac93be8a0db15f1ccab681cda405/litex/soc/software/bios/sdram.c#L618) something like:
```
printf("0x%02x 0x%02x  ",
    prs[p][  NBMODULES-1-module] ^ tst[  NBMODULES-1-module],
    prs[p][2*NBMODULES-1-module] ^ tst[2*NBMODULES-1-module]
);
```

enjoy-digital, 2020-03-10:
Interesting, so it's probably a timing issue and not an electrical issue. The PHY is configured with `cmd_latency` set to 1, while the default is 0. Is there a reason for this? If there is no reason, i would recommend using 0.

The next step if this is still failing with `cmd_latency=0` will be to understand if it's a read issue or write issue. To do so, DDR4 memories now provide an integrated pre-defined pattern that can be enabled as done in `sdrmpr` function:
````
sdrmpron(MPR0_SEL);
command_prd(DFII_COMMAND_CAS|DFII_COMMAND_CS|DFII_COMMAND_RDDATA);
cdelay(15);
````

You could enable` MPR0` before the read_leveling and instead of checking for the written data, check for MPR0 pattern. If this is succeeding with that, it means the reads are probably working an issue are on the writes.

Depending the results, i'll try to give more indications.

jedrzejboczar, 2020-03-10:
Thanks! `cmd_latency` was wrong indeed. Now everything works correctly.
I'll create a PR in a moment.

closed 2020-03-10 by enjoy-digital

enjoy-digital, 2020-03-10:
Great! Can you post the full BIOS prompt just to have a trace somewhere of the calibration results?

jedrzejboczar, 2020-03-10:
Sure, here it is:
```
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2020 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Mar 10 2020 15:50:15
 BIOS CRC passed (1992d462)

 Migen git sha1: e2e6c72
 LiteX git sha1: 72af1b39

--=============== SoC ==================--
CPU:       VexRiscv @ 125MHz
ROM:       32KB
SRAM:      4KB
L2:        8KB
MAIN-RAM:  524288KB

--========== Initialization ============--
Initializing SDRAM...
SDRAM now under software control
Write leveling:
m0: |00000001111111111111111| delay: 106
m1: |00000001111111111111111| delay: 109
Read leveling:
m0, b0: |00000000000000000000000000000000| delays: -
m0, b1: |01111111111111110000000000000000| delays: 122+-121
m0, b2: |00000000000000000011111111111111| delays: 394+-118
m0, b3: |00000000000000000000000000000000| delays: -
m0, b4: |00000000000000000000000000000000| delays: -
m0, b5: |00000000000000000000000000000000| delays: -
m0, b6: |00000000000000000000000000000000| delays: -
m0, b7: |00000000000000000000000000000000| delays: -
best: m0, b1 delays: 122+-120
m1, b0: |00000000000000000000000000000000| delays: -
m1, b1: |11111111111111000000000000000000| delays: 104+-104
m1, b2: |00000000000000001111111111111110| delays: 362+-115
m1, b3: |00000000000000000000000000000000| delays: -
m1, b4: |00000000000000000000000000000000| delays: -
m1, b5: |00000000000000000000000000000000| delays: -
m1, b6: |00000000000000000000000000000000| delays: -
m1, b7: |00000000000000000000000000000000| delays: -
best: m1, b2 delays: 362+-117
SDRAM now under hardware control
Memtest OK
Memspeed Writes: 311Mbps Reads: 393Mbps

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
             Timeout
No boot medium found

--============= Console ================--
litex>
```

enjoy-digital, 2020-03-10:
Thanks, the write/read leveling scans are looking good.