litex-hub/litex-boards #63: Does the de1-soc target supports using the hard processor?

ilesser, 2020-04-04:


ilesser, 2020-04-04:
If it does not, I would like to contribute and try to add the support for the HPS

rob-ng15, 2020-04-04:
That would be good, as it would unlock additional capabilities on other de
boards.

ilesser, 2020-04-04:
Do you mean for all Cyclone V FPGAs?

From what I understand of the code I should modify the [SoCCore](https://github.com/enjoy-digital/litex/blob/master/litex/soc/integration/soc_core.py#L52) or create a new class.

I should modify the add_cpu() method to use the HPS instead of a softcpu.

Should I do that inside the [de1soc.py](https://github.com/litex-hub/litex-boards/blob/master/litex_boards/targets/de1soc.py)?

ilesser, 2020-04-04:
According the the [documentation](https://www.intel.com/content/www/us/en/programmable/documentation/cru1463082726154.html) the device only supports AXI-3.

[image]

But the conde from litex [axi interconnect](https://github.com/enjoy-digital/litex/blob/master/litex/soc/interconnect/axi.py) implements AXI-4 Full/Lite.

Will it be compatible to AXI-3?

enjoy-digital, 2020-04-13:
@ilesser: adding support for the HPS could be useful yes (as we are supporting creating SoCs aroudn the Zynq PS). We could had support to AXI-3, the difference with AXI-4 should not be very big and mostly located on the burst capabilities i think. You could look at [SoCZynq](https://github.com/enjoy-digital/litex/blob/master/litex/soc/integration/soc_zynq.py) to find some inspiration.

ilesser, 2020-04-13:
> @ilesser: adding support for the HPS could be useful yes (as we are supporting creating SoCs aroudn the Zynq PS). We could had support to AXI-3, the difference with AXI-4 should not be very big and [...]

Thanks!
I was reading about axi3 and axi4 and there's are only a few extra signals between them.

I have already forked litex and I am working in a soc_cv.py.
I will create an issue there and a pull request.
I want to use axi instead of wishbone and that will take me a while.
I am also going through some qsys projects to try to isolate an IP that can be used for the instantiation.

enjoy-digital, 2020-04-13:
The easiest solution is to bridge the AXI to the Wishbone as we are doing on the Zynq. This will be easier to merge and there are also some work planned to support AXI directly in the SoC.

enjoy-digital, 2020-04-14:
To avoid too much duplication, let's continue the discussion in https://github.com/enjoy-digital/litex/issues/466.

closed 2020-04-14 by enjoy-digital