litex-hub/litex-boards #66: ulx3s.py bitstream fails memory test

gojimmypi, 2020-04-12:
the [ulx3s.py](https://github.com/litex-hub/litex-boards/blob/master/litex_boards/targets/ulx3s.py) is now failing memory test:

```
$ $WORKSPACE/ulx3s-examples/bin/ujprog.exe top.bit
ULX2S / ULX3S JTAG programmer v 3.0.92 (built Feb 18 2019 10:55:47)
Using USB cable: ULX3S FPGA 12K v3.0.3
Programming: 100%
Completed in 63.63 seconds.
gojimmypi:/mnt/c/workspace/litex-boards/litex_boards/targets/soc_basesoc_ulx3s/gateware
$ litex_term --serial-boot --kernel bios.bin /dev/ttyS15
[LXTERM] Starting....

litex> memtest
Memtest bus failed: 30/256 errors
Memtest data failed: 48626/524288 errors
Memtest addr failed: 594/8192 errors
```
my [standard build is here](https://github.com/gojimmypi/ulx3s-toolchain/blob/68a7048b6654f9cd6ba05c82162e5c353a543592/soft_cpu.sh#L22), but I've also tried a more explicit match of RAM chip that I have with the same results:

```
./ulx3s.py --sdram-module AS4C32M16 --device LFE5U-85F --cpu-type picorv32
```

I've run [emard's memtest](https://github.com/emard/ulx3s-bin/tree/master/fpga/memtest), specifically the `ulx3s_85f_memtest_32MB_190MHz_300deg.bit`  and confirmed memory seems ok. 
[image]

(top row yellow not used; next row: numbers of tests; bottom row: errors) 

is this a known problem or perhaps I am doing something wrong here? I had this working not long ago.

enjoy-digital, 2020-04-13:
The default ULX3S target in litex-boards has been tested successfully a few days ago. An issue in Nextpnr was preventing it from working reliably which could explain some instability. Some changes have also been done in LiteX to be sure IO registers are instantiated. The issue was discussed in https://github.com/enjoy-digital/litex/issues/461. Can you be sure you are using an up to date Nextpnr and up to date LiteX/LiteDRAMLiteX-boards repositories and do the test again?

gojimmypi, 2020-04-14:
Well - memtest is working again! (although I'm not sure why)

What's the best way to determine versions are installed and use? (e.g. `nextpnr` is not in my path)

I found this python egg file:
```
usr\local\lib\python3.6\dist-packages\migen-0.8.dev0-py3.6.egg
```
This is my `migen` repo:
```
gojimmypi:/mnt/c/workspace/migen
$ git log
commit 19d5eae29b9ff57ea11540252f79ad1a9526ff3a (HEAD -> master, origin/master, origin/HEAD)
Author: Sebastien Bourdeauducq <sb@m-labs.hk>
Date:   Sat Apr 11 15:50:25 2020 +0800

    zc706: fix user_led I/O standards

    It seems that the board user guide is incorrect, but the schematic is right.
```
this is my NextPNR:
```
gojimmypi:/mnt/c/workspace/nextpnr
$ git log
commit 96c14abd1f79a152f3e5a1b4897ca2e7a91e7b27 (HEAD -> master, origin/master, origin/HEAD)
Author: Miodrag Milanovic <mmicko@gmail.com>
Date:   Sat Apr 11 22:05:30 2020 +0200

    Add TRELLIS_PROGRAM_PREFIX

gojimmypi:/mnt/c/workspace/nextpnr
$ git fetch
gojimmypi:/mnt/c/workspace/nextpnr
$ git status
On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean
```
The only instances of `nextpnr` on my entire drive are directories in: `\rootfs\home\gojimmypi\temp` and `rootfs\home\gojimmypi\workspace`

my local LiteX litedram:
```
gojimmypi:/mnt/c/workspace/litedram
$ git log
commit 64c2be5d63d388f98ddd0585ed13254ec048a122 (HEAD -> master, origin/master, origin/HEAD)
Author: Florent Kermarrec <florent@enjoy-digital.fr>
Date:   Sat Apr 11 18:55:16 2020 +0200

    README: switch to markdown.

gojimmypi:/mnt/c/workspace/litedram

$ git fetch
gojimmypi:/mnt/c/workspace/litedram
$ git status
On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean
```
the only thing I had done is go back in time to try to see if I could get it working again when it was last known good:
```
git checkout master@{3/22/2020}
```
but I undid that at fetched the latest `litex_boards` and it is working again:
```
gojimmypi:/mnt/c/workspace/litex-boards/litex_boards/targets/soc_basesoc_ulx3s/gateware
$ git reset --hard
Checking out files: 100% (69/69), done.
HEAD is now at 188d4a4 targets: use DDROutput on sdram_clock and similar configuration for all SDRAM targets.
gojimmypi:/mnt/c/workspace/litex-boards/litex_boards/targets/soc_basesoc_ulx3s/gateware
$ git checkout master
Already on 'master'
Your branch is behind 'origin/master' by 3 commits, and can be fast-forwarded.
  (use "git pull" to update your local branch)
gojimmypi:/mnt/c/workspace/litex-boards/litex_boards/targets/soc_basesoc_ulx3s/gateware
$ git pull
Updating 188d4a4..52c9648
Fast-forward
 litex_boards/platforms/arty_s7.py | 213 ++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
 litex_boards/targets/arty_s7.py   |  90 +++++++++++++++++++++++++++++++++
 test/test_targets.py              |   3 ++
 3 files changed, 306 insertions(+)
 create mode 100644 litex_boards/platforms/arty_s7.py
 create mode 100755 litex_boards/targets/arty_s7.py
gojimmypi:/mnt/c/workspace/litex-boards/litex_boards/targets/soc_basesoc_ulx3s/gateware
$ git fetch
gojimmypi:/mnt/c/workspace/litex-boards/litex_boards/targets/soc_basesoc_ulx3s/gateware
$ git pull
Already up to date.
gojimmypi:/mnt/c/workspace/litex-boards/litex_boards/targets/soc_basesoc_ulx3s/gateware
```

enjoy-digital, 2020-04-14:
Hi @gojimmypi, thanks for the feedback and great it's working. In the last days, an issue has been found and fixed in Nextpnr. Before this fix, the ULX3S target has been working, but it was not reliable between builds. Before the Nextpnr issue was found, we also did some investigation/improvements while investigating to be sure that IO regs are used with GENSDRPHY. The recommended way for developers to have up to date LiteX is to use the `litex_setup.py` script and use the `update` command. Nextpnr, LiteX are moving a lot with new features/improvements/fixes, so i would recommend using up to date versions when possible. The ULX3S will be regularly tested and i think we can close this issue.

closed 2020-04-14 by enjoy-digital