litex-hub/litex-boards #68: Mercury XU5: memory initialization fails at 125 MHz

jedrzejboczar, 2020-04-27:
When working on https://github.com/enjoy-digital/litex/pull/472 I had to decrease the `sys_clk_freq` because at frequency of 125 MHz the memory initialization fails. The highest working frequency I could get was 120 MHz (with 480 MHz for IDELAYCTRL).

It looks as if, for higher frequency, clock/command delay has much bigger influence than DQ/DQS delay, e.g.
```
litex> sdram_cdly 27
Write leveling:
M0: |00000000000000000000000| delay: -1
M1: |00000000000000000000000| delay: -1

litex> sdram_cdly 28
Write leveling:
M0: |11111111111111111111111| delay: -1
M1: |00000000000000000000000| delay: -1

litex> sdram_cdly 29
Write leveling:
M0: |11111111111111111111111| delay: -1
M1: |00000000000000000000000| delay: -1

litex> sdram_cdly 30
Write leveling:
M0: |11111111111111111111111| delay: -1
M1: |00000000000000000000000| delay: 197

litex> sdram_cdly 31
Write leveling:
M0: |11111111111111111111111| delay: -1
M1: |11111111111111111111111| delay: -1
```
Sometimes (as here for cdly 30) a single scan can be different than the rest. But when I added printing the number of the 1s registered, it was always either 0/128 or 128/128, so all the loops for a given delay were giving the same result. It looks like the problem is in incrementing the write delay, but inserting additional delays in software didn't help.

enjoy-digital, 2020-05-06:
@jedrzejboczar: it seems that with the new cmd/clk delay calibration, using cmd_latency of 1 gives better results on the different boards. Could you do a test with the XU5 with the current upstream code? (cmd_latency has been set to 1 with https://github.com/litex-hub/litex-boards/commit/da61aabc5bc44f0dda9007c68381c2df797300f1)

jedrzejboczar, 2020-05-06:
Unfortunately it doesn't work. 
To have a clear view I've tested the following configurations with all repos on current master:

| cmd_latency | sys_clk_freq | write leveling | read leveling |
| ----------- | ------------ | -------------- | ------------- |
|           0 |      100 MHz |             ok |            ok |
|           0 |      120 MHz |             ok |            ok |
|           0 |      125 MHz |           fail |          fail |
|           1 |      100 MHz |             ok |          fail |
|           1 |      120 MHz |   fail (m1 ok) |          fail |
|           1 |      125 MHz |           fail |          fail |

enjoy-digital, 2020-05-06:
Thanks for the tests, i'll do more tests on other boards to try to understand what is happening. I prefer we keep this open and don't reduce the sys_clk_freq for now since this should be working.

enjoy-digital, 2020-11-19:
@jedrzejboczar: do you still have access to the board? If so, could you do a test to verify the behavior with the recent improvements?

jedrzejboczar, 2020-11-19:
Yes, I will try to test it tomorrow.

jedrzejboczar, 2020-11-20:
Unfortunatelly there are still problems. It seems that I was able to calibrate it by forcing cmd_delay, but writing does not work then. This output is from test with not L2 cache, but I tested also with cache. I used a fresh Litex install.
```
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2020 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Nov 20 2020 14:30:32
 BIOS CRC passed (81779db8)

 Migen git sha1: a5cc037
 LiteX git sha1: 8e39060d

--=============== SoC ==================--
CPU:            VexRiscv @ 125MHz
BUS:            WISHBONE 32-bit @ 4GiB
CSR:            32-bit data
ROM:            32KiB
SRAM:           8KiB
L2:             0KiB
SDRAM:          524288KiB 16-bit @ 1000MT/s (CL-9 CWL-9)

--========== Initialization ============--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Write leveling:
  Cmd/Clk scan:
  |11110000  |11111  |11111  |11111| best: 0
  Setting Cmd/Clk delay to 0 taps.
  Data scan:
  m0: |00000011111111111111111| delay: 93
  m1: |00000001111111111111111| delay: 97
Write latency calibration:
m0:0 m1:0
Read leveling:
  m0, b0: |00000000000000000000000000000000| delays: -
  m0, b1: |00000000000000000000000000000000| delays: -
  m0, b2: |00000000000000000000000000000000| delays: -
  m0, b3: |00000000000000000000000000000000| delays: -
  m0, b4: |00000000000000000000000000000000| delays: -
  m0, b5: |00000000000000000000000000000000| delays: -
  m0, b6: |00000000000000000000000000000000| delays: -
  m0, b7: |00000000000000000000000000000000| delays: -
  best: m0, b00 delays: -
  m1, b0: |00000000000000000000000000000000| delays: -
  m1, b1: |00000000000000000000000000000000| delays: -
  m1, b2: |00000000000000000000000000000000| delays: -
  m1, b3: |00000000000000000000000000000000| delays: -
  m1, b4: |00000000000000000000000000000000| delays: -
  m1, b5: |00000000000000000000000000000000| delays: -
  m1, b6: |00000000000000000000000000000000| delays: -
  m1, b7: |00000000000000000000000000000000| delays: -
  best: m1, b00 delays: -
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2MiB)...
  Write: 0x40000000-0x40200000 2MiB
   Read: 0x40000000-0x40200000 2MiB
  bus errors:  256/256
  addr errors: 8192/8192
  data errors: 524288/524288
Memtest KO
Memory initialization failed

--============= Console ================--

litex> mem_read 0x40000000 0x100
[... 77 lines cut ...]
0x400000d0  78 c4 de da 9c 2f 7d c8 ce 97 3e 64 e7 4b 1f 32  x..../}...>d.K.2
0x400000e0  dd ef c4 60 f8 d2 97 4c 7c e9 4b 26 be f4 25 13  ...`...L|.K&..%.
0x400000f0  58 02 94 df 2c 7d e9 84 96 be 74 42 4b 5f 3a 21  X...,}....tBK_:!

litex> mem_write 0x4000000c 0xaabbccdd 0x20

litex> mem_read 0x40000000 0x100
Memory dump:
0x40000000  07 09 15 2a 00 00 00 00 00 00 00 00 00 00 00 00  ...*............
0x40000010  40 c4 df dc 02 00 36 d8 01 00 1b 6c 03 80 2d b6  @.....6....l..-.
0x40000020  ce ad cc 04 01 60 9b 6d 03 b0 ed b6 02 d8 56 db  .....`.m......V.
0x40000030  00 88 de ce 03 b6 f5 b6 02 db 5a db 81 6d ad 6d  ..........Z..m.m
0x40000040  5a dd 84 a0 62 5b 5b db b1 ad ad 6d db d6 f6 b6  Z...b[[....m....
0x40000050  c8 52 6e ee b7 b5 ad 6d d8 da f6 b6 6c 6d 7b 5b  .Rn....m....lm{[
0x40000060  e3 de 04 40 5b db de 16 ae 6d 4f 8b d7 b6 a7 45  ...@[....mO....E
0x40000070  6c 4c dc dc b4 ed 79 51 da f6 bc 28 6d 7b 5e 14  lL....yQ...(m{^.
0x40000080  ce ce ce 44 d9 9e 27 c5 6f cf b3 e2 b4 e7 79 f1  ...D..'.o.....y.
0x40000090  4a 84 ce e4 ed 79 5e 3c f5 3c 0f 9e 79 9e 27 cf  J....y^<.<..y.'.
0x400000a0  fe d6 f0 60 9c e7 f9 f3 ce f3 fc 79 e7 79 fe 3c  ...`.......y.y.<
0x400000b0  4c 80 cd c1 78 9e 2f 4f 3c cf 97 27 9e e7 cb 13  L...x./O<..'....
0x400000c0  ee 8f 54 4c e4 f9 d2 84 f2 7c 69 42 79 be 34 21  ..TL.....|iBy.4!
0x400000d0  78 c4 de da 9c 2f 7d c8 ce 97 3e 64 e7 4b 1f 32  x..../}...>d.K.2
0x400000e0  dd ef c4 60 f8 d2 97 4c 7c e9 4b 26 be f4 25 13  ...`...L|.K&..%.
0x400000f0  58 02 94 df 2c 7d e9 84 96 be 74 42 4b 5f 3a 21  X...,}....tBK_:!
```

jedrzejboczar, 2020-11-20:
It worked after changing `cmd_latency` to 0:
```
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2020 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Nov 20 2020 14:57:59
 BIOS CRC passed (02f12266)

 Migen git sha1: a5cc037
 LiteX git sha1: 8e39060d

--=============== SoC ==================--
CPU:            VexRiscv @ 125MHz
BUS:            WISHBONE 32-bit @ 4GiB
CSR:            32-bit data
ROM:            32KiB
SRAM:           8KiB
L2:             0KiB
SDRAM:          524288KiB 16-bit @ 1000MT/s (CL-9 CWL-9)

--========== Initialization ============--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Write leveling:
  Cmd/Clk scan:
  |11110000  |11111  |11111  |11111| best: 0
  Setting Cmd/Clk delay to 0 taps.
  Data scan:
  m0: |00000011111111111111111| delay: 93
  m1: |00000001111111111111111| delay: 97
Write latency calibration:
m0:0 m1:0
Read leveling:
  m0, b0: |00000000000000000000000000000000| delays: -
  m0, b1: |01111111111111110000000000000000| delays: 128+-122
  m0, b2: |00000000000000000011111111111111| delays: 393+-118
  m0, b3: |00000000000000000000000000000000| delays: -
  m0, b4: |00000000000000000000000000000000| delays: -
  m0, b5: |00000000000000000000000000000000| delays: -
  m0, b6: |00000000000000000000000000000000| delays: -
  m0, b7: |00000000000000000000000000000000| delays: -
  best: m0, b01 delays: 129+-123
  m1, b0: |00000000000000000000000000000000| delays: -
  m1, b1: |11111111111111100000000000000000| delays: 111+-111
  m1, b2: |00000000000000000111111111111111| delays: 379+-122
  m1, b3: |00000000000000000000000000000000| delays: -
  m1, b4: |00000000000000000000000000000000| delays: -
  m1, b5: |00000000000000000000000000000000| delays: -
  m1, b6: |00000000000000000000000000000000| delays: -
  m1, b7: |00000000000000000000000000000000| delays: -
  best: m1, b02 delays: 378+-120
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2MiB)...
  Write: 0x40000000-0x40200000 2MiB
   Read: 0x40000000-0x40200000 2MiB
Memtest OK
Memspeed at 0x40000000 (2MiB)...
  Write speed: 41MiB/s
   Read speed: 19MiB/s

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
             Timeout
No boot medium found

--============= Console ================--

litex> sdram_test
Memtest at 0x40000000 (16MiB)...
  Write: 0x40000000-0x41000000 16MiB
   Read: 0x40000000-0x41000000 16MiB
Memtest OK

litex>
```

closed 2020-11-20 by enjoy-digital