litex-hub/litex-boards #93: Missing TinyFPGA BX target

nalzok, 2020-07-28:
TinyFPGA BX is listed as a supported open-source board in README, but there are no corresponding script in `litex-boards/litex_boards/targets/`.

While we do have `litex-boards/litex_boards/platforms/tinyfpga_bx.py`, I’m not sure if it’s ready to use?

enjoy-digital, 2020-11-12:
Sorry @nalzok for the delay, a tinyfpga_bx target has been added with https://github.com/litex-hub/litex-boards/commit/7c6df677394c4bd77309dbb56dde501cbc08c2be. The iCE40 is too small for the default VexRiscv CPU, but you can build a SERV based SoC with:` ./tinyfpga_bx.py --cpu-type=serv --build`

nalzok, 2020-11-12:
Thanks for the great work! I will check it out when I can get my hands on my TinyFPGA BX.

> The iCE40 is too small for the default VexRiscv CPU

I see... Would it take a lot of effort to use [Murax SOC](https://github.com/SpinalHDL/VexRiscv#murax-soc) instead?

enjoy-digital, 2020-11-12:
Sorry, it's possible to use VexRiscv but not with the standard variant that is too large for the iCE40 (and would also be with Murax SoC). But you can use the Lite variant of Vexriscv:` ./tinyfpga_bx.py --cpu-type=vexriscv --cpu-variant=lite --build`

nalzok, 2020-11-12:
> (and would also be with Murax SoC)

I'm a little surprised to learn this. [ICESugar](https://github.com/wuxx/icesugar), an iCE40UP5K-powered development board, [is able to run Murax SOC](https://www.muselab-tech.com/zai-icesugarshang-yun-xing-muraxsoc/) (link in Chinese), and I can confirm that since I also own that board. TinyFPGA BX has 8K logic elements, so I would expect it to do more. I'll do some investigation once I have some free time...

mithro, 2020-11-12:
@nalzok - The VexRISCV definitely fits on the iCE40UP5K and the TinyFPGA BX. You just have to disable things like the hard multiplier and divider which is what the `lite` variant does.

nalzok, 2020-11-12:
Got it! Thanks @mithro  for the explanation. Feel free to close the issue if you have nothing to add! @enjoy-digital

mithro, 2020-11-12:
I tried to start documenting some of this stuff at https://github.com/enjoy-digital/litex/wiki/Soft-CPU

closed 2020-11-19 by enjoy-digital