litex-hub/litex-boards #97: RAM and Ethernet support for Zybo

rlipperts, 2020-08-04:
I was working with the Zybo 7z020 and am not sure if there is an easy way to add an ethernet port (for the wishbone-tool) and the RAM to my target (which is using a softcore on the PL).
As I still am fairly new in all of this I wondered if you can provide me with some insights.

- The [zybo_z7 platform](https://github.com/litex-hub/litex-boards/blob/master/litex_boards/platforms/zybo_z7.py) introduces ps7_mio and ps7_ddram as extensions without ever specifying concrete Pins, how does that work?
- As the ethernet port is directly connected to a PS controller (which may be accessible over ps7_mio?) is there an easy way add this to a target?
- The Zybo Rev. b 7z020 platform and target I created have the same functionality as the ones for the zybo rev. A which you added here, should I work on a pull-request for that?

enjoy-digital, 2020-11-19:
Hi @rlipperts,

sorry the delay. I think i did something very similar to what you want to do while testing LiteX on the SDS1104X-E Oscilloscpe:
https://twitter.com/enjoy_digital/status/1327212606256541702

You can find the platform file here:
https://github.com/litex-hub/litex-boards/blob/master/litex_boards/platforms/sds1104xe.py
and target file here:
https://github.com/litex-hub/litex-boards/blob/master/litex_boards/targets/sds1104xe.py

Since MIO and DDRAM are fixed and already known by Vivado, providing them is not mandatory, but this can also be done, as it has been done on the ZedBoard: https://github.com/litex-hub/litex-boards/blob/master/litex_boards/platforms/zedboard.py#L60-L97.

For the Zybo Rev.b, if there are some similarity between boards, we could try to add support for both revisions in the same platform, as done for example on the OrangeCrab: https://github.com/litex-hub/litex-boards/blob/master/litex_boards/platforms/orangecrab.py.

rlipperts, 2020-11-27:
I appreciate the heads up. I am currently not working with the Zybo anymore (And quite busy with my masters thesis). If I pick up the zybo again I will definetly look into this, thank you!

enjoy-digital, 2020-11-27:
@rlipperts thanks for the feedback.

closed 2020-11-27 by enjoy-digital