litex-hub/litex-boards #99: zybo_z7: ENET0_MDIO : must be mio instead of EMIO and SDIO0 WP is not available

trabucayre, 2020-08-06:
According to the schematic MDIO (MDC, MDIO) must be connected to PS_MIO052_501 and PS_MIO053_501 respectivelly.

In xci file the line 607 is wrong:
```
<spirit:configurableElementValue spirit:referenceId="PARAM_VALUE.PCW_ENET0_GRP_MDIO_IO">EMIO</spirit:configurableElementValue>
```
and must be 
```
<spirit:configurableElementValue spirit:referenceId="PARAM_VALUE.PCW_ENET0_GRP_MDIO_IO">MIO 52 .. 53</spirit:configurableElementValue>
```

SDIO_WP in not used
line 666:
```
<spirit:configurableElementValue spirit:referenceId="PARAM_VALUE.PCW_EN_EMIO_WP_SDIO0">1</spirit:configurableElementValue>
```
and line 1118
```
<spirit:configurableElementValue spirit:referenceId="PARAM_VALUE.PCW_SD0_GRP_WP_IO">EMIO</spirit:configurableElementValue>
```
must be

```
<spirit:configurableElementValue spirit:referenceId="PARAM_VALUE.PCW_EN_EMIO_WP_SDIO0">0</spirit:configurableElementValue>
<spirit:configurableElementValue spirit:referenceId="PARAM_VALUE.PCW_SD0_GRP_WP_ENABLE">0</spirit:configurableElementValue>
<spirit:configurableElementValue spirit:referenceId="PARAM_VALUE.PCW_SD0_GRP_WP_IO">&lt;Select></spirit:configurableElementValue>
```

I would be appy to apply by myself this change but I don't know how to provide modified file since it isn't under version control.

enjoy-digital, 2020-11-19:
Sorry for the delay @trabucayre, i'll try to look at this soon.

trabucayre, 2020-11-19:
No worry.
Thanks

enjoy-digital, 2022-03-24:
This has finally been updated... :) with https://github.com/litex-hub/litex-boards/commit/c081177d77f37a4ea6cff150d42a69bd6f0abbc2.

closed 2022-03-24 by enjoy-digital