litex-hub/litex-boards #105: icebreaker - no BIOS on serial

biot, 2020-08-31:
When I follow the instructions on https://github.com/enjoy-digital/litex (point 4 is missing BTW) I don't get the BIOS output on serial as shown in point 5.

using `icebreaker.py --build` builds fine without errors, using `--load` also works fine but does nothing, and using `--flash` looks like it flashes the icebreaker, but still no serial output. However in that case the board does a slow LED flashing, alternating between green and red LEDs.

Can you confirm I didn't miss anything? Maybe flesh out point 4 in the docs?

enjoy-digital, 2020-09-01:
Thanks @biot for reporting. It seems the execution of the BIOS from SPI Flash was broken. I updated the target and fixed that, could you do a test with latest LiteX/LiteX-boards? `./icebreaker.py --cpu-type=serv --build --flash`

biot, 2020-09-01:
Yes, that's fixed it, thanks very much!

Incidentally with the "serv" CPU core I get:
   BIOS CRC failed (expected 000000000000000000, got 0000000000000000g0)
   The system will continue, but expect problems.

...but all is well with vexriscv.

closed 2020-09-01 by biot