litex-hub/litex-boards #107: DRAM failed on AC701

elliott-wen, 2020-09-05:
Hi, boys and gals. Anyone has any idea why DRAM fails on AC701 boards?

I have two low-end AC701 boards, they all failed on dram test. I tried self-test programs from Xilinx, DRAM seems fine.

I didn't change anything on source code. Built the bitstream using official repo straight away. 

--============= Console ================--

litex> 
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2020 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Sep  5 2020 17:33:11
 BIOS CRC passed (73288079)

 Migen git sha1: 7bc4eb1
 LiteX git sha1: f7b6dd05

--=============== SoC ==================--
CPU:		VexRiscv @ 100MHz
BUS:		WISHBONE 32-bit @ 4GiB
CSR:		8-bit data
ROM:		32KiB
SRAM:		8KiB
L2:		8KiB
SDRAM:		1048576KiB 64-bit @ 800Mbps/pin

--========== Initialization ============--
Initializing DRAM @0x40000000...
SDRAM now under software control
SDRAM now under software control
Read leveling:
m0, b00: |00000000000000000000000000000000| delays: -
m0, b01: |00000000000000000000000000000000| delays: -
m0, b02: |00000000000000000000000000000000| delays: -
m0, b03: |00000000000000000000000000000000| delays: -
m0, b04: |00000000000000000000000000000000| delays: -
m0, b05: |00000000000000000000000000000000| delays: -
m0, b06: |00000000000000000000000000000000| delays: -
m0, b07: |00000000000000000000000000000000| delays: -
m0, b08: |00000000000000000000000000000000| delays: -
m0, b09: |00111111111110000000000000000000| delays: 07+-05
m0, b10: |00000000000000000011111111111000| delays: 23+-05
m0, b11: |00000000000000000000000000000000| delays: -
m0, b12: |00000000000000000000000000000000| delays: -
m0, b13: |00000000000000000000000000000000| delays: -
m0, b14: |00000000000000000000000000000000| delays: -
m0, b15: |00000000000000000000000000000000| delays: -
best: m0, b09 delays: 07+-05
m1, b00: |00000000000000000000000000000000| delays: -
m1, b01: |00000000000000000000000000000000| delays: -
m1, b02: |00000000000000000000000000000000| delays: -
m1, b03: |00000000000000000000000000000000| delays: -
m1, b04: |00000000000000000000000000000000| delays: -
m1, b05: |00000000000000000000000000000000| delays: -
m1, b06: |00000000000000000000000000000000| delays: -
m1, b07: |00000000000000000000000000000000| delays: -
m1, b08: |00000000000000000000000000000000| delays: -
m1, b09: |01111111111110000000000000000000| delays: 07+-06
m1, b10: |00000000000000000111111111111000| delays: 23+-06
m1, b11: |00000000000000000000000000000000| delays: -
m1, b12: |00000000000000000000000000000000| delays: -
m1, b13: |00000000000000000000000000000000| delays: -
m1, b14: |00000000000000000000000000000000| delays: -
m1, b15: |00000000000000000000000000000000| delays: -
best: m1, b09 delays: 07+-06
m2, b00: |00000000000000000000000000000000| delays: -
m2, b01: |00000000000000000000000000000000| delays: -
m2, b02: |00000000000000000000000000000000| delays: -
m2, b03: |00000000000000000000000000000000| delays: -
m2, b04: |00000000000000000000000000000000| delays: -
m2, b05: |00000000000000000000000000000000| delays: -
m2, b06: |00000000000000000000000000000000| delays: -
m2, b07: |00000000000000000000000000000000| delays: -
m2, b08: |00000000000000000000000000000000| delays: -
m2, b09: |00000000000000000000000000000000| delays: -
m2, b10: |00000000000000000000000000000000| delays: -
m2, b11: |00000000000000000000000000000000| delays: -
m2, b12: |00000000000000000000000000000000| delays: -
m2, b13: |00000000000000000000000000000000| delays: -
m2, b14: |00000000000000000000000000000000| delays: -
m2, b15: |00000000000000000000000000000000| delays: -
best: m2, b00 delays: -
m3, b00: |00000000000000000000000000000000| delays: -
m3, b01: |00000000000000000000000000000000| delays: -
m3, b02: |00000000000000000000000000000000| delays: -
m3, b03: |00000000000000000000000000000000| delays: -
m3, b04: |00000000000000000000000000000000| delays: -
m3, b05: |00000000000000000000000000000000| delays: -
m3, b06: |00000000000000000000000000000000| delays: -
m3, b07: |00000000000000000000000000000000| delays: -
m3, b08: |00000000000000000000000000000000| delays: -
m3, b09: |00000000000000000000000000000000| delays: -
m3, b10: |00000000000000000000000000000000| delays: -
m3, b11: |00000000000000000000000000000000| delays: -
m3, b12: |00000000000000000000000000000000| delays: -
m3, b13: |00000000000000000000000000000000| delays: -
m3, b14: |00000000000000000000000000000000| delays: -
m3, b15: |00000000000000000000000000000000| delays: -
best: m3, b00 delays: -
m4, b00: |00000000000000000000000000000000| delays: -
m4, b01: |00000000000000000000000000000000| delays: -
m4, b02: |00000000000000000000000000000000| delays: -
m4, b03: |00000000000000000000000000000000| delays: -
m4, b04: |00000000000000000000000000000000| delays: -
m4, b05: |00000000000000000000000000000000| delays: -
m4, b06: |00000000000000000000000000000000| delays: -
m4, b07: |00000000000000000000000000000000| delays: -
m4, b08: |00000000000000000000000000000000| delays: -
m4, b09: |00000000000000000000000000000000| delays: -
m4, b10: |00000000000000000000000000000000| delays: -
m4, b11: |00000000000000000000000000000000| delays: -
m4, b12: |00000000000000000000000000000000| delays: -
m4, b13: |00000000000000000000000000000000| delays: -
m4, b14: |00000000000000000000000000000000| delays: -
m4, b15: |00000000000000000000000000000000| delays: -
best: m4, b00 delays: -
m5, b00: |00000000000000000000000000000000| delays: -
m5, b01: |00000000000000000000000000000000| delays: -
m5, b02: |00000000000000000000000000000000| delays: -
m5, b03: |00000000000000000000000000000000| delays: -
m5, b04: |00000000000000000000000000000000| delays: -
m5, b05: |00000000000000000000000000000000| delays: -
m5, b06: |00000000000000000000000000000000| delays: -
m5, b07: |00000000000000000000000000000000| delays: -
m5, b08: |00000000000000000000000000000000| delays: -
m5, b09: |00000000000000000000000000000000| delays: -
m5, b10: |00000000000000000000000000000000| delays: -
m5, b11: |00000000000000000000000000000000| delays: -
m5, b12: |00000000000000000000000000000000| delays: -
m5, b13: |00000000000000000000000000000000| delays: -
m5, b14: |00000000000000000000000000000000| delays: -
m5, b15: |00000000000000000000000000000000| delays: -
best: m5, b00 delays: -
m6, b00: |00000000000000000000000000000000| delays: -
m6, b01: |00000000000000000000000000000000| delays: -
m6, b02: |00000000000000000000000000000000| delays: -
m6, b03: |00000000000000000000000000000000| delays: -
m6, b04: |00000000000000000000000000000000| delays: -
m6, b05: |00000000000000000000000000000000| delays: -
m6, b06: |00000000000000000000000000000000| delays: -
m6, b07: |00000000000000000000000000000000| delays: -
m6, b08: |00000000000000000000000000000000| delays: -
m6, b09: |00000000000000000000000000000000| delays: -
m6, b10: |00000000000000000000000000000000| delays: -
m6, b11: |00000000000000000000000000000000| delays: -
m6, b12: |00000000000000000000000000000000| delays: -
m6, b13: |00000000000000000000000000000000| delays: -
m6, b14: |00000000000000000000000000000000| delays: -
m6, b15: |00000000000000000000000000000000| delays: -
best: m6, b00 delays: -
m7, b00: |00000000000000000000000000000000| delays: -
m7, b01: |00000000000000000000000000000000| delays: -
m7, b02: |00000000000000000000000000000000| delays: -
m7, b03: |00000000000000000000000000000000| delays: -
m7, b04: |00000000000000000000000000000000| delays: -
m7, b05: |00000000000000000000000000000000| delays: -
m7, b06: |00000000000000000000000000000000| delays: -
m7, b07: |00000000000000000000000000000000| delays: -
m7, b08: |00000000000000000000000000000000| delays: -
m7, b09: |00000000000000000000000000000000| delays: -
m7, b10: |00000000000000000000000000000000| delays: -
m7, b11: |00000000000000000000000000000000| delays: -
m7, b12: |00000000000000000000000000000000| delays: -
m7, b13: |00000000000000000000000000000000| delays: -
m7, b14: |00000000000000000000000000000000| delays: -
m7, b15: |00000000000000000000000000000000| delays: -
best: m7, b00 delays: -
SDRAM now under hardware control
Memtest at 0x40000000...
[########################################]
[########################################]
- bus errors:  256/256
- addr errors: 4095/8192
- data errors: 524285/524288
Memtest KO
Memory initialization failed

--============= Console ================--

enjoy-digital, 2020-09-05:
Hi @elliott-wen,

can you try with https://github.com/litex-hub/litex-boards/commit/e4cdbe0f7ad0653e825556d992d233a1723273e3, other users have used this configuration successfully, this reduces the dram data-width to 32-bits but should allow you to get things working.

enjoy-digital, 2020-09-05:
Here is the generated bitstream with this configuration. [ac701_2020_09_05.zip](https://github.com/litex-hub/litex-boards/files/5177871/ac701_2020_09_05.zip)

elliott-wen, 2020-09-05:
Hi, thanks for your fast reply.

Unfortunately, I tried both the generated bitstream and the modified
ac701.py. None of them seems to work.

I am using vivado 2020 to generate my own bitstream, will it be the problem?

itex>
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2020 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Sep  6 2020 00:05:14
 BIOS CRC passed (23a0bfac)

 Migen git sha1: 7bc4eb1
 LiteX git sha1: f7b6dd05

--=============== SoC ==================--
CPU: VexRiscv @ 100MHz
BUS: WISHBONE 32-bit @ 4GiB
CSR: 8-bit data
ROM: 32KiB
SRAM: 8KiB
L2: 8KiB
SDRAM: 524288KiB 32-bit @ 800Mbps/pin

--========== Initialization ============--
Initializing DRAM @0x40000000...
SDRAM now under software control
SDRAM now under software control
Read leveling:
m0, b00: |00000000000000000000000000000000| delays: -
m0, b01: |00000000000000000000000000000000| delays: -
m0, b02: |00000000000000000000000000000000| delays: -
m0, b03: |00000000000000000000000000000000| delays: -
m0, b04: |00000000000000000000000000000000| delays: -
m0, b05: |00000000000000000000000000000000| delays: -
m0, b06: |00000000000000000000000000000000| delays: -
m0, b07: |00000000000000000000000000000000| delays: -
m0, b08: |00000000000000000000000000000000| delays: -
m0, b09: |00111111111111000000000000000000| delays: 08+-06
m0, b10: |00000000000000000001111111111110| delays: 25+-06
m0, b11: |00000000000000000000000000000000| delays: -
m0, b12: |00000000000000000000000000000000| delays: -
m0, b13: |00000000000000000000000000000000| delays: -
m0, b14: |00000000000000000000000000000000| delays: -
m0, b15: |00000000000000000000000000000000| delays: -
best: m0, b09 delays: 08+-06
m1, b00: |00000000000000000000000000000000| delays: -
m1, b01: |00000000000000000000000000000000| delays: -
m1, b02: |00000000000000000000000000000000| delays: -
m1, b03: |00000000000000000000000000000000| delays: -
m1, b04: |00000000000000000000000000000000| delays: -
m1, b05: |00000000000000000000000000000000| delays: -
m1, b06: |00000000000000000000000000000000| delays: -
m1, b07: |00000000000000000000000000000000| delays: -
m1, b08: |00000000000000000000000000000000| delays: -
m1, b09: |11111111111110000000000000000000| delays: 06+-06
m1, b10: |00000000000000000111111111111000| delays: 22+-06
m1, b11: |00000000000000000000000000000000| delays: -
m1, b12: |00000000000000000000000000000000| delays: -
m1, b13: |00000000000000000000000000000000| delays: -
m1, b14: |00000000000000000000000000000000| delays: -
m1, b15: |00000000000000000000000000000000| delays: -
best: m1, b09 delays: 07+-06
m2, b00: |00000000000000000000000000000000| delays: -
m2, b01: |00000000000000000000000000000000| delays: -
m2, b02: |00000000000000000000000000000000| delays: -
m2, b03: |00000000000000000000000000000000| delays: -
m2, b04: |00000000000000000000000000000000| delays: -
m2, b05: |00000000000000000000000000000000| delays: -
m2, b06: |00000000000000000000000000000000| delays: -
m2, b07: |00000000000000000000000000000000| delays: -
m2, b08: |00000000000000000000000000000000| delays: -
m2, b09: |00000000000000000000000000000000| delays: -
m2, b10: |00000000000000000000000000000000| delays: -
m2, b11: |00000000000000000000000000000000| delays: -
m2, b12: |00000000000000000000000000000000| delays: -
m2, b13: |00000000000000000000000000000000| delays: -
m2, b14: |00000000000000000000000000000000| delays: -
m2, b15: |00000000000000000000000000000000| delays: -
best: m2, b00 delays: -
m3, b00: |00000000000000000000000000000000| delays: -
m3, b01: |00000000000000000000000000000000| delays: -
m3, b02: |00000000000000000000000000000000| delays: -
m3, b03: |00000000000000000000000000000000| delays: -
m3, b04: |00000000000000000000000000000000| delays: -
m3, b05: |00000000000000000000000000000000| delays: -
m3, b06: |00000000000000000000000000000000| delays: -
m3, b07: |00000000000000000000000000000000| delays: -
m3, b08: |00000000000000000000000000000000| delays: -
m3, b09: |00000000000000000000000000000000| delays: -
m3, b10: |00000000000000000000000000000000| delays: -
m3, b11: |00000000000000000000000000000000| delays: -
m3, b12: |00000000000000000000000000000000| delays: -
m3, b13: |00000000000000000000000000000000| delays: -
m3, b14: |00000000000000000000000000000000| delays: -
m3, b15: |00000000000000000000000000000000| delays: -
best: m3, b00 delays: -
SDRAM now under hardware control
Memtest at 0x40000000...
[########################################]
[########################################]
- bus errors:  256/256
- addr errors: 0/8192
- data errors: 524282/524288
Memtest KO
Memory initialization failed

enjoy-digital, 2020-09-07:
Hi @elliott-wen,

given your results, if you want something working but with limited bandwidth/size, you can use `PHYPadsReducer(platform.request("ddram"), [0, 1])`. I will check the electrical settings to see if something could be missing or not configured correctly.

elliott-wen, 2020-09-08:
Hi, currently I am using the DRAM IP core from Xilinx. But I am still quite
looking forward to litex-dram, which is significantly better in terms of
resource consumption.

Cheers
Elliott

enjoy-digital, 2020-11-19:
Hi @elliott-wen,

write latency calibration has been added recently to LiteDRAM and this could improve the behavior on the AC701. I don't have this board unfortunately, would you mind doing a test with the current version of LiteDRAM/LiteX?

elliott-wen, 2020-11-19:
Will have a check and report back to you.

enjoy-digital, 2022-05-27:
We can probably close this now, the first 4 modules are working on AC701. More time would need to be spend to get the 8 working.

closed 2022-05-27 by enjoy-digital