litex-hub/litex-boards #119: Custom Zync board

flaminggoat, 2020-10-26:
I am trying to develop a custom zynq board. Can anyone point to how I can generate this xci file that the zybo target uses?

https://github.com/litex-hub/litex-boards/blob/931f6667ac424d0bcd7dca0cf90d3e0033cce5df/litex_boards/targets/zybo_z7.py#L59

flaminggoat, 2020-11-02:
I found that the XCI file can be generated by using "Create and package new IP", then copying the ps7 xci file from the ip directory. However when using my xci file I get the following error: 
```
INFO: [Synth 8-6157] synthesizing module 'FDPE' [/home/theo/Xilinx/Vivado/Vivado/2020.1/scripts/rt/data/unisim_comp.v:13664]
        Parameter INIT bound to: 1'b1 
        Parameter IS_C_INVERTED bound to: 1'b0 
        Parameter IS_D_INVERTED bound to: 1'b0 
        Parameter IS_PRE_INVERTED bound to: 1'b0 
ERROR: [Synth 8-6156] failed synthesizing module 'FDPE' [/home/theo/Xilinx/Vivado/Vivado/2020.1/scripts/rt/data/unisim_comp.v:13664]
```

enjoy-digital, 2020-11-19:
Hi @flaminggoat,

would you mind sharing your platform/target file and .xci to allow me to try to reproduce the issue?

flaminggoat, 2020-11-27:
@enjoy-digital I have committed the files in my fork here: https://github.com/flaminggoat/litex-boards/commit/0faf8cfa077fc4a8d728bd489a805bf8b04b16b5 Thanks for looking into this.

developandplay, 2021-08-12:
@flaminggoat
What are the exact steps you took to generate the xci file?
Did you just add `ZYNQ2 Processing System` from the IP Catalog and then take the `processing_system7_0.xci` file?
I'm assuming you also customized the IP according to your board?
I'm trying to add the PYNQ Z2 board and since the softcores keep resetting I wanted to try out if the same issue applies to the built in arm core.

enjoy-digital, 2022-03-24:
The .xci can be directly generated with Vivado and then re-imported in LiteX. We've added support for various Zynq boards in the last 6 months (Pynq Z1, Zybo, Zeboard,  KV260, Alinx AX7010, etc...) so these can probably be reused as example. If there are still issues, please re-open.

closed 2022-03-24 by enjoy-digital