litex-hub/litex-boards #126: about add QMTech Wukong Board

SanadaShinken, 2020-11-24:
Hi, All

I had a FPGA board, the chip is Xilinx XC7A100T. The URL is https://github.com/ChinaQMTECH/QM_XC7A100T_WUKONG_BOARD.

after some study on how to add this board into litex-boards project, the current result is the attached files. but I got some errors. Does anyone could do me a favor to show me how to get rid of this errors.

move make.txt to make.py

move qmtech_wukong.platform.txt to litex_boards/platforms/qmtech_wukong.py
move qmtech_wukong.target.txt to litex_boards/targets/qmtech_wukong.py

The first problem is about clock input pin. in the schematic, the fpga's clock is in M22.
the output message's meaning is due to M22 is negative pin, and couldn't be used as clock input. 

the second problem is about DDR3 memory. the DDR3 memory chip is as the same as ARTY-A7.
as a good reference, and reassign the pin, the DDR3  memory setting is done. But during building phase, the error message shows up the reference voltage of the bank 16 to DDR3 pins is 1.35V. after some trying, by adding [litex_boards/platforms/qmtech_wukong.py] in targets setting file, the error seems gone. but I'm not sure this is the right way or not.

[make.txt](https://github.com/litex-hub/litex-boards/files/5589531/make.txt)
[qmtech_wukong.platform.txt](https://github.com/litex-hub/litex-boards/files/5589532/qmtech_wukong.platform.txt)
[qmtech_wukong.target.txt](https://github.com/litex-hub/litex-boards/files/5589534/qmtech_wukong.target.txt)

Many Thanks.

BR, Sanada

enjoy-digital, 2020-11-27:
Hi @SanadaShinken,

thanks for sharing the files. I'll have a look in the next days.

SanadaShinken, 2020-11-28:
Hi @enjoy-digital :

Thank you very much. BTW, I also create a board which come from alinx.
I will share to you. 

BR, Sanada

> Hi @SanadaShinken,
> 
> thanks for sharing the files. I'll have a look in the next days.

SanadaShinken, 2020-12-09:
Hi @enjoy-digital :

the below file is what I done until now. it could boot the fpga board, and run into bios. 
the qmtech_wukong.platform.txt is been put into platforms, and the qmtech_wukong.target.txt is been put into targets. the output terminal log is wukong.txt.
if you open the log file, you can see there is a problem about ddr3 memory initialization. 
it is failed.

Could you show me some hint to make the ddr3 memory initialization succeed?

If ddr3 memory initialization succeed, the next step is put linux image into flash, and
run into linux world.

[qmtech_wukong.platform.txt](https://github.com/litex-hub/litex-boards/files/5664280/qmtech_wukong.platform.txt)
[qmtech_wukong.target.txt](https://github.com/litex-hub/litex-boards/files/5664282/qmtech_wukong.target.txt)
[wukong.txt](https://github.com/litex-hub/litex-boards/files/5664283/wukong.txt)

BR, Sanada

enjoy-digital, 2020-12-17:
@SanadaShinken If you don't mind, I'm closing this one since a duplicate of https://github.com/litex-hub/litex-boards/issues/133, let's continue the discussion there.

closed 2020-12-17 by enjoy-digital

ozbenh, 2021-09-22:
There seem to be something wrong here:

```
    ("cpu_reset",  0, Pins("J8"),  IOStandard("LVCMOS33")),
```

J8 is the CMD pin of the SD-CARD... unless there are different versions of this board that is.

The examples for this board tend to use H7 (KEY1) as the reset button...

ozbenh, 2021-09-22:
Even more confusing, you have clk50 on M22, but on the docs I have it's M21

ozbenh, 2021-09-22:
Another one: you have the DRAM cs_n on E22 but the schematics I have show E22 it as unconnected and DDR_CS pulled down

ozbenh, 2021-09-22:
And another one... it uses GMIIMII for liteeth but the clock frequency default to 100Mhz which appears insufficient (it asserts if I try to use gen.py to generate a standalone core for this board)

SanadaShinken, 2021-09-22:
Hi, @ozbenh :

the following is the schematic which I use to make the targer / platform. 
https://github.com/ChinaQMTECH/QM_XC7A100T_WUKONG_BOARD

BTW, could you tell me the fpga chip type? currently the fpga chip is XC7A100T. 
As I know, qmtech seems make a newer one, which uses XC7A200T.

BR, Sanada

ozbenh, 2021-09-23:
Hi ! I talked to QMTech yesterday (they were very responsive via AliExpress "contact vendor" button :-)

So there's a V1 and a V2 of the board. V2 exists both in A100T and A200T model and has some fixed up pinout (though they don't currently sell the A200T one as the FPGA itself is currently too expensive).

The docs for V2 are at https://github.com/ChinaQMTECH/XC7A100T-200T_Wukong_Board.git

That said, I'm happy to provide patches (currently porting Microwatt standalone but I'll fix LiteX later). The main question is do we prefer having a separate qmterch_wukong_v2.py target (and platform) or should I just add some kind of --version argument to the existing one ?

ozbenh, 2021-09-23:
Are you confident your FPGA is a speed grade -2 ? According to QMTech they are -1 but the LiteX code hard codes -2

ozbenh, 2021-09-23:
Also can you double check this ? 

```
     ("jp3", " AF7  AE7  AD8  AC8  AF9  AE9 AD12 AC10",                                                                                         
            "AA11 AB11 AF11 AE11 AD14 AC14 AF13 AE13",                                                                                         
            "AD12 AC12"),                                                                                                                      
```
I think on the first line it should be AD10 AC10 and not AD12 AC10

SanadaShinken, 2021-09-23:
Hi, @ozbenh :

Thank you for the board version information. for board version's issue, I think you could ask the project owner and get the suggestion. I don't have any idea about how to do is the better way.

for FPGA speed grade, setting -2 is running very well. Maybe I could do some test with setting to -1.

for jp3, what you said is right.

Would you like to send a PR for the update?

Many Thanks.

BR, Sanada

ozbenh, 2021-09-23:
I will send a PR fixing this while adding v2. That said, for -2 vs -1, well if you have a -2 you are lucky it's faster :-) QMtech told me that they are supposed to be all -1's (at least the v2 versions). I'm going to add a parameter to choose the speed grade

ozbenh, 2021-09-24:
https://github.com/litex-hub/litex-boards/pull/273