litex-hub/litex-boards #133: adbout ARTIX7 XC7A100T based FPGA DDR3 memory initialized failed

SanadaShinken, 2020-12-16:
Hi, All

I had a FPGA board, the chip is Xilinx XC7A100T. The URL is https://github.com/ChinaQMTECH/QM_XC7A100T_WUKONG_BOARD. the board could be found in the following sites.
https://www.aliexpress.com/item/4000170042795.html
http://www.chinaqmtech.com/download_fpga

The below file is what I done until now. it could boot the fpga board, and run into bios.
the qmtech_wukong.platform.txt is been put into platforms, and the qmtech_wukong.target.txt is been put into targets. the output terminal log is wukong.txt.
if open the log file, there is a problem about ddr3 memory initialization. it is failed.

Could any one do me a favor to  show me some hint to make the ddr3 memory initialization succeed?

If ddr3 memory initialization succeed, the next step is put linux image into flash, and
run into linux world.

[qmtech_wukong.target.txt](https://github.com/litex-hub/linux-on-litex-vexriscv/files/5683069/qmtech_wukong.target.txt)
[qmtech_wukong.platform.txt](https://github.com/litex-hub/linux-on-litex-vexriscv/files/5683068/qmtech_wukong.platform.txt)
[wukong.txt](https://github.com/litex-hub/linux-on-litex-vexriscv/files/5683070/wukong.txt)

Thanks in advance

BR, Sanada

enjoy-digital, 2020-12-16:
Thanks @SanadaShinken for the initial support. I will look at it very soon. For now I transfered the issue to LiteX-Boards.

SanadaShinken, 2020-12-17:
> Thanks @SanadaShinken for the initial support. I will look at it very soon. For now I transfered the issue to LiteX-Boards.

Hi Sir:

Many Thanks LiteX Project, and You!!!!!

I'd made a workable version Linux + VexRiscV  running on qmtech wukong board.
But I still need your help to clearing some settings. I'll upload the setting file when I done some tests.

BR, Sanada

enjoy-digital, 2020-12-17:
Hi @SanadaShinken,

great you manage to make it work yourself, that's the better way to learn :) (I was going to look at it). Feel free to create a PR with support for this board if it's working. Even if not perfect, I could review and clean it up.

Chgavilana, 2020-12-17:
Hi:
You can try with my files over same board (I will load support maybe next year to official repository but more hands could be great). It's tested with Vexriscv linux, linux-smp up to 16 cores (first version without DMA and some additional configuration over image build), Ethernet (with some doubts about its max baudrate), spisdcard boot (with a pmod) and terminal over usb onboard. Now I'm working to activate HDMI in/out port and a 5Mpix cmos camera (litevideo is under study) but first I will test new repository version first.

Check it out and tell us your feedback
@enjoy-digital take a quick look at it. I hope this helps.

BR. Carlos.

[qmtech.txt](https://github.com/litex-hub/litex-boards/files/5712150/qmtech.txt)
[QMTECH_platform.txt](https://github.com/litex-hub/litex-boards/files/5712151/QMTECH_platform.txt)
[QMTECH_target.txt](https://github.com/litex-hub/litex-boards/files/5712152/QMTECH_target.txt)

SanadaShinken, 2020-12-19:
> Hi @SanadaShinken,
> 
> great you manage to make it work yourself, that's the better way to learn :) (I was going to look at it). Feel free to create a PR with support for this board if it's working. Even if not perfect, I could review and clean it up.

Hello @enjoy-digital : 

Thank you very much. the setting board setting files are in the work.zip file. this setting files could boot up with https://github.com/litex-hub/linux-on-litex-vexriscv/files/5281217/kc705_2020_09_25.zip, but will fail with 2020-1215 prebuilt bit-stream after I update litex related repositories. 

BR, Sanada

SanadaShinken, 2020-12-19:
> Hi:
> You can try with my files over same board (I will load support maybe next year to official repository but more hands could be great). It's tested with Vexriscv linux, linux-smp up to 16 cores [...]

Hi, @Chgavilana :

Very thanks for your sharing. I did some tests based on your spi sd card setting, but still failed. the linux kernel will crash after spi sd card initialized. Could you do me some favor to give me some hint about how to debug this error?

below is my environments:

1. Vivado : 2020.1
2. Litex : update to 2020-1218
3. Linux-on-litex-vexriscv : update to 2020-1218

BR, Sanada

Chgavilana, 2020-12-20:
Hi @SanadaShinken 
Well, for me works fine, and with last version, kernel recognize sd card and mount over /dev directory. I recommend first check your hardware connection (I don't know what pmod you use, mine is a little rustic but with all signal with pull-up and 3.3V power from connector), next, copy files again over SD (and check your sd memory), and finally, check your make.py file adding Qmtech board (short way to generate all). 

```
class Qmtech(Board):
#    SPIFLASH_PAGE_SIZE    = 256
#    SPIFLASH_SECTOR_SIZE  = 64*kB
#    SPIFLASH_DUMMY_CYCLES = 11
    def __init__(self):
        from litex_boards.targets import QMTECH
        Board.__init__(self, QMTECH.BaseSoC, soc_capabilities={
            # Communication
            "serial",
            "ethernet",
            # Storage
#            "spiflash",
#            "sdcard",
            "spisdcard",
            # GPIOs
            "leds",
            # Video
#            "framebuffer",
            # Monitoring
            #"xadc",
        }, bitstream_ext=".bit")
```
Send a screen of your error and I can check what other problem could be.
Carlos.

SanadaShinken, 2020-12-21:
Hello, @Chgavilana :

Thank you very much. the attachment file is what i use for wukong board.
[work.zip](https://github.com/litex-hub/litex-boards/files/5724691/work.zip)

the following message is the fail log after linux kernel boot up.

mmc_spi spi0.0: setup: unsupported mode bits 4
mmc_spi spi0.0: can't change chip-select polarity
mmc_spi spi0.0: SD/MMC host mmc0, no WP, no poweroff, cd polling
Unable to handle kernel NULL pointer dereference at virtual address 00000000
Oops [#1]
CPU: 0 PID: 1 Comm: swapper/0 Not tainted 5.0.0 #1
sepc: c00fa820 ra : c00fa7c8 sp : cf82dc40
 gp : c05a36e8 tp : cf828000 t0 : cfdf8148
 t1 : 00000002 t2 : 000000c0 s0 : cf82dc60
 s1 : cf801cc0 a0 : 00000000 a1 : 006000c0
 a2 : c05cce58 a3 : cfdf6890 a4 : ffffffff
 a5 : ffffffff a6 : 000007cb a7 : cf8a4700
 s2 : 006000c0 s3 : c00c7e64 s4 : 00000013
 s5 : 00000000 s6 : 00000000 s7 : cf8a4640
 s8 : 00000001 s9 : 000041ed s10: 00080000
 s11: cf8a4620 t3 : ffffffff t4 : fffffffe
 t5 : 00000400 t6 : cf8a4713
sstatus: 00000120 sbadaddr: 00000000 scause: 0000000d
---[ end trace e4985d40f850f455 ]---
Kernel panic - not syncing: Attempted to kill init! exitcode=0x0000000b
---[ end Kernel panic - not syncing: Attempted to kill init! exitcode=0x0000000b ]---

Many Thanks.

BR, Sanada

SanadaShinken, 2020-12-26:
Hi All:

Many Thanks for Greate Litex Project!!! @enjoy-digital , @Chgavilana, many thanks!!

After update all related projects, the final gateware could bring up linux OS successfully!

the following devices is ok for the attachment setting files.
[backup.zip](https://github.com/litex-hub/litex-boards/files/5743149/backup.zip)

1. DDR3
2. SPI SD Card
3. ethernet
4. serial

the boot log file is setting vexriscv CPU running at 100MHz and 150MHz.
[bootup_ok.txt](https://github.com/litex-hub/litex-boards/files/5743152/bootup_ok.txt)
[bootup_ok_150MHz.txt](https://github.com/litex-hub/litex-boards/files/5743153/bootup_ok_150MHz.txt)

If @enjoy-digital want to add to litex, it is very ok for my setting. this issues could be closed.

Best Regards,
Akio

SanadaShinken, 2020-12-26:
Hi All:

Many Thanks for Greate Litex Project!!! @enjoy-digital , @Chgavilana, many thanks!!

After update all related projects, the final gateware could bring up linux OS successfully!

the following devices is ok for the attachment setting files.
[backup.zip](https://github.com/litex-hub/litex-boards/files/5743156/backup.zip)

    DDR3
    SPI SD Card
    ethernet
    serial

the boot log file is setting vexriscv CPU running at 100MHz and 150MHz.
[bootup_ok.txt](https://github.com/litex-hub/litex-boards/files/5743157/bootup_ok.txt)
[bootup_ok_150MHz.txt](https://github.com/litex-hub/litex-boards/files/5743158/bootup_ok_150MHz.txt)

If @enjoy-digital want to add to litex, it is very ok to me. this issues could be closed.

Best Regards,
BR, Sanada

enjoy-digital, 2020-12-28:
Hi @SanadaShinken,

thanks for the feedback. I can add the board support to LiteX-Boards ye, if you don't want to create a PR yourself can you provide me a name and email I can use for the copyright?

SanadaShinken, 2020-12-29:
> Hi @SanadaShinken,
> 
> thanks for the feedback. I can add the board support to LiteX-Boards ye, if you don't want to create a PR yourself can you provide me a name and email I can use for the copyright?

@enjoy-digital :

name : Shinken Sanada
email : sanadashinken@gmail.com

Many Thanks. 

BR, Sanada

enjoy-digital, 2020-12-29:
@SanadaShinken: This has been integrated. I also did some review/cleanup on the Platform file in https://github.com/litex-hub/litex-boards/commit/b67b18caad1d3f352b30de0aaf2b28b6df54dd71.

closed 2020-12-29 by enjoy-digital