litex-hub/litex-boards #135: Fomu: usb.out_ev_pending.status: 'CSR' object has no attribute 'status'

tcal-x, 2020-12-21:
litex-boards: 36b7fb1033390c78b7551076ecf88c0dc040aafd
valentyusb: (hw_cdc_eptri) 371526e432a858b768ddc561cee0d7a0c8f40f42 

Hello,
I encountered the following error.  Apologies if I made a mistake with setup:

```
$ ./fomu.py --build
Cloning into 'valentyusb'...
remote: Enumerating objects: 16, done.
remote: Counting objects: 100% (16/16), done.
remote: Compressing objects: 100% (12/12), done.
remote: Total 3056 (delta 5), reused 11 (delta 4), pack-reused 3040
Receiving objects: 100% (3056/3056), 743.77 KiB | 1.08 MiB/s, done.
Resolving deltas: 100% (1980/1980), done.
INFO:SoC:        __   _ __      _  __  
INFO:SoC:       / /  (_) /____ | |/_/  
INFO:SoC:      / /__/ / __/ -_)>  <    
INFO:SoC:     /____/_/\__/\__/_/|_|  
INFO:SoC:  Build your hardware, easily!
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Creating SoC... (2020-12-20 23:45:50)
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:FPGA device : ice40-up5k-uwg30.
INFO:SoC:System clock: 12.00MHz.
INFO:SoCBusHandler:Creating Bus Handler...
INFO:SoCBusHandler:32-bit wishbone Bus, 4.0GiB Address Space.
INFO:SoCBusHandler:Adding reserved Bus Regions...
INFO:SoCBusHandler:Bus Handler created.
INFO:SoCCSRHandler:Creating CSR Handler...
INFO:SoCCSRHandler:8-bit CSR Bus, 32-bit Aligned, 16.0KiB Address Space, 2048B Paging, big Ordering (Up to 32 Locations).
INFO:SoCCSRHandler:Adding reserved CSRs...
INFO:SoCCSRHandler:CSR Handler created.
INFO:SoCIRQHandler:Creating IRQ Handler...
INFO:SoCIRQHandler:IRQ Handler (up to 32 Locations).
INFO:SoCIRQHandler:Adding reserved IRQs...
INFO:SoCIRQHandler:IRQ Handler created.
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Initial SoC:
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:32-bit wishbone Bus, 4.0GiB Address Space.
INFO:SoC:8-bit CSR Bus, 32-bit Aligned, 16.0KiB Address Space, 2048B Paging, big Ordering (Up to 32 Locations).
INFO:SoC:IRQ Handler (up to 32 Locations).
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoCCSRHandler:ctrl CSR allocated at Location 0.
INFO:SoCBusHandler:io0 Region added at Origin: 0x80000000, Size: 0x80000000, Mode: RW, Cached: False Linker: False.
INFO:SoCBusHandler:cpu_bus0 added as Bus Master.
INFO:SoCBusHandler:cpu_bus1 added as Bus Master.
INFO:SoCCSRHandler:cpu CSR allocated at Location 1.
INFO:SoCCSRHandler:identifier_mem CSR allocated at Location 2.
Traceback (most recent call last):
  File "./fomu.py", line 169, in <module>
    main()
  File "./fomu.py", line 160, in main
    **soc_core_argdict(args)
  File "./fomu.py", line 93, in __init__
    **kwargs)
  File "/home/tim/timvideos/litex-buildenv/third_party/litex/litex/soc/integration/soc_core.py", line 181, in __init__
    self.add_uart(name=uart_name, baudrate=uart_baudrate, fifo_depth=uart_fifo_depth)
  File "/home/tim/timvideos/litex-buildenv/third_party/litex/litex/soc/integration/soc.py", line 1108, in add_uart
    self.submodules.uart = cdc_eptri.CDCUsb(usb_iobuf)
  File "valentyusb/valentyusb/usbcore/cpu/cdc_eptri.py", line 102, in __init__
    self.submodules.phy = phy = ClockDomainsRenamer("usb_12")(CDCUsbPHY(iobuf, debug=debug, vid=vid, pid=pid, product=product, manufacturer=manufacturer))
  File "valentyusb/valentyusb/usbcore/cpu/cdc_eptri.py", line 461, in __init__
    If(usb.out_ev_pending.status,
AttributeError: 'CSR' object has no attribute 'status'
```

enjoy-digital, 2020-12-21:
Hi @tcal-x,

the integration of ValentyUSB is a bit hacky for now. Changes have been done to ValentyUSB to fix this and we are using a different repository. You can just do `rm -rf valentyusb` in the same directory you are runng the target, is should solves the issue.

tcal-x, 2020-12-21:
Thank you @enjoy-digital , 

Is the goal of removing the valentyusb directory to disable UART, or to refresh the clone of valentyusb?

In any case, I see the same issue -- after removing valentyusb/ and rerunning, the ./fomu.py script performs the clone again, and the same error results.   

If time allows, I will try to debug the issue further.

```
$ cd valentyusb/
$ git remote -v
origin  https://github.com/litex-hub/valentyusb (fetch)
origin  https://github.com/litex-hub/valentyusb (push)
$ git branch
* hw_cdc_eptri
```

enjoy-digital, 2020-12-22:
@tcal-x: the goal was to do a clean clone of ValentyUSB, but I think the issue is that LiteX is not up to date on your setup so ValentyUSB is not building correctly.

tcal-x, 2020-12-23:
Thank you @enjoy-digital ;  I had been trying to update my LiteX using `litex_setup.py update`, but that didn't fix the problem.   

When I did a completely new install of everything, starting with an empty directory, wgetting `litex_setup.py`, and then running `litex_setup.py init install` in a virtual environment, then it worked. 

Well, the LiteX side of it worked.   I encountered the following.   I ran using tools from fomu-toolchain v1.5.6, from March; maybe I need newer versions of Yosys and nextpnr-ice40.
```
Info: Device utilisation:
Info:            ICESTORM_LC:  6436/ 5280   121%
Info:           ICESTORM_RAM:    28/   30    93%
Info:                  SB_IO:    11/   96    11%
Info:                  SB_GB:     8/    8   100%
Info:           ICESTORM_PLL:     1/    1   100%
Info:            SB_WARMBOOT:     0/    1     0%
Info:           ICESTORM_DSP:     4/    8    50%
Info:         ICESTORM_HFOSC:     0/    1     0%
Info:         ICESTORM_LFOSC:     0/    1     0%
Info:                 SB_I2C:     0/    2     0%
Info:                 SB_SPI:     0/    2     0%
Info:                 IO_I3C:     0/    2     0%
Info:            SB_LEDDA_IP:     0/    1     0%
Info:            SB_RGBA_DRV:     0/    1     0%
Info:         ICESTORM_SPRAM:     4/    4   100%

Info: Placed 14 cells based on constraints.
ERROR: Unable to place cell 'main_basesoc_timer_value_status_SB_DFFESR_Q_27_DFFLC', no Bels remaining of type 'ICESTORM_LC'
```

closed 2020-12-23 by tcal-x

tcal-x, 2020-12-23:
In case anyone finds this, my problem above is because I was not specifying the correct VexRiscv variant.   From @enjoy-digital 's [tweet ](https://twitter.com/enjoy_digital/status/1313788215409684481) in October, these are the known good recipes:
```
./fomu.py --cpu-type=serv --build --flash
./fomu.py --cpu-type=vexriscv --cpu-variant=minimal --build --flash
```
You should then see this from dmesg:
```
[290570.358648] usb 1-2.4: USB disconnect, device number 31
[290570.440320] usb 1-2.4: new full-speed USB device number 32 using xhci_hcd
[290570.541616] usb 1-2.4: New USB device found, idVendor=1209, idProduct=5bf2, bcdDevice= 1.01
[290570.541622] usb 1-2.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[290570.541626] usb 1-2.4: Product: OrangeCrab CDC
[290570.541629] usb 1-2.4: Manufacturer: GsD
[290570.550521] cdc_acm 1-2.4:1.0: ttyACM0: USB ACM device
```
...and now you can connect via litex\_term, picocom, etc.