litex-hub/litex-boards #137: tinyfpga_bx: vexriscv build error

devobliquezer0, 2020-12-24:
Build: `./litex-boards/litex_boards/targets/tinyfpga_bx.py --build --cpu-type=vexriscv`

Error: `ERROR: Unable to place cell 'mem.2.0.0_RAM', no Bels remaining of type 'ICESTORM_RAM'`

Full output: https://paste.ubuntu.com/p/wY9rH3bDdT/

serv core builds fine

enjoy-digital, 2020-12-24:
As reported by the tools, the tinyfpga_bx is indeed too small to use the `standard` variant of VexRiscv. You can however use the `minimal` variant:

`./tinyfpga_bx.py --cpu-type=vexriscv --cpu-variant=minimal --build`

devobliquezer0, 2020-12-24:
got it. it works now.

This brings up an idea which might be a lot of work to setup, but what if litex had a fpga budgeting system where it could calculate the fpga requirements for the final soc config and fails if it starts to go over the available elements.

enjoy-digital, 2020-12-24:
Thanks for the feedback. You idea is interesting but it would be indeed very difficult  to evaluate these numbers during the design elaboration (and when using external sources it would even be impossible).

closed 2020-12-24 by enjoy-digital