litex-hub/litex-boards #139: ecpix5 target sys clk frequency seems wrong 75MHz instead of 100MHz

bvernoux, 2020-12-26:
It seems https://github.com/litex-hub/litex-boards/blob/master/litex_boards/targets/ecpix5.py sys_clk_freq is wrong it shall be set to 100MHz instead of 75MHz thanks to clarify and fix that

enjoy-digital, 2020-12-28:
Hi @bvernoux,

the 100MHz clock is indeed the primary clock of the board that is  converted to the internal `sys_clk` through a PLL.  The default `sys_clk` of 75MHz is the one that has been tested with the default configuration (VexRiscv standard CPU, LiteDRAM controller) and that is both functional and passing timings. That's also the `sys_clk` frequency we have been using on other ECP5 dev boards. 100MHz also seems to be working on hardware (I just tested:` ./ecpix5.py --sys-clk-freq=100e6 --build --load`) but timing are not fully satisfied:
 
```
Warning: Max frequency for clock              '$glbnet$sys_clk': 99.25 MHz (FAIL at 100.00 MHz)
Info: Max frequency for clock          '$glbnet$crg_clkout1': 165.23 MHz (PASS at 25.00 MHz)
Info: Max frequency for clock '$glbnet$clk100$TRELLIS_IO_IN': 388.05 MHz (PASS at 100.00 MHz)
```

So we could probably increase the frequency a bit but 75MHz seems to be a good compromise for now and gives some timing margin when adding other peripherals.

enjoy-digital, 2021-01-07:
I think the question has been answered, closing.

closed 2021-01-07 by enjoy-digital