litex-hub/litex-boards #143: vcu118 target with vexriscv_smp causes vexriscv to fail

SolitaryThinker, 2021-01-04:
Not sure if this is the right repo for this issue. Please let me know if I should repost this issue to another repo.

When running this command `./vcu118.py --build --cpu-type vexriscv_smp  --cpu-variant linux`
Litex will attempt to generate the following file: `VexRiscvLitexSmpCluster_Cc1_Iw32Is4096Iy1_Dw32Ds4096Dy1_Ldw512`. 

Vexrisvc uses the following command from the dev branch: 
`vexriscv.demo.smp.VexRiscvLitexSmpClusterCmdGen --cpu-count=1 --ibus-width=32 --dbus-width=32 --dcache-size=4096 --icache-size=4096 --dcache-ways=1 --icache-ways=1 --litedram-width=512 --aes-instruction=False --netlist-name=VexRiscvLitexSmpCluster_Cc1_Iw32Is4096Iy1_Dw32Ds4096Dy1_Ldw512 --netlist-directory=/home/will/research/softcore/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog`

However, VexRiscv fails with the following error during this process:

> [info] Step 14
> [info] Build dBusCoherent
> [info] Build dBusCoherent_bmb_slaveModel_invalidationGen
> [info] Build 
> [info] Step 15
> [info] Build cores_0_cpu_dBus_connector_invalidationPlanner
> [info] Build dBusCoherent_bmb_slaveModel_arbiterGen
> [info] Build cores_0_cpu_dBus_masterModel_invalidationRequirementsGen2
> [info] Step 16
> [info] [Progress] at 1.145 : Checks and transforms
> [error] Exception in thread "main" spinal.core.SpinalExit: 
> [error]  
> [error] ********************************************************************************
> [error] ********************************************************************************
> [error] WIDTH MISMATCH on (toplevel/iBridge_logic/io_input_upSizer_io_output_unburstify/buffer_beat :  UInt[0 bits]) := (UInt - UInt)[1 bits] at 
> [error]     spinal.lib.bus.bmb.BmbUnburstify$$anon$2$$anonfun$1.apply$mcV$sp(BmbUnburstify.scala:40)
> [error]     spinal.lib.bus.bmb.BmbUnburstify$$anon$2.<init>(BmbUnburstify.scala:39)
> [error]     spinal.lib.bus.bmb.BmbUnburstify.<init>(BmbUnburstify.scala:28)
> [error]     spinal.lib.bus.bmb.Bmb.unburstify(Bmb.scala:552)
> [error]     vexriscv.demo.smp.BmbToLiteDram.<init>(Misc.scala:166)
> [error]     vexriscv.demo.smp.BmbToLiteDramGenerator$$anonfun$12.apply(Misc.scala:264)
> [error]     vexriscv.demo.smp.BmbToLiteDramGenerator$$anonfun$12.apply(Misc.scala:264)
> [error]     spinal.lib.generator.Generator$Task.build(Generator.scala:216)
> [error]     spinal.lib.generator.Generator$$anonfun$generateIt$1$$anonfun$apply$mcV$sp$1.apply(Generator.scala:276)
> [error]     spinal.lib.generator.Generator$$anonfun$generateIt$1$$anonfun$apply$mcV$sp$1.apply(Generator.scala:275)
> [error]     spinal.lib.generator.Generator$$anonfun$generateIt$1.apply$mcV$sp(Generator.scala:275)
> [error]     spinal.lib.generator.Generator$$anonfun$generateIt$1.apply(Generator.scala:275)
> [error]     spinal.lib.generator.Generator$$anonfun$generateIt$1.apply(Generator.scala:275)
> [error]     spinal.lib.generator.Generator.apply(Generator.scala:240)
> [error]     spinal.lib.generator.Generator.generateIt(Generator.scala:274)
> [error]     spinal.lib.generator.GeneratorCompiler$$anonfun$build$3.apply(Generator.scala:354)
> [error]     spinal.lib.generator.GeneratorCompiler$$anonfun$build$3.apply(Generator.scala:352)
> [error]     spinal.lib.generator.GeneratorCompiler.build(Generator.scala:352)
> [error]     spinal.lib.generator.GeneratorComponent.<init>(Generator.scala:390)
> [error]     spinal.lib.generator.Generator.toComponent(Generator.scala:297)
> [error]     vexriscv.demo.smp.VexRiscvLitexSmpClusterCmdGen$.dutGen(VexRiscvSmpLitexCluster.scala:117)
> [error]     vexriscv.demo.smp.VexRiscvLitexSmpClusterCmdGen$$anonfun$18.apply(VexRiscvSmpLitexCluster.scala:122)
> [error]     vexriscv.demo.smp.VexRiscvLitexSmpClusterCmdGen$$anonfun$18.apply(VexRiscvSmpLitexCluster.scala:122)
> [error]     vexriscv.demo.smp.VexRiscvLitexSmpClusterCmdGen$.delayedEndpoint$vexriscv$demo$smp$VexRiscvLitexSmpClusterCmdGen$1(VexRiscvSmpLitexCluster.scala:122)
> [error]     vexriscv.demo.smp.VexRiscvLitexSmpClusterCmdGen$delayedInit$body.apply(VexRiscvSmpLitexCluster.scala:59)
> [error]     vexriscv.demo.smp.VexRiscvLitexSmpClusterCmdGen$.main(VexRiscvSmpLitexCluster.scala:59)
> [error]     vexriscv.demo.smp.VexRiscvLitexSmpClusterCmdGen.main(VexRiscvSmpLitexCluster.scala)
> [error] ********************************************************************************
> [error] ********************************************************************************
> [error] WIDTH MISMATCH on (toplevel/dBridge_logic/io_input_upSizer_io_output_unburstify/buffer_beat :  UInt[0 bits]) := (UInt - UInt)[1 bits] at 
> [error]     spinal.lib.bus.bmb.BmbUnburstify$$anon$2$$anonfun$1.apply$mcV$sp(BmbUnburstify.scala:40)
> [error]     spinal.lib.bus.bmb.BmbUnburstify$$anon$2.<init>(BmbUnburstify.scala:39)
> [error]     spinal.lib.bus.bmb.BmbUnburstify.<init>(BmbUnburstify.scala:28)
> [error]     spinal.lib.bus.bmb.Bmb.unburstify(Bmb.scala:552)
> [error]     vexriscv.demo.smp.BmbToLiteDram.<init>(Misc.scala:166)
> [error]     vexriscv.demo.smp.BmbToLiteDramGenerator$$anonfun$12.apply(Misc.scala:264)
> [error]     vexriscv.demo.smp.BmbToLiteDramGenerator$$anonfun$12.apply(Misc.scala:264)
> [error]     spinal.lib.generator.Generator$Task.build(Generator.scala:216)
> [error]     spinal.lib.generator.Generator$$anonfun$generateIt$1$$anonfun$apply$mcV$sp$1.apply(Generator.scala:276)
> [error]     spinal.lib.generator.Generator$$anonfun$generateIt$1$$anonfun$apply$mcV$sp$1.apply(Generator.scala:275)
> [error]     spinal.lib.generator.Generator$$anonfun$generateIt$1.apply$mcV$sp(Generator.scala:275)
> [error]     spinal.lib.generator.Generator$$anonfun$generateIt$1.apply(Generator.scala:275)
> [error]     spinal.lib.generator.Generator$$anonfun$generateIt$1.apply(Generator.scala:275)
> [error]     spinal.lib.generator.Generator.apply(Generator.scala:240)
> [error]     spinal.lib.generator.Generator.generateIt(Generator.scala:274)
[... 21 lines cut ...]
> [error] 	at spinal.core.internals.SpinalVerilogBoot$$anonfun$singleShot$2$$anonfun$apply$116.apply(Phase.scala:2429)
> [error] 	at scala.collection.mutable.ResizableArray$class.foreach(ResizableArray.scala:59)
> [error] 	at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:48)
> [error] 	at spinal.core.internals.SpinalVerilogBoot$$anonfun$singleShot$2.apply(Phase.scala:2429)
> [error] 	at spinal.core.internals.SpinalVerilogBoot$$anonfun$singleShot$2.apply(Phase.scala:2368)
> [error] 	at spinal.core.ScopeProperty$.sandbox(ScopeProperty.scala:31)
> [error] 	at spinal.core.internals.SpinalVerilogBoot$.singleShot(Phase.scala:2368)
> [error] 	at spinal.core.internals.SpinalVerilogBoot$.apply(Phase.scala:2363)
> [error] 	at spinal.core.Spinal$.apply(Spinal.scala:353)
> [error] 	at spinal.core.SpinalConfig.generateVerilog(Spinal.scala:160)
> [error] 	at vexriscv.demo.smp.VexRiscvLitexSmpClusterCmdGen$.delayedEndpoint$vexriscv$demo$smp$VexRiscvLitexSmpClusterCmdGen$1(VexRiscvSmpLitexCluster.scala:122)
> [error] 	at vexriscv.demo.smp.VexRiscvLitexSmpClusterCmdGen$delayedInit$body.apply(VexRiscvSmpLitexCluster.scala:59)
> [error] 	at scala.Function0$class.apply$mcV$sp(Function0.scala:34)
> [error] 	at scala.runtime.AbstractFunction0.apply$mcV$sp(AbstractFunction0.scala:12)
> [error] 	at scala.App$$anonfun$main$1.apply(App.scala:76)
> [error] 	at scala.App$$anonfun$main$1.apply(App.scala:76)
> [error] 	at scala.collection.immutable.List.foreach(List.scala:392)
> [error] 	at scala.collection.generic.TraversableForwarder$class.foreach(TraversableForwarder.scala:35)
> [error] 	at scala.App$class.main(App.scala:76)
> [error] 	at vexriscv.demo.smp.VexRiscvLitexSmpClusterCmdGen$.main(VexRiscvSmpLitexCluster.scala:59)
> [error] 	at vexriscv.demo.smp.VexRiscvLitexSmpClusterCmdGen.main(VexRiscvSmpLitexCluster.scala)
> [error] Nonzero exit code returned from runner: 1
> [error] (Compile / runMain) Nonzero exit code returned from runner: 1
> [error] Total time: 3 s, completed Jan 4, 2021, 11:14:48 AM
>

enjoy-digital, 2021-01-07:
Thanks @SolitaryThinker, the elaboration of this VexRiscv-SMP for this configuration indeed seems to fail. I'll do more tests and report it.

enjoy-digital, 2021-01-08:
@SolitaryThinker: As a temporary workaround, you could reduce the DDR3 data with with `PHYPadsReducer `(https://github.com/enjoy-digital/litex_vexriscv_smp/blob/master/genesys2.py#L81 for example).

enjoy-digital, 2021-01-08:
@Dolu1990: do you eventually have an idea why large `litedram-width` causes this (IIRC it starts at 512 since 256 is working fine). Should I report if to VexRiscv repo or do you prefer I try to handle data-width convertion directly in LiteDRAM for large `litedram-width` cases?

Dolu1990, 2021-01-08:
Hi

Yes sure.
--litedram-width=512
=> 512/8 = 64 =  byte per cache line = length max on the bmb bus =>  no need to unburstify accesses => beat counter of zero bits => beat counter + 1 assigned into beat counter (of zero bits) =>  width missmatch.

Should be easy to fix. i'm looking at it

Dolu1990, 2021-01-08:
I pushed a fix, normaly if you pull https://github.com/litex-hub/pythondata-cpu-vexriscv-smp.git and its submodules (in ext folders) that should be fix. let' me know if not the case ^^

enjoy-digital, 2021-01-09:
Thanks @Dolu1990, this indeed fixes it. The VexRiscv-SMP cluster is now generated correctly with the `litedram-width` of 512.

Can you give it a try @SolitaryThinker?, in `pythondata-cpu-vexriscv_smp` just do: 
`git pull`
`git submodule update`
 and try to rebuild the design.

enjoy-digital, 2021-01-25:
Closing since resolved.

closed 2021-01-25 by enjoy-digital