litex-hub/litex-boards #150: Arty: bios with ethernet support does not fit into ROM

kgugala, 2021-01-14:
running:
```
python3 litex-boards/litex_boards/targets/arty.py --with-ethernet
```

fails with:
```
riscv64-unknown-elf-ld: bios.elf section `.rodata' will not fit in region `rom'
riscv64-unknown-elf-ld: region `rom' overflowed by 1608 bytes
```
@enjoy-digital did the default ROM size shrink recently, or the binary got bigger?

enjoy-digital, 2021-01-14:
@kgugala: that's indeed possible the ROM size increased recently.  I want to have a closer at this while also looking at that as part of https://github.com/enjoy-digital/litex/issues/698. For now can you use `--integrated-rom-size=xxyy` to increase the ROM size manually?

kgugala, 2021-01-14:
@enjoy-digital I did change the ROM size in my use case. Just reported the issue so we can fix it :)

enjoy-digital, 2021-01-14:
@kgugala: ah ok thanks :) So yes we need to make sure the ROM size is optimal and that enabling optional features does not cause this kind of errors since the ROM size will vary depending the features enabled (with https://github.com/enjoy-digital/litex/issues/698). The predefined size was a good compromise in the past, but with the additional features added to the BIOS (Ethernet, SDCard, etc...) it's no longer optimal/convenient.

enjoy-digital, 2021-03-26:
https://github.com/enjoy-digital/litex/issues/698 has been implemented, we can now close this.

closed 2021-03-26 by enjoy-digital