litex-hub/litex-boards #156: Alveo U280 DDR4 problem

smosanu, 2021-01-27:
Hello all,

I am working on adding support for the Alveo U280 board (in my case it is ES1) and am using the Alveo U250 _platform_ and _target_ as reference. I was able to generate a minimal SoC and bitstream and boot LiteX, however memory initialization fails. Included below is the boot log including a failing memory test. The board files are at my [fork under alveo_u280 branch](https://github.com/hplp/litex-boards/tree/alveo_u280).

The Alveo U280 is similar to the U250, with a few differences. It has 2 (not 4) DDR4 channels and HBM2. The main clock sources are of 100MHz (not 300MHz). The same MTA18ASF2G72PZ DDR4 module is used. I verified the configuration settings of two identical Vivado MIG+MicroBlaze default designs for the two boards and only the CLK and pins are different.

Any idea what I am doing wrong? Any suggestions are greatly appreciated. Thank you!

```
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2020 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Jan 26 2021 21:29:42
 BIOS CRC passed (125b79b6)

 Migen git sha1: --------
 LiteX git sha1: 737ed9d6

--=============== SoC ==================--
CPU:            VexRiscv SMP-LINUX @ 125MHz
BUS:            WISHBONE 32-bit @ 4GiB
CSR:            32-bit data
ROM:            64KiB
SRAM:           8KiB
L2:             0KiB
SDRAM:          1048576KiB 64-bit @ 1000MT/s (CL-9 CWL-9)

--========== Initialization ============--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Write leveling:
  Cmd/Clk scan (0-324)
  |000000  |0000  |0000  |0000| best: -1
  Setting Cmd/Clk delay to -1 taps.
  Data scan:
  m0: |1111111111111111111111| delay: -
  m1: |1111111111111111111111| delay: -
  m2: |1111111111111111111111| delay: -
  m3: |1111111111111111111111| delay: -
  m4: |1111111111111111111111| delay: -
  m5: |1111111111111111111111| delay: -
  m6: |1111111111111111111111| delay: -
  m7: |1111111111111111111111| delay: -
Write latency calibration:
m0:0 m1:0 m2:0 m3:0 m4:0 m5:0 m6:0 m7:0
Read leveling:
  m0, b0: |00000000000000000000000000000000| delays: -
  m0, b1: |00000000000000000000000000000000| delays: -
  m0, b2: |00000000000000000000000000000000| delays: -
  m0, b3: |00000000000000000000000000000000| delays: -
  m0, b4: |00000000000000000000000000000000| delays: -
  m0, b5: |00000000000000000000000000000000| delays: -
  m0, b6: |00000000000000000000000000000000| delays: -
  m0, b7: |00000000000000000000000000000000| delays: -
  best: m0, b00 delays: -
  m1, b0: |00000000000000000000000000000000| delays: -
  m1, b1: |00000000000000000000000000000000| delays: -
  m1, b2: |00000000000000000000000000000000| delays: -
  m1, b3: |00000000000000000000000000000000| delays: -
  m1, b4: |00000000000000000000000000000000| delays: -
  m1, b5: |00000000000000000000000000000000| delays: -
  m1, b6: |00000000000000000000000000000000| delays: -
  m1, b7: |00000000000000000000000000000000| delays: -
  best: m1, b00 delays: -
  m2, b0: |00000000000000000000000000000000| delays: -
  m2, b1: |00000000000000000000000000000000| delays: -
  m2, b2: |00000000000000000000000000000000| delays: -
  m2, b3: |00000000000000000000000000000000| delays: -
  m2, b4: |00000000000000000000000000000000| delays: -
  m2, b5: |00000000000000000000000000000000| delays: -
  m2, b6: |00000000000000000000000000000000| delays: -
  m2, b7: |00000000000000000000000000000000| delays: -
  best: m2, b00 delays: -
  m3, b0: |00000000000000000000000000000000| delays: -
  m3, b1: |00000000000000000000000000000000| delays: -
  m3, b2: |00000000000000000000000000000000| delays: -
  m3, b3: |00000000000000000000000000000000| delays: -
  m3, b4: |00000000000000000000000000000000| delays: -
  m3, b5: |00000000000000000000000000000000| delays: -
  m3, b6: |00000000000000000000000000000000| delays: -
  m3, b7: |00000000000000000000000000000000| delays: -
  best: m3, b00 delays: -
  m4, b0: |00000000000000000000000000000000| delays: -
  m4, b1: |00000000000000000000000000000000| delays: -
  m4, b2: |00000000000000000000000000000000| delays: -
  m4, b3: |00000000000000000000000000000000| delays: -
  m4, b4: |00000000000000000000000000000000| delays: -
  m4, b5: |00000000000000000000000000000000| delays: -
  m4, b6: |00000000000000000000000000000000| delays: -
  m4, b7: |00000000000000000000000000000000| delays: -
  best: m4, b00 delays: -
  m5, b0: |00000000000000000000000000000000| delays: -
  m5, b1: |00000000000000000000000000000000| delays: -
  m5, b2: |00000000000000000000000000000000| delays: -
  m5, b3: |00000000000000000000000000000000| delays: -
  m5, b4: |00000000000000000000000000000000| delays: -
  m5, b5: |00000000000000000000000000000000| delays: -
  m5, b6: |00000000000000000000000000000000| delays: -
  m5, b7: |00000000000000000000000000000000| delays: -
  best: m5, b00 delays: -
  m6, b0: |00000000000000000000000000000000| delays: -
  m6, b1: |00000000000000000000000000000000| delays: -
  m6, b2: |00000000000000000000000000000000| delays: -
  m6, b3: |00000000000000000000000000000000| delays: -
  m6, b4: |00000000000000000000000000000000| delays: -
  m6, b5: |00000000000000000000000000000000| delays: -
  m6, b6: |00000000000000000000000000000000| delays: -
  m6, b7: |00000000000000000000000000000000| delays: -
  best: m6, b00 delays: -
  m7, b0: |00000000000000000000000000000000| delays: -
  m7, b1: |00000000000000000000000000000000| delays: -
  m7, b2: |00000000000000000000000000000000| delays: -
  m7, b3: |00000000000000000000000000000000| delays: -
  m7, b4: |00000000000000000000000000000000| delays: -
  m7, b5: |00000000000000000000000000000000| delays: -
  m7, b6: |00000000000000000000000000000000| delays: -
  m7, b7: |00000000000000000000000000000000| delays: -
  best: m7, b00 delays: -
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2MiB)...
  Write: 0x40000000-0x40200000 2MiB     
   Read: 0x40000000-0x40200000 2MiB     
  bus errors:  256/256
  addr errors: 8192/8192
  data errors: 524288/524288
Memtest KO
Memory initialization failed

--============= Console ================--

litex> mem_list

Available memory regions:
PLIC      0xf0c00000 0x400000 
CLINT     0xf0010000 0x10000 
ROM       0x00000000 0x10000 
SRAM      0x10000000 0x2000 
MAIN_RAM  0x40000000 0x40000000 
FIRMWARE_RAM  0x20000000 0x8000 
OPENSBI   0x40f00000 0x80000 
CSR       0xf0000000 0x10000 

litex> mem_test 0x40000000 1024

Memtest at 0x40000000 (1KiB)...
  Write: 0x40000000-0x40000400 1KiB   
   Read: 0x40000000-0x40000400 1KiB   
  bus errors:  256/256
  addr errors: 256/256
  data errors: 256/256
Memtest KO

```

enjoy-digital, 2021-01-27:
Hi @smosanu,

thanks for the log. This behavior generally means there are pinout errors, constraint errors or missing INTERNAL_VREF constraints. Setting INTERNAL_VREF to 0.84 on the banks used by the LiteDRAM controller is mandatory to get the DDR4 working and I don't see the constraints in your platform file. For an example you can have a look at: https://github.com/litex-hub/litex-boards/blob/master/litex_boards/platforms/alveo_u250.py#L351-L366

Once added, could you provide an updated log if still failing?

smosanu, 2021-01-29:
Thank you, that's a good point. Excluding those constraints was a silly mistake. I now added them as listed below and there is a noticeable improvement, but it is still failing. Please see the new log and let me know what you think! Thank you!

```
# DDR4 memory channel C1 Internal Vref
self.add_platform_command("set_property INTERNAL_VREF 0.84 [get_iobanks 64]")
self.add_platform_command("set_property INTERNAL_VREF 0.84 [get_iobanks 65]")
self.add_platform_command("set_property INTERNAL_VREF 0.84 [get_iobanks 66]")
```
```
litex> reboot

        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2020 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Jan 27 2021 15:07:31
 BIOS CRC passed (a6e1ae18)

 Migen git sha1: --------
 LiteX git sha1: 737ed9d6

--=============== SoC ==================--
CPU:            VexRiscv SMP-LINUX @ 125MHz
BUS:            WISHBONE 32-bit @ 4GiB
CSR:            32-bit data
ROM:            64KiB
SRAM:           8KiB
L2:             0KiB
SDRAM:          1048576KiB 64-bit @ 1000MT/s (CL-9 CWL-9)

--========== Initialization ============--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Write leveling:
  Cmd/Clk scan (0-324)
  |000000  |0000  |0000  |0000| best: -1
  Setting Cmd/Clk delay to -1 taps.
  Data scan:
  m0: |1111111111000000000000| delay: -
  m1: |1111111110000000000000| delay: -
  m2: |1111100000000000000001| delay: -
  m3: |1111000000000000000011| delay: -
  m4: |0000000000000001111111| delay: 238
  m5: |1110000000000000000011| delay: 305
  m6: |1111110000000000000000| delay: -
  m7: |1111111100000000000000| delay: -
Write latency calibration:
m0:0 m1:0 m2:0 m3:0 m4:0 m5:6 m6:0 m7:0
Read leveling:
  m0, b0: |00000000000000000000000000000000| delays: -
  m0, b1: |00000000000000000000000000000000| delays: -
  m0, b2: |00000000000000000000000000000000| delays: -
  m0, b3: |00000000000000000000000000000000| delays: -
  m0, b4: |00000000000000000000000000000000| delays: -
  m0, b5: |00000000000000000000000000000000| delays: -
  m0, b6: |00000000000000000000000000000000| delays: -
  m0, b7: |00000000000000000000000000000000| delays: -
  best: m0, b00 delays: -
  m1, b0: |00000000000000000000000000000000| delays: -
  m1, b1: |00000000000000000000000000000000| delays: -
  m1, b2: |00000000000000000000000000000000| delays: -
  m1, b3: |00000000000000000000000000000000| delays: -
  m1, b4: |00000000000000000000000000000000| delays: -
  m1, b5: |00000000000000000000000000000000| delays: -
  m1, b6: |00000000000000000000000000000000| delays: -
[... 65 lines cut ...]

--============= Console ================--
litex> mem_list

Available memory regions:
PLIC      0xf0c00000 0x400000
CLINT     0xf0010000 0x10000
ROM       0x00000000 0x10000
SRAM      0x10000000 0x2000
MAIN_RAM  0x40000000 0x40000000
FIRMWARE_RAM  0x20000000 0x8000
OPENSBI   0x40f00000 0x80000
CSR       0xf0000000 0x10000

litex> mem_test 0x40000000 1024

Memtest at 0x40000000 (1KiB)...
  Write: 0x40000000-0x40000400 1KiB
   Read: 0x40000000-0x40000400 1KiB
  bus errors:  224/256
  addr errors: 254/256
  data errors: 256/256
Memtest KO

```

enjoy-digital, 2021-01-29:
That's already better and you have first sign of life on m5. The calibration still requires some initial adjustments and will requires some tests:

Can you try to do the following sequence in the BIOS:
```
sdram_force_cmd_delay N
sdram_cal
sdram_test
```
With N in from 0 to 512 with steps of ~50 and share results.

If no improvements, could you try with `cmd_latency` set to 1 here: https://github.com/enjoy-digital/litedram/blob/master/litedram/phy/usddrphy.py#L33? And first see with the automatic calibration, then if not working doing the same `sdram_force_cmd_delay test` than the one done with the default `cmd_latency`?

smosanu, 2021-02-01:
Thank you for the instructions! I tried all of them plus some more but still didn't manage to get the sdram OK.

I added the C1 channel, in addition to C0, and double-checked that all the pin constraints and the IOstandard are OK. I compared with working MicroBlaze projects that use either C0 or C1 and the pins correspond. 

I tested both C0 and C1 with LiteX with cmd_latency set to 0 and then 1. I also tried the sequence above. Setting cmd_latency to 1 had some positive effect (multiple regions were showing signs of life) but the error rate remained high. Increasing the cmd_delay didn't have much effect. Attached is a log file with the tests with N increasing from 0 to 500 with cmd_latency-1.
[au280_litedram_log.txt](https://github.com/litex-hub/litex-boards/files/5907115/au280_litedram_log.txt)

I am not sure how to fix this... :(

smosanu, 2021-02-01:
Pinging in @daveshah1 - perhaps one more set of eyes can be of help!?

enjoy-digital, 2021-02-02:
Thanks for the tests/results @smosanu, it's not far from working. I assume the results you provided are for C1? (because write leveling is not working while it was with C0). For C1 have you also set the `INTERNAL_VREF `constraints? (could you do a PR for C1 addition so that I can also check?). If it was for C1, could you provide the same log with C0?

smosanu, 2021-02-02:
Yes, the results provided are for C1. The board I have is an engineering sample, and C0 gave me some trouble with a previous design that worked with C1, so I decided to test with C1. But the board is perfectly fine and works well under numerous other tasks such as SDx or MicroBlaze so I hope to get it to work with LiteX/LiteDRAM too!

The `INTERNAL_VREF` constraints are set for both C0 and C1. One signal that I am missing, when compared with the alveo_u250 board, is the `ddram_reset_gate`, though this signal is not used by the generated SoC. I did a PR that matches my last tests - thank you very much for having a look at it! I'll re-run the tests for C0 as soon as I get a chance and post the log. On C0 I was only seeing some life at m5; the other regions didn't show any improvements for any N.
Again, thank you!

smosanu, 2021-02-08:
Updating Issue with log of channel C0, still _Memtest KO_ but also quite a bit of life detected.
[au280_C0_litedram_log.txt](https://github.com/litex-hub/litex-boards/files/5947319/au280_C0_litedram_log.txt)

enjoy-digital, 2021-02-09:
Thanks @smosanu, that's encouraging. The last issue coudl be related to wrong electrical settings. Using `ddr4_mr_gen` (https://github.com/enjoy-digital/litedram/blob/master/bench/ddr4_mr_gen.py)  could allow you to do some tests with various settings without re-compiling the bistream. I'll try to give more indications on this soon.

smosanu, 2021-03-16:
Thank you @enjoy-digital,
I had one more look at the issue yesterday together with @vasimr22. We looked at the clock settings, electrical settings, constraints, looked for possible incorrect settings in general, and also played with the resistance and delay values using `ddr4_mr_gen` but were still unable to find what the issue is, not even a clue to what it might be. :( I'm attaching the log of the tests we run (using ddr4 channel 1). Basically we changed the register values as instructed by `ddr4_mr_gen` and re-run the tests. Perhaps you can see something we missed? Any suggestions of new tests we could run that might tell us someting? Thank you very much!
[ddr4bench.txt](https://github.com/litex-hub/litex-boards/files/6150339/ddr4bench.txt)

smosanu, 2021-03-24:
Updated a few dqs_n pins, constraints, and frequency speed (up to 300MHz, 1:4 leads to 1200MHz = native for DDR4 PHY) and am seeing improved (I hope!) initialization values. It is still failing though.
[LitexAU280at300MHz.txt](https://github.com/litex-hub/litex-boards/files/6194174/LitexAU280at300MHz.txt)

enjoy-digital, 2021-03-24:
Thanks @smosanu for the update. 300MHz sys_clk is a bit high for now and LiteDRAM hasn't been tested at this speed yet. I would recommend using 150MHz sys_clk to start with and then increase progressively when working.

smosanu, 2021-03-24:
Here are the boot log and reports at 150MHz for C1. Should I try varying CL/CWL? Funny but I actually thought the boot at 300MHz looked better.
[LitexAU280at150MHzCh1.txt](https://github.com/litex-hub/litex-boards/files/6198606/LitexAU280at150MHzCh1.txt)
[alveo_u280_timing.txt](https://github.com/litex-hub/litex-boards/files/6198631/alveo_u280_timing.txt)
[alveo_u280_utilization_hierarchical_place.txt](https://github.com/litex-hub/litex-boards/files/6198632/alveo_u280_utilization_hierarchical_place.txt)

smosanu, 2021-03-24:
Same as above, 150MHz, now with `cmd_latency = 1`, seems promising.
[LitexAU280at150MHzCh1CMDLtcy1.txt](https://github.com/litex-hub/litex-boards/files/6200432/LitexAU280at150MHzCh1CMDLtcy1.txt)

smosanu, 2021-04-07:
Created a pull request with the recent fixes: https://github.com/litex-hub/litex-boards/pull/197
Closing this issue since the remaining ddr4 errors can be a separate issue.

closed 2021-04-07 by smosanu