litex-hub/litex-boards #166: DDR3 only work at high speed on ZTEX 2.13

rdolbeau, 2021-01-31:
Following #164 , the onboard memory of the ZTEX 2.13 works fine at fairly high speed (core @ 90-100 MHz; working was tested with VexRiscV, failing with VexRiscV & Rocket), but fails at lower speed. Initialization simply fails at 100%.

The following magic commands in the BIOS (suggested by @enjoy-digital ) enable the memory to work:
```
sdram_mr_write 0 2608
sdram_mr_write 1 2054
sdram_mr_write 2 512
sdram_cal
sdram_test
```

Alternatively, the following patch also solves the problem (tested with Rocket):
```
diff --git a/litedram/init.py b/litedram/init.py                                                                                                                                                                                                                               
index d6cc83e..5baaf7e 100644                                                                                                                                                                                                                                                  
--- a/litedram/init.py                                                                                                                                                                                                                                                         
+++ b/litedram/init.py                                                                                                                                                                                                                                                         
@@ -116,7 +116,7 @@ def get_ddr2_phy_init_sequence(phy_settings, timing_settings):                                                                                                                                                                                             
 # DDR3 ---------------------------------------------------------------------------------------------                                                                                                                                                                          
                                                                                                                                                                                                                                                                               
 def get_ddr3_phy_init_sequence(phy_settings, timing_settings):                                                                                                                                                                                                                
-    cl  = phy_settings.cl                                                                                                                                                                                                                                                     
+    cl  = phy_settings.cl + 1                                                                                                                                                                                                                                                 
     bl  = 8                                                                                                                                                                                                                                                                   
     cwl = phy_settings.cwl                                                                                                                                                                                                                                                    
                                                                                                                                                                                                                                                                               
```

developandplay, 2021-06-08:
@enjoy-digital
Could you explain what the magic commands do?
Also is there a good intro guide to litedram? I'm completly lost when I encounter memory errors for rocket or blackparrot.

enjoy-digital, 2021-06-08:
@developandplay: The "magic comands" are used to reconfigure the DDR3 timings. 

This has been generated with: https://github.com/enjoy-digital/litedram/blob/master/bench/ddr3_mr_gen.py and:
`python3 ddr3_mr_gen.py --cl=7 --cwl=5`

My aim was probably to increase CL value.

enjoy-digital, 2021-09-14:
This has been fixed with https://github.com/enjoy-digital/litedram/commit/6f323f6a7aa032c8342b5931f4a562d2e28969e9.

closed 2021-09-14 by enjoy-digital