litex-hub/litex-boards #181: kc705 doesn't build with --integrated-main-ram-size set

jersey99, 2021-02-25:
Looks like SoCCore adds main_ram as a region already and kc705.py trys to add_sdram separately. What is the recommended way? Thanks!

```
$ python kc705.py --integrated-main-ram-size 0x100000 --build
INFO:SoC:        __   _ __      _  __  
INFO:SoC:       / /  (_) /____ | |/_/  
INFO:SoC:      / /__/ / __/ -_)>  <    
INFO:SoC:     /____/_/\__/\__/_/|_|  
INFO:SoC:  Build your hardware, easily!
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Creating SoC... (2021-02-25 09:25:33)
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:FPGA device : xc7k325t-ffg900-2.
INFO:SoC:System clock: 125.00MHz.
INFO:SoCBusHandler:Creating Bus Handler...
INFO:SoCBusHandler:32-bit wishbone Bus, 4.0GiB Address Space.
INFO:SoCBusHandler:Adding reserved Bus Regions...
INFO:SoCBusHandler:Bus Handler created.
INFO:SoCCSRHandler:Creating CSR Handler...
INFO:SoCCSRHandler:32-bit CSR Bus, 32-bit Aligned, 16.0KiB Address Space, 2048B Paging, big Ordering (Up to 32 Locations).
INFO:SoCCSRHandler:Adding reserved CSRs...
INFO:SoCCSRHandler:CSR Handler created.
INFO:SoCIRQHandler:Creating IRQ Handler...
INFO:SoCIRQHandler:IRQ Handler (up to 32 Locations).
INFO:SoCIRQHandler:Adding reserved IRQs...
INFO:SoCIRQHandler:IRQ Handler created.
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Initial SoC:
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:32-bit wishbone Bus, 4.0GiB Address Space.
INFO:SoC:32-bit CSR Bus, 32-bit Aligned, 16.0KiB Address Space, 2048B Paging, big Ordering (Up to 32 Locations).
INFO:SoC:IRQ Handler (up to 32 Locations).
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoCCSRHandler:ctrl CSR allocated at Location 0.
INFO:SoCBusHandler:io0 Region added at Origin: 0x80000000, Size: 0x80000000, Mode: RW, Cached: False Linker: False.
INFO:SoCBusHandler:cpu_bus0 added as Bus Master.
INFO:SoCBusHandler:cpu_bus1 added as Bus Master.
INFO:SoCCSRHandler:cpu CSR allocated at Location 1.
INFO:SoCBusHandler:rom Region added at Origin: 0x00000000, Size: 0x00008000, Mode: R, Cached: True Linker: False.
INFO:SoCBusHandler:rom added as Bus Slave.
INFO:SoC:RAM rom added Origin: 0x00000000, Size: 0x00008000, Mode: R, Cached: True Linker: False.
INFO:SoCBusHandler:sram Region added at Origin: 0x01000000, Size: 0x00002000, Mode: RW, Cached: True Linker: False.
INFO:SoCBusHandler:sram added as Bus Slave.
INFO:SoC:RAM sram added Origin: 0x01000000, Size: 0x00002000, Mode: RW, Cached: True Linker: False.
INFO:SoCBusHandler:main_ram Region added at Origin: 0x40000000, Size: 0x00100000, Mode: RW, Cached: True Linker: False.
INFO:SoCBusHandler:main_ram added as Bus Slave.
INFO:SoC:RAM main_ram added Origin: 0x40000000, Size: 0x00100000, Mode: RW, Cached: True Linker: False.
INFO:SoCCSRHandler:identifier_mem CSR allocated at Location 2.
INFO:SoCCSRHandler:uart_phy CSR allocated at Location 3.
INFO:SoCCSRHandler:uart CSR allocated at Location 4.
INFO:SoCIRQHandler:uart IRQ allocated at Location 0.
INFO:SoCCSRHandler:timer0 CSR allocated at Location 5.
INFO:SoCIRQHandler:timer0 IRQ allocated at Location 1.
INFO:S7MMCM:Creating S7MMCM, speedgrade -2.
INFO:S7MMCM:Registering Differential ClkIn of 200.00MHz.
INFO:S7MMCM:Creating ClkOut0 sys of 125.00MHz (+-10000.00ppm).
INFO:S7MMCM:Creating ClkOut1 sys4x of 500.00MHz (+-10000.00ppm).
INFO:S7MMCM:Creating ClkOut2 idelay of 200.00MHz (+-10000.00ppm).
INFO:SoCCSRHandler:ddrphy CSR allocated at Location 6.
@#$@#$ sdram made it here
INFO:SoCCSRHandler:sdram CSR allocated at Location 7.
ERROR:SoCBusHandler:main_ram already declared as Region:
ERROR:SoCBusHandler:32-bit wishbone Bus, 4.0GiB Address Space.
IO Regions: (1)
io0                 : Origin: 0x80000000, Size: 0x80000000, Mode: RW, Cached: False Linker: False
Bus Regions: (3)
rom                 : Origin: 0x00000000, Size: 0x00008000, Mode: R, Cached: True Linker: False
sram                : Origin: 0x01000000, Size: 0x00002000, Mode: RW, Cached: True Linker: False
main_ram            : Origin: 0x40000000, Size: 0x00100000, Mode: RW, Cached: True Linker: False
Bus Masters: (2)
- cpu_bus0
- cpu_bus1
Bus Slaves: (3)
- rom
- sram
- main_ram
Traceback (most recent call last):
  File "kc705.py", line 178, in <module>
    main()
  File "kc705.py", line 160, in main
    soc = BaseSoC(
  File "kc705.py", line 77, in __init__
    self.add_sdram("sdram",
  File "/home/w/software/litex/litex/litex/soc/integration/soc.py", line 1248, in add_sdram
    self.bus.add_region("main_ram", SoCRegion(origin=origin, size=sdram_size))
  File "/home/w/software/litex/litex/litex/soc/integration/soc.py", line 170, in add_region
    raise

```

jersey99, 2021-02-25:
@enjoy-digital I think I just understood the problem. 

Line 72 of targets/kc705.py should read:
```
if not self.integrated_main_ram_size:
```

I will leave this open, till somebody fixes it. I will send a PR in a bit. But if you get to it first feel free to close this issue.

enjoy-digital, 2021-03-01:
@jersey99: Sorry, I'm not sure to understand. I'm not able to reproduce with: ` ./kc705.py --integrated-main-ram-size 0x100000 --build`

jersey99, 2021-03-02:
@enjoy-digital Ugh, sorry, I traced my problems back to not having the latest version of litedram checked out :facepalm:  I can no longer reproduce the problem either, which is a good thing! Closing ..

closed 2021-03-02 by jersey99

enjoy-digital, 2021-03-02:
@jersey99 No problem, thanks for the feedback.