litex-hub/litex-boards #194: project requires installing dependencies for all platforms/targets

shuffle2, 2021-03-27:
I've recently updated the migen/litex dependencies in a project tree I saw that litex-boards had been split off into it's own repo (yes, I'm slow on the update .. :))

Looking at https://github.com/litex-hub/litex-boards/blob/master/litex_boards/__init__.py , it seems like now litex will try to install dependencies for every possible platform+target, even if it's not actually being used. Or am I just doing something incorrectly? For now I've just locally modified __init__.py to reference only the single target I care about (lattice_versa_ecp5).

enjoy-digital, 2021-03-28:
Thanks for the feedback @shuffle2, I added this for retro-compatibility and will probably revisit it. What is the exact dependency issue you see? The fact that since alls the targets are imported all the cores used in the different targets are required?

shuffle2, 2021-03-29:
> The fact that since alls the targets are imported all the cores used in the different targets are required?

Yes...IMO it kind of defeats the purpose of having stuff split between different repos, with support code for downloading dependencies. Currently it seems like the typical git submodule use case (where one is expected to recursively clone submodules, which are just split to make managing large amounts of data tolerable in git), except submodules aren't being used.

enjoy-digital, 2021-03-30:
I understand, this is just temporary until I spend time finding a better solution to provide the platform/target name retro-compatibility. (The platforms/targets had to be renamed to allow supporting more boards, but breaking name  retro-compatibility would also have been confusing for users). Will try to  look at this soon.

PR #199 (2021-04-11): boards/compat: Skip targets with unsatisfied dependencies.
Fixes #194
commit 2e204285 (2021-04-11): boards/compat: Skip targets with unsatisfied dependencies.

Fixes #194
closed 2021-04-11 by enjoy-digital