litex-hub/litex-boards #219: Add DDR3 Memory into the Zybo Z7 board target for Using it as RAM for OS as in Arty A7 board in the targets/digilent_arty.py

bhatmahadev6, 2021-05-21:
I am not able to understand how we can make use of  DDR3 as the main memory for Vexrisc-v Core to boot an RTOS (for example NuttX) in Zybo Z7/Zybo (**as DDR3 is tied to ZYNQ processing system**)

Please tell me how can  I use DDR3 in Zybo as Ram !!

Thanks

enjoy-digital, 2021-05-23:
Hello @bhatmahadev6,

the DDR3 is indeed connected to the Zynq we don't yet support access to the DDR3 of the Zynq from the fabric in LiteX SoCs. This is probably only a matter of using the right PS configuration on the Zynq and connecting the LiteX SoC through an AXI Master interface to the Zynq, but this hasn't been tested yet.

mithro, 2021-05-24:
I believe Antmicro have done something like this. @kgugala thoughts?

kgugala, 2021-05-24:
what @enjoy-digital writes is correct, all you need to do is to connect the system bus (via bridge) to one of the PS7 AXI buses where FPGA is a master (e.g. S_AXI_GP0). Also, you need map the bridge as memory at `0x0` (this may require moving ROM somewhere else), or insert some logic to do address mapping in the design. Keep in mind that the first 256 kB of Zynq's memory space is in fact on-chip RAM (OCR) (not really DDR). This can be changed in one of the Zynq's System Level Control Registers (SLCR) - you can move OCR to the end of the 4GB memory space.

I actually ran LiteX/Vex SoC booting Linux on Zybo using PS DDR3 some time ago and it worked fine. You basically need to boot the Zynq up to the point where DDR controller is initialized and enabled. Next you can simply load bitstream and let LiteX run.

enjoy-digital, 2021-05-25:
Thanks @kgugala, that's interesting to know you already tested it (and to have the know  the small subtleties to get it working).

bhatmahadev6, 2021-06-14:
Hi @kgugala, Do you have any example for the above one?
I am new to Litex and quickly want to ramp up with the NuttX operating system.

enjoy-digital, 2022-03-24:
@bhatmahadev6: I think I and @kgugala provided the necessary information to do what you initially wanted to do, but this requires a bit of FPGA development experience since is not handled by LiteX directly. Before exploring this, I would recommend being more familiar with LiteX with a board well supported by LiteX, for example the Digilent Arty where Soft-CPU and Soft-DDR3 controller are possible and that will allow you to play with Zephyr, NuttX, Linux, etc...

We'll probably improve the aspects you wanted to explore in the future in LiteX as part of development done for specific projects but things haven't been planned yet.

closed 2022-03-24 by enjoy-digital