litex-hub/litex-boards #224: SD card support broken on ulx3s

developandplay, 2021-06-10:
When using vexriscv or rocket with the ulx3s the `--with-sdcard` flag is not working for me.
This applies both to booting from the SD card as well as accessing the SD card in linux.

Using the `--with-spi-sdcard` flag works for both booting as well as linux.

```
--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
Timeout
Booting from SDCard in SD-Mode...
Booting from boot.json...
Booting from boot.bin...
SDCard boot failed.
No boot medium found

--============= Console ================--
```

developandplay, 2021-06-10:
When enabling SD card debug output this is what I am getting:

```
CMD55: APP_CMD
cmdevt: 00000000
cmdevt: 00000001
ff7fffff ffff7fff ffffff7f ffffffff
ACMD41: APP_SEND_OP_COND, arg: 70ff8000
cmdevt: 00000000
cmdevt: 00000001
ff7fffff ffff7fff ffffff7f ffffffff
```

enjoy-digital, 2022-03-24:
The issue was related to pin changes between 1.7 and 2.0 revisions. This has been fixed and tested with https://github.com/litex-hub/litex-boards/commit/68c23e92514a83e314e75d150dd087f6bc4f80e4.

closed 2022-03-24 by enjoy-digital

sebeaumont, 2022-04-27:
I have this board: 
```v3.0.8    INEM-KONČAR   jul 2020   v20   IS42S16160G-7TL  W25Q128JVSIQ```

And encounter the exactly same problem with build after that commit was merged.

sebeaumont, 2022-04-27:
Should this be `spiserialboot` as I have got to exactly the same point with the `organgecrab 85k`; the SD card however is correctly detected by linux. Not so on the ULX3.

Irvise, 2023-01-05:
I am in the same scenario as ept. However, on top of that, the SDCard does not work if I use the Wishbone bus, but it does work with the AXI one.