litex-hub/litex-boards #225: enclustra_mercury_kx2 won't build

cbakalis, 2021-06-15:
Hello,

Using the master branch of `litex` and `litex_boards`, I cannot build the `enclustra_mercury_kx2` project.

After navigating to `itex_boards/targets`, I run `$ python3 enclustra_mercury_kx2.py --build`. During Vivado's P&R, I get the following error:

```
ERROR: [Place 30-110] The following IOBs use the Digitally Controlled Impedance feature (DCI) and have been constrained (LOC constraint) to the I/O bank 32.
This feature requires the VRN and VRP pins within the same I/O bank to be connected to reference resistors.
The following VR pins are currently locked and can't be used to supply the necessary reference.
IO Standard: SioStd: SSTL15_T_DCI VREF = 0.75 VCCO = 1.5 Termination: 2  TermDir:  BiDi VccAuxIOType: 1 IdelayId: 1 Bank: 32 Placed :
     List of placed IOBs :
	Term: ddram_dq[32]
	Term: ddram_dq[33]
	Term: ddram_dq[34]
	Term: ddram_dq[35]
	Term: ddram_dq[36]
        [...]
	Term: ddram_dq[63]

List of Occupied VR Sites: 
	VR Pin: V13 is occupied by term: user_led2
	VR Pin: W13 is occupied by term: user_led3

ERROR: [Place 30-374] IO placer failed to find a solution
Below is the partial placement that can be analyzed to see if any constraint modifications will make the IO placement problem easier to solve.

+-------------------------------------------------------------------------------------------------------------------------------------------------------------------------+
|                                                                     IO Placement : Bank Stats                                                                           |
+----+-------+-------+------------------------------------------------------------------------+------------------------------------------+--------+--------+--------+-----+
| Id | Pins  | Terms |                               Standards                                |                IDelayCtrls               |  VREF  |  VCCO  |   VR   | DCI |
+----+-------+-------+------------------------------------------------------------------------+------------------------------------------+--------+--------+--------+-----+
|  0 |     0 |     0 |                                                                        |                                          |        |        |        |     |
| 12 |    50 |     0 |                                                                        |                                          |        |        |        |     |
| 13 |    50 |     0 |                                                                        |                                          |        |        |        |     |
| 14 |    50 |     0 |                                                                        |                                          |        |        |        |     |
| 15 |    50 |     0 |                                                                        |                                          |        |        |        |     |
| 16 |    50 |     1 | LVCMOS25(1)                                                            |                                          |        |  +2.50 |    YES |     |
| 32 |    50 |    47 | LVCMOS15(3)  SSTL15(4)  SSTL15_T_DCI(32)  DIFF_SSTL15(8)               |                                          |  +0.75 |  +1.50 |    YES | YES |
| 33 |    50 |    32 | LVCMOS15(4)  SSTL15(24)  DIFF_SSTL15(2)  LVDS(2)                       |                                          |  +0.75 |  +1.80 |    YES |     |
| 34 |    50 |    44 | SSTL15(4)  SSTL15_T_DCI(32)  DIFF_SSTL15(8)                            |                                          |  +0.75 |  +1.50 |    YES | YES |
+----+-------+-------+------------------------------------------------------------------------+------------------------------------------+--------+--------+--------+-----+
|    |   400 |   124 |                                                                        |                                          |        |        |        |     |
+----+-------+-------+------------------------------------------------------------------------+------------------------------------------+--------+--------+--------+-----+
```

Any help is much appreciated!

AnttiLukats, 2021-06-16:
it seems those pins are really used by the LEDs! Maybe mercury board use DCI borrowing and that is not supported by litex?

enjoy-digital, 2021-06-16:
Thanks @AnttiLukats, the LEDs pins indeed seems conflicting with the DRAM pins. @cbakalis the build has been fixed with https://github.com/litex-hub/litex-boards/commit/fa045e6fa4febed5a90616ac8d9ce5b6b9507861.