litex-hub/litex-boards #226: major error building

AnttiLukats, 2021-06-16:
it  seems most boards fail with following error

B:\GitHub\litex-boards\litex_boards\targets>python terasic_de0nano.py --build                                                                                                                                      Traceback (most recent call last):                                                                                                                                                                                   File "terasic_de0nano.py", line 17, in <module>                                                                                                                                                                      from litex_boards.platforms import de0nano                                                                                                                                                                       File "C:\Users\antti\AppData\Local\Programs\Python\Python38\lib\site-packages\litex_boards-0.0.0-py3.8.egg\litex_boards\__init__.py", line 65, in <module>                                                           t = importlib.import_module(f"litex_boards.targets.{target}")                                                                                                                                                    File "C:\Users\antti\AppData\Local\Programs\Python\Python38\lib\importlib\__init__.py", line 127, in import_module                                                                                                   return _bootstrap._gcd_import(name[level:], package, level)                                                                                                                                                      File "C:\Users\antti\AppData\Local\Programs\Python\Python38\lib\site-packages\litex_boards-0.0.0-py3.8.egg\litex_boards\targets\antmicro_lpddr4_test_board.py", line 21, in <module>                                 from litedram.modules import MT53E256M16D1                                                                                                                                                                     ImportError: cannot import name 'MT53E256M16D1' from 'litedram.modules' (b:\github\litex\litedram\litedram\modules.py) 

tried 4 different boards from 2 different vendors same error

enjoy-digital, 2021-06-16:
Hi @AnttiLukats,

the LiteX ecosystem is splitted into different repositories, please make sure to update the different repositories manually (git pull) or use `litex_setup.py update`.

Here clearly, LiteDRAM repository is not up to date on your machine, since you can see that MT53E256M16D1 is present here: https://github.com/enjoy-digital/litedram/blob/master/litedram/modules.py#L984

So you just need to update the dependencies. For future issues, please make sure to read https://github.com/enjoy-digital/litex/wiki/Feedback-Contribution-Support#how-to-create-an-issue-when-it-does-not-work, please try to do minimal verification on your side and give the maximum of context, it's always easier/faster to investigate on a local machine, than ask us to imagine what is happening on an external machine :)

closed 2021-06-16 by enjoy-digital

AnttiLukats, 2021-06-16:
thanks, it was github desktop issue, I did update from there but it did not pull new files, Sorry!