litex-hub/litex-boards #231: Colorlight-5a vexriscv no console bios

stanleyseow, 2021-06-25:
I encounter a weird issue, with --cpu-type vexriscv vs --cpu-type  picorv32

Using the SAME hardware / serial port ( FT2232 ) I was able to get a bios using --cpu-type picorv32 but not able to get bios using vexriscv cpu

Only differences are--cpu-type : 
colorlight_5a_75x.py --cpu-type picorv32 --build
colorlight_5a_75x.py --cpu-type vexriscv --build

I got a pre-compiled vexriscv.svf n it can load and show the bios. I wonder why I can't syn my own softcpu ???

I m using the default 7.0 pinout ..

```
_io_v7_0 = [ # Documented by @miek
    # Clk
    ("clk25", 0, Pins("P6"), IOStandard("LVCMOS33")),

    # Led
    ("user_led_n", 0, Pins("P11"), IOStandard("LVCMOS33")),

    # Button
    ("user_btn_n", 0, Pins("M13"), IOStandard("LVCMOS33")),

    # Serial
    ("serial", 0,
        Subsignal("tx", Pins("P11")), # led (J19 DATA_LED-)
        Subsignal("rx", Pins("M13")), # btn (J19 KEY+)
        IOStandard("LVCMOS33")
    ),
```

enjoy-digital, 2021-06-28:
@stanleyseow: Hi, can you try to do a `rm -rf build ` and rebuild the VexRiscv SoC? This coudl be related to https://github.com/enjoy-digital/litex/issues/950.

stanleyseow, 2021-06-28:
Thanks...

solved with rm -rf build 

closing this issue...

closed 2021-06-28 by stanleyseow