litex-hub/litex-boards #232: qmtech_ep4ce15.py serial port definition "rx", Pins("J3:8") or Pins("J3:9") in linux-on-litex-vexriscv application

marconi1964, 2021-06-27:
Hi,

I got a chance to know LiteX recently from my friend who gave me a QMTech_EP4CE15F23C8N board, and practice linux-on-litex-vexriscv on it.

My friend told me to connect my USB-UART serial port to J3:7 and J3:9 (J3 is U8 on the qmtech board), which works well. But when I check this settings litex-boards/litex_boards/platforms/qmtech_ep4ce15.py regarding Serial port definition below, I am confused. The setting is J3:7 and J3:8 (instead of 9). So I am wondering if the Serial port used in linux-on-litex-vexriscv is defined elsewhere, and not here?

```
    # Serial
    ("serial", 0,
        # Compatible with cheap FT232 based cables (ex: Gaoominy 6Pin Ftdi Ft232Rl Ft232)
        Subsignal("tx", Pins("J3:7"), IOStandard("3.3-V LVTTL")), # GPIO_07 (JP1 Pin 10)
        Subsignal("rx", Pins("J3:8"), IOStandard("3.3-V LVTTL"))  # GPIO_05 (JP1 Pin 8)
    ),

# ......

    ("J3", {
        # odd row     even row
         7: "AA13",   8: "AB13",
         9: "AA14",  10: "AB14",

```

Thank you,

enjoy-digital, 2022-03-24:
Hi @marconi1964,

the pinout used by Linux-on-LiteX-Vexrsicv for the serial is the one defined in LiteX-Boards, so if there is an issue/inconsistency, it's in `qmtech_daughterboard.py/``qmtech_ep4cex5.py` platform files.

@hansfbaier: Do you have an idea on this? Are you using J3:8 as FPGA TX and J3:7 as FPGA RX?

hansfbaier, 2022-03-24:
@marconi1964 The board platform files for EP4CEx5 specify:
```python
            Subsignal("tx", Pins("J3:7"), IOStandard("3.3-V LVTTL")),
            Subsignal("rx", Pins("J3:8"), IOStandard("3.3-V LVTTL"))
```
Which means you should wire your UART tx to J3:8 and UART tx to J3:7 (if I'm not mistaken).
The linux on Vexriscv project uses the serial from the platform file.

enjoy-digital, 2022-05-27:
We can probably close this now.

closed 2022-05-27 by enjoy-digital