litex-hub/litex-boards #235: Thoughts on adding the option to remove the LedChaser submodule from targets?

JosephBushagour, 2021-06-29:
The LedChaser submodule is a great diagnostic tool but it is somewhat costly in terms of LCs (around 70 LCs on ice40up5k boards like Fomu/iCEbreaker). I was thinking of submitting a PR that adds a `led_chaser=True` keyword argument to BaseSoCs of ice40up5k boards so that a user can turn off the LedChaser if they don't want it, but first I wanted to hear your thoughts on if that's a good or bad idea.

enjoy-digital, 2021-07-02:
@JosephBushagour That's a good idea yes, maybe we could just name it `with_led_chaser` for consistency with the other optional features (`with_pcie`, `with_etherne`t, etc...).

smosanu, 2021-07-05:
I'd also prefer having `with_led_chaser` as an option because my ES1 board has an issue with the way the LEDs are soldered and one of the LEDs creates a short that crashes the entire machine, so I had to disable it so far anyway.

enjoy-digital, 2021-07-21:
Thanks for PR #236 @JosephBushagour, this has been merged so we can close this.

closed 2021-07-21 by enjoy-digital