litex-hub/litex-boards #237: Avnet Board Design

garrettxw, 2021-07-09:
Hello,
I'm working with a research group that is doing radiation testing on a variety of designs including Linux on VexRiscv which Litex supports. We have been using the KCU105 board, but there are power regulation issues so we would rather use the Avnet boards that use the same XCKU040 chip on. I was wondering if, before the Avnet boards were discontinued if there was a preliminary build in progress or at least some things to look for in order to port the design to the Avnet.

enjoy-digital, 2021-07-10:
Hi @garrettxw,

we indeed don't have that much Avnet board supported but if your Avnet board uses a XCKU040 with a DDR3/DDR4 and peripherals already supported in LiteX, porting LiteX to this board should be easily (would mostly consist defining the platform and create the target by taking inspiration from existing boards that are very close).

garrettxw, 2021-07-13:
Hey @enjoy-digital 
Ok, that's basically what we assumed, we just thought it would be good to check with you. Thanks!

closed 2021-07-13 by garrettxw