litex-hub/litex-boards #238: gowin gw_sh script doesn't recognize commands

Martoni, 2021-07-10:
Hello,
I'm trying to build LiteX for the board trenz-tec0117 like described in this [blog-article](http://www.fabienm.eu/flf/test-dune-carte-gowin-avec-litex/) (in french).

But there is an error when Litex run the tcl gowin command line : 

```
...
make: Leaving directory '/home/user/myapp/litex/myapp/build/trenz_tec0117/software/bios'
INFO:SoC:Initializing ROM rom with contents (Size: 0x51cc).
INFO:SoC:Auto-Resizing ROM rom from 0x6000 to 0x51cc.
 *** Gowin Tcl command line console  *** 
invalid command name "set_device"
    while executing
"set_device -name GW1NR-9 GW1NR-LV9QN88C6/I5"
    (file "run.tcl" line 1)
Traceback (most recent call last):
  File "../litex-boards/litex_boards/targets/trenz_tec0117.py", line 210, in <module>
    main()
  File "../litex-boards/litex_boards/targets/trenz_tec0117.py", line 198, in main
    builder.build(run=args.build)
  File "/home/user/myapp/litex/litex/litex/soc/integration/builder.py", line 293, in build
    vns = self.soc.build(build_dir=self.gateware_dir, **kwargs)
  File "/home/user/myapp/litex/litex/litex/soc/integration/soc.py", line 1113, in build
    return self.platform.build(self, *args, **kwargs)
  File "/home/user/myapp/litex/litex/litex/build/gowin/platform.py", line 40, in build
    return self.toolchain.build(self, *args, **kwargs)
  File "/home/user/myapp/litex/litex/litex/build/gowin/gowin.py", line 126, in build
    raise OSError("Error occured during Gowin's script execution.")
OSError: Error occured during Gowin's script execution.
```

I can launch gw_ide without problem,  and I'm using floating licence server like described [here](http://tangnano.sipeed.com/en/get_started/install-the-ide.html).

Does somebody had a similar issue ?

I'm using gowin eda `Version 1.9.2 Beta build(34175)`

@enjoy-digital

enjoy-digital, 2021-07-10:
Hi @Martoni,

I'm not sure to remember which version of Gowin EDA I was using/testing with. (Could be related to a change in the API?). I'll be able to do tests on Monday and update Gowin EDA.

Martoni, 2021-07-10:
I don't think that is a LiteX bug in fact. Because if I run manualy the tcl console binarie named `gw_sh` I get the same error : 

```
$ ./gw_sh
 *** Gowin Tcl command line console  *** 
% set_device -name GW1NR-9 GW1NR-LV9QN88C6/I5
invalid command name "set_device"
% 
```

I don't have this bug on my work computer. I'll see wich version of gowin I've installed on it.

Thanks

Martoni, 2021-07-12:
Gowin eda version is more recent on my work computer : `1.9.7.01 Beta build (47178)`
Maybe is the problem.

Martoni, 2021-07-12:
It was this: gowin eda version problem.
Solved with installing last version 1.9.7.

closed 2021-07-12 by Martoni