litex-hub/litex-boards #241: Question about `bios_flash_offset` in `targets/kosagi_fomu.py`

JosephBushagour, 2021-07-16:
In the `kosagi_fomu.py` file the `--bios-flash-offset` argument has a default of `0x60000` but the flash function disregards its `bios_flash_offset` argument and always places the bios at `0x0020000`. I would have thought this was a bug but using the file I can successfully boot into bios while leaving the `bios_flash_offset` value at `0x60000`. I could be missing something, why does this work?
https://github.com/litex-hub/litex-boards/blob/4b48f15265c902dc2931809444b5eb803023d0b5/litex_boards/targets/kosagi_fomu.py#L120-L161

mithro, 2021-07-17:
@xobs / @umarcor - Any idea?

xobs, 2021-07-18:
I'm not sure, but I suspect it's because the bios is built out of bram, and isn't XIP from SPI flash or anywhere like that.

enjoy-digital, 2021-07-21:
@JosephBushagour It indeed seems the flash function and flash offset support was not completely finished, which I just did with https://github.com/litex-hub/litex-boards/commit/a455713e0c75538832d01fc35bf50a4e1d4a2820.

0x60000 - 0x20000 = 0x40000 which I suppose is the offset added by Foboot (for the DFU bitstream + firmware) which i now made explicit. @xobs can you confirm this?

xobs, 2021-07-21:
With Foboot, it doesn't matter /where/ the BIOS goes since:

1. RISC-V code is mostly PIC within a 4 MB window, and
2. BRAM can shift anywhere since it's not actually reading from flash

The program that it jumps to is usually located at offset `0x4_0000` from the start of flash, or address `0x2004_0000` in memory-mapped RAM (SPI is mapped XIP at `0x2000_0000). See https://github.com/im-tomu/foboot/blob/master/doc/FLASHLAYOUT.md for the flash layout -- I wish we had memory maps generated as part of the documentation.

I'm not sure what the situation is with this command, since I've never used this to prepare an image. However, if you're loading a bitstream from SPI flash, then indeed it will be located on the flash chip at offset `0x4_0000`.

enjoy-digital, 2021-07-21:
Thanks @xobs, I indeed wanted to have confirmation of the offset introduced by the DFU programmer (ie that a write at 0 is physically localed at `0x4_0000` in the SPI Flash).

In this target, we are relying on DFU for the flashing and then using XiP for the BIOS, so we just need to handle this offset which is now the case.

enjoy-digital, 2021-07-21:
@JosephBushagour: I think we can now close this since fixed (while doing the fix I also tested it with different `--bios_flash_offset` values). Please re-open if you have troubles.

closed 2021-07-21 by enjoy-digital