litex-hub/litex-boards #244: example with LiteDRAM but no mmu

Aljeshka, 2021-07-19:
Heyy

is there an exampple demo (preferably arty), that uses LiteDRAM, but NOMMU? I simply want to build an example design, where i can read from DRAM on a specific PHYSICAL address, and actually see that address on the DRAM port in the form of ROW_BANK_COLUMN.

Example: sdram_force_rdphase 0x4FFFFFF0.
Result: ROW_BANK_COLUMN <= 0xFFFFFF.

enjoy-digital, 2021-07-21:
Hello,

we don't have a MMU in LiteDRAM, the user address is directly mapped to Rows/Banks/Columns with a ROW_BANK_COLUMN mapping:

Imagine you use this SDRAM module: https://github.com/enjoy-digital/litedram/blob/master/litedram/modules.py#L448-L455
It has:
- 9 bit for COL (512)
- 2 bit for BANKS (4)
- 13 bits for ROWs (8192)

The mapping of your address (in words) will just be the following:
COL   = address[0:9]
BANK = address[9:11]
ROW = address[11:24]

Aljeshka, 2021-07-21:
Maybe I am misunderstanding something. In the litex interface I have the commands sdram_test and sdram_init. I think sdram_test reads and writes from 0x40000000 - 0x40800000. and sdram_init reads and writes from 0x40000000 - 0x40200000. 

I have a 256MB DDR3 with 8 banks. For the sdram_test command i would expect to see the address 0x401234567 at some point. This would mean the "effective" address is 0x123456, or in binary 0001 0010 0011 0100 0101 0110, right?
So first command is activate row[13:0] = 00001 0010 0011 01 in ba[2:0] = 000. The second command is column acces with col[6:0] = 101 0110.

When I'm measuring these signals i get totally different rows and columns though. So i thought there is some MMU and the addresses are virtual?

enjoy-digital, 2021-07-21:
@Aljeshka: To simplify the work on this, I would recommend using `litex_sim`and enable the SDRAM (--with-sdram) and also specify a SDRAM module similar to your hardware(--sdram-module/--sdram-data-width). You can then add more instrumentation to the [model](https://github.com/enjoy-digital/litedram/blob/master/litedram/phy/model.py) with `DIsplay` to look at the addresses.

Aljeshka, 2021-07-21:
i will try it out thanks

Aljeshka, 2021-07-21:
@enjoy-digital: I added some Display commands to [model](https://github.com/enjoy-digital/litedram/blob/master/litedram/phy/model.py) and `--with-sdram --sdram-module MT41K128M16` to `litex_sim`. Simulation works, but none of the Display commands is showing. Any idea what I am doing wrong?

enjoy-digital, 2021-09-08:
@Aljeshka: Sorry for the late answer, be sure to add `self.sync += ` before your Display command.

enjoy-digital, 2022-01-07:
I think the question has been answered, closing.

closed 2022-01-07 by enjoy-digital