litex-hub/litex-boards #247: No PLL config in hackaday_hadbadge

MateuszKosmala, 2021-07-28:
While building hackaday_hadbadge target I've got  exception like:
```
INFO:SoC:        __   _ __      _  __  
INFO:SoC:       / /  (_) /____ | |/_/  
INFO:SoC:      / /__/ / __/ -_)>  <    
INFO:SoC:     /____/_/\__/\__/_/|_|  
INFO:SoC:  Build your hardware, easily!
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Creating SoC... (2021-07-28 09:53:22)
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:FPGA device : LFE5U-45F-8CABGA381.
INFO:SoC:System clock: 48.00MHz.
INFO:SoCBusHandler:Creating Bus Handler...
INFO:SoCBusHandler:32-bit wishbone Bus, 4.0GiB Address Space.
INFO:SoCBusHandler:Adding reserved Bus Regions...
INFO:SoCBusHandler:Bus Handler created.
INFO:SoCCSRHandler:Creating CSR Handler...
INFO:SoCCSRHandler:32-bit CSR Bus, 32-bit Aligned, 16.0KiB Address Space, 2048B Paging, big Ordering (Up to 32 Locations).
INFO:SoCCSRHandler:Adding reserved CSRs...
INFO:SoCCSRHandler:CSR Handler created.
INFO:SoCIRQHandler:Creating IRQ Handler...
INFO:SoCIRQHandler:IRQ Handler (up to 32 Locations).
INFO:SoCIRQHandler:Adding reserved IRQs...
INFO:SoCIRQHandler:IRQ Handler created.
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Initial SoC:
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:32-bit wishbone Bus, 4.0GiB Address Space.
INFO:SoC:32-bit CSR Bus, 32-bit Aligned, 16.0KiB Address Space, 2048B Paging, big Ordering (Up to 32 Locations).
INFO:SoC:IRQ Handler (up to 32 Locations).
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoCBusHandler:io0 Region added at Origin: 0x80000000, Size: 0x80000000, Mode: RW, Cached: False Linker: False.
INFO:SoCBusHandler:cpu_bus0 added as Bus Master.
INFO:SoCBusHandler:cpu_bus1 added as Bus Master.
INFO:SoCBusHandler:rom Region added at Origin: 0x00000000, Size: 0x00020000, Mode: R, Cached: True Linker: False.
INFO:SoCBusHandler:rom added as Bus Slave.
INFO:SoC:RAM rom added Origin: 0x00000000, Size: 0x00020000, Mode: R, Cached: True Linker: False.
INFO:SoCBusHandler:sram Region added at Origin: 0x10000000, Size: 0x00002000, Mode: RW, Cached: True Linker: False.
INFO:SoCBusHandler:sram added as Bus Slave.
INFO:SoC:RAM sram added Origin: 0x10000000, Size: 0x00002000, Mode: RW, Cached: True Linker: False.
INFO:SoCIRQHandler:uart IRQ allocated at Location 0.
INFO:SoCIRQHandler:timer0 IRQ allocated at Location 1.
INFO:ECP5PLL:Creating ECP5PLL.
INFO:ECP5PLL:Registering Single Ended ClkIn of 8.00MHz.
INFO:ECP5PLL:Creating ClkOut0 sys of 48.00MHz (+-10000.00ppm).
INFO:ECP5PLL:Creating ClkOut1 sys_ps of 48.00MHz (+-10000.00ppm).
INFO:SoCBusHandler:main_ram Region added at Origin: 0x40000000, Size: 0x02000000, Mode: RW, Cached: True Linker: False.
INFO:SoCBusHandler:main_ram added as Bus Slave.
Traceback (most recent call last):
  File "/home/m3et/Antmicro/litex-boards/litex_boards/targets/./hackaday_hadbadge.py", line 101, in <module>
    main()
  File "/home/m3et/Antmicro/litex-boards/litex_boards/targets/./hackaday_hadbadge.py", line 98, in main
    builder.build(**builder_kargs, run=args.build)
  File "/home/m3et/Antmicro/litex/litex/soc/integration/builder.py", line 261, in build
    self.soc.finalize()
  File "/home/m3et/Antmicro/migen/migen/fhdl/module.py", line 156, in finalize
    subfragments = self._collect_submodules()
  File "/home/m3et/Antmicro/migen/migen/fhdl/module.py", line 149, in _collect_submodules
    r.append((name, submodule.get_fragment()))
  File "/home/m3et/Antmicro/migen/migen/fhdl/module.py", line 102, in get_fragment
    self.finalize()
  File "/home/m3et/Antmicro/migen/migen/fhdl/module.py", line 156, in finalize
    subfragments = self._collect_submodules()
  File "/home/m3et/Antmicro/migen/migen/fhdl/module.py", line 149, in _collect_submodules
    r.append((name, submodule.get_fragment()))
  File "/home/m3et/Antmicro/migen/migen/fhdl/module.py", line 102, in get_fragment
    self.finalize()
  File "/home/m3et/Antmicro/migen/migen/fhdl/module.py", line 157, in finalize
    self.do_finalize(*args, **kwargs)
  File "/home/m3et/Antmicro/litex/litex/soc/cores/clock/lattice_ecp5.py", line 135, in do_finalize
    config = self.compute_config()
  File "/home/m3et/Antmicro/litex/litex/soc/cores/clock/lattice_ecp5.py", line 114, in compute_config
    raise ValueError("No PLL config found")
ValueError: No PLL config found

```
I noticed that PhaseDetector Input Frequency should be between 10 and 400 MHz ([pfd_freq](https://github.com/enjoy-digital/litex/blob/8415e5d6acdf8ba35eee48489b324c281edaaa23/litex/soc/cores/clock/lattice_ecp5.py#L23)) and hackaday_hadbadge doesn't meet that requirements since frequency input to the PLL in board ([clock freq](https://github.com/litex-hub/litex-boards/blob/a3f479837c0950d7f33f6a9e7bc582eef28c01b2/litex_boards/targets/hackaday_hadbadge.py#L49)) is **8e6**, so maybe It is idea to change(lower) **pfd_freq_range**?

enjoy-digital, 2021-07-28:
Thanks for the feedback @MateuszKosmala, I was indeed almost sure there were some case where pfd_freq was below the requirement and introducing a check in ECP5PLL would break it :) Using 8e6 has been working correctly on hardware, so we could indeed lower it. We can probably keep the current code in LiteX and just lower it for the hackaday_hadbadge. I'm going to look at this.

enjoy-digital, 2021-07-28:
This is fixed with https://github.com/litex-hub/litex-boards/commit/64eadd8012d1ed97bf496e02f36281ca0a48b66f.

enjoy-digital, 2021-07-28:
The defauit configuration of all targets is now also tested in CI, this should help catching similar issues in the future.

closed 2021-07-28 by enjoy-digital