litex-hub/litex-boards #253: need some help for a new Altera Cyclone IV board

alainlou, 2021-08-14:
Litex is really cool! I want to use it on my FPGA board!

I have a RZRD Cyclone IV board which isn't one of the currently supported boards

I think I need to create two new python files under platforms/ and targets/

These I assume I can base off some of the existing Cyclone IV boards

Is this the correct approach?

jevinskie, 2021-08-20:
Yes. (Also answered on IRC)

alainlou, 2021-08-22:
So I've got a nice LED snake with a period of ~1 second :smile: but I'm having some trouble getting to the neat BIOS menu

I'm using `[azl@fedora targets]$ ./rz_easyfpga.py --cpu-type=vexriscv --cpu-variant=minimal --build --load` to build and load the board with the platform and target files from [here](https://github.com/alainlou/litex-boards)

The output from the serial port I'm getting is this:
```
[azl@fedora bin]$ lxterm /dev/ttyUSB0 
�yrRgI�������
             ���
                �
                 ������
                       ��
                         �.�Vi�^.�V�������
                                          �
                                           ��{
                                              ��

                                                *�;�;
                                                     K��Vi�^.�V������
                                                                     �

                                                                      �
                                                                       �
                                                                        �+
                                                                          ��;���Vi�^.�V�����

                                                                                            **ZT�T��hT��hPhP�P��Vi�^.�V����Uj:��k���{��t���T:�����:j�K��Vi�ޯ�{�j�˅�j��)z����ֶVm������Z
/�����������e�g=;����{maY�;-'�!#�}1�������������������

```

Using SignalTap (the ILA for Intel FPGAs) I get this on `serial_tx` when triggering on the first falling edge for `serial_tx~reg0`:
[image]
```
0010100001 -> 'P'
0101100001 -> '°'
0110110001 -> 'Ø'
011
```

The sampling clock is the 50 MHz system clock. Each bit is about ~433 periods and there is a start and end bit, which suggests the UART is set up correctly on the board side.

The USB serial device I'm using is [this](https://www.amazon.ca/gp/product/B07BBPX8B8/ref=ppx_yo_dt_b_asin_title_o09_s01?ie=UTF8&psc=1) which I have [used before without problems](https://github.com/alainlou/synth)

Don't know really how to investigate further, any ideas would help a lot!

tcal-x, 2021-08-22:
Getting the LEDchaser working is a great step!

Have you tried explicitly setting the lxterm TTY speed to 115200, as in `lxterm --speed 115200 /dev/ttyUSB0` ?

alainlou, 2021-08-23:
So I think I was doing something really stupid :rofl: 

I used to sketchily put jump wires into the DB9 female connector on the board which worked in previous projects, but gives me garbage now. I connected them to header pins on the board (the proper way) and it seems fine now.

Maybe it's because previously I had only been sending one character a time (with a fairly long pause where the line is held high between each character) but now the characters are transmitted back-to-back.

---

But now I have to look into why the BIOS stalls after the CRC check, I've narrowed it down to halting at this line of code:
`tmp[i++] = digits[num % base];`
in the `number` function in `lib.c` which was called by
`printf(" BIOS CRC passed (%08x)\n", actual_crc);` in `helpers.c`

I suspect something weird with `__udivsi3` since `putchar((char)((num%16) + 48));` (which does `andi    a0,s11,15`) works but `putchar((char)((num%base) + 48));` doesn't

tcal-x, 2021-08-23:
Calling the integer division emulation routine is expected since you've built the VexRiscv without mul/div.   I'm puzzled why it would hang though; usually issues in this area result in link-time errors.

I *do* often see differences between different versions of the RISC-V toolchain.   Do you know which version you're using?

alainlou, 2021-08-23:
LOL I did a `rm -rf` on the `build` directory (in litex-boards/litex_boards/targets) and rebuilding it works! Thanks @gatecat and leons! (couldn't find leons' github tag)

Seems it was because I had stale software binaries from when I tried with `lite` or `standard`

---

But now I'm back to looking UART issues :smile: 

[image]

I think I'll try different uart speeds and fifo depths, and see if something maybe is wrong with my connection to the board...

alainlou, 2021-08-23:
Update!

Did a `./litex_setup.py update` and we're good!

[image]

Seems some problems needed to get resolved upstream, perhaps [this stuff about stack alignment](https://github.com/enjoy-digital/litex/pull/988)

Thanks so much folks!

alainlou, 2021-08-30:
So now I'm having some trouble booting an application

[image]

I'm debugging with led_write's before the call to boot helper and on the first line of main in the application and I can see that the led_write before boot helper happens but not the one of the first line of main.

Things I know work: boot menu functionality, sdram_test, there seem to be no CRC errors when loading the code over UART
Things I've tried: `rm -rf build`, rm -rf all my litex stuff and running litex_setup.py all over again, `li x13, 0x40000000` before the `jr x13` in boot_helper

Any tips/other ways I could look at this?

alainlou, 2021-09-12:
Wow, so turns about 50 MHz was too fast for the external SDRAM chip. Even though it passes `sdram_test` and there are no errors when loading application code over serial, somehow there are problems when trying to actually boot. Bumping sys-clk-freq down to 25 MHz solves the issue! Behold the world's slowest spinning donut:
[image]

I found this out after I bought a logic analyzer, and connected it up to pins on the address bus of the DRAM, then `sdram_test` gave problems. 
[image]

Then I thought, oh maybe that's because of the additional capacitive load of the cables, why don't I slow down the clock ... boom, magic, it works! I think 50 MHz before was just on the edge (flaky enough for code to not boot, but not for the tests to fail), then adding the cables pushed it over.

Not quite sure if this a problem with my setup or not but I don't think I want to investigate further otherwise I'll lose my mind ...

If anyone is curious, this is what the board looks like (maybe EE majors will know if the traces on the board indicate bad design, since the [SDRAM datasheet](https://www.datasheets360.com/pdf/8926917598108841422) seems to suggest even >100 MHz is fine)
[image]

I think I could convert this into a PR?!

alainlou, 2021-09-14:
Woah, so it actually doesn't work if I remove the header cables from the address pins on the board:

Tried these clk rates: 2, 10, 12, 25 MHz

Adding header cables (they don't even need to be hooked up to the logic analyzer on the other side) solves the problem. Why would that be?

tcal-x, 2021-09-14:
Hi @alainlou , I'm really happy you're sticking with this and trying to figure it out.

My only guess with the cables is that there's noise that they're helping to flatten?   Or perhaps there's a floating node not tied to vdd or ground?    And as you mention, the capacitive load of the cables keep the voltage from going all over the place randomly.

What is your source for the board?   Would it be possible for me or you to acquire another one?

alainlou, 2021-09-14:
Hey @tcal-x, thanks for the tips!

Honestly, I would not recommend getting another one. I originally purchased it from Amazon but the listing is no longer online :laughing:

> My only guess with the cables is that there's noise that they're helping to flatten? Or perhaps there's a floating node not tied to vdd or ground? And as you mention, the capacitive load of the cables keep the voltage from going all over the place randomly.

Hmm, it is the address bus, so I don't think there should be any floating nodes ...

I do see from the logic analyzer when I hook it up that there are blips between subsequent addresses but I assume that's the DRAM controller trying to refresh other cells.
[image]

tcal-x, 2021-09-15:
@alainlou , one other thing occurred to me, something that you may have already checked -- is whether the correct IO standards are being specified for each group of pins.

But as far as the PR, since it works, even if slowly, go ahead and send it -- you're just adding new files, right?   Make sure there's a pointer to this issue.

alainlou, 2021-09-16:
Yea, I think the IO standards are correct - certainly for the memory IC (LVTTL)

I asked on [Reddit](https://www.reddit.com/r/AskElectronics/comments/poqppl/why_would_adding_header_cables_solve_any_problems/) and [Stack Exchange](https://electronics.stackexchange.com/questions/586820/why-would-adding-header-cables-solve-any-problems/586848#586848) - folks seem to suggest that adding the wires fixes some electrical issues (either hold time violations or problems with noise/ringing)

I think I might double check the memory timings for the module in litedram to be sure!

alainlou, 2021-09-16:
Huh, so I actually narrowed it down - I only actually need to add a wire on one pin, `sdram_a[5]`

I'm really not sure why that's special, perhaps a problem with just my board?

closed 2021-10-03 by alainlou