litex-hub/litex-boards #261: sqrl_acorn fails to build with vivado 2020.1 and crossover uart enabled

teknoman117, 2021-09-02:
Hello!

Just recently managed to get ahold of some Acorn CLE-215+ boards (nice upgrade over the LiteFury board I bought last year).

I was trying to build the base project with the crossover uart, but I'm getting some cryptic Vivado errors about no valid objects found for main_crg_clkout0. I pulled the LiteX tree clean as of last night

This command fails:
```./sqrl_acorn.py --with-pcie --build --driver --uart-name=crossover```

However, building it without the crossover uart succeeds, and I was able to flash it to the acorn and run the dma and scratch register tests without problems:
```./sqrl_acorn.py --with-pcie --build --driver```

I've attached the vivado logs for both.
[vivado-notworking.log](https://github.com/litex-hub/litex-boards/files/7102179/vivado-notworking.log)
[vivado-working.log](https://github.com/litex-hub/litex-boards/files/7102180/vivado-working.log)

sajattack, 2021-09-04:
I commented out these lines and it works for me https://github.com/litex-hub/litex-boards/blob/master/litex_boards/targets/sqrl_acorn.py#L108-L111

teknoman117, 2021-09-07:
That did fix it for me as well.

enjoy-digital, 2021-09-08:
Thanks for the feedback @teknoman117, I was aware of possible breakage and this is fixed with https://github.com/enjoy-digital/litex/commit/e0e9311cebcfa7334b867041850f5aca2dc50f05 and https://github.com/litex-hub/litex-boards/commit/129b95f9b514dd469674d22e2304f99e9515c805.

closed 2021-09-08 by enjoy-digital

WannaBeKratos, 2024-04-30:
Do other toolchain work for this board ? For example openXC7?