litex-hub/litex-boards #262: linsn_rv901t target fails to build

pauluzs, 2021-09-04:
_./linsn_rv901t.py --build_   
Fails to build some 19 out of 20 times.

Been trying create the gateware for this board on current linux install, a vm and a fresh ubuntu 20.04 (64) install. All with ISE 14.7. For this i'm using the default settings and guide on the litex github page

The build starts to fail at phase:

Phase 9.8  Global Placement
Phase 10.8  Global Placement
Phase 11.9  Local Placement Optimization
ERROR:Place:543 - This design does not fit into the number of slices available

Phase 11.9  Local Placement Optimization
ERROR:Place:120 - There were not enough sites to place all selected components.

Phase 11.9  Local Placement Optimization
ERROR:Pack:1654 - The timing-driven placement phase encountered an error.

Tried setting the environment variable XIL_PAR_ENABLE_LEGALIZER to 1 > No luck
Tried lowering some setting variables for the build > No luck
Tried different cpu and bus types > Still no luck

Tried the only other spartan6 target, albeit 25 instead the linsn's 16k
_./scarabhardware_minispartan6.py --build_ > Builds fine

[litexlog.txt](https://github.com/litex-hub/litex-boards/files/7109864/litexlog.txt)

enjoy-digital, 2021-09-06:
Hi @pauluzs,

it's possible that the default CPU/config is now too big for this board. Can you try with: `./linsn_rv901t.py --cpu-type=vexriscv --cpu-variant=lite --build`?

pauluzs, 2021-09-06:
Hi, 

It builds fine  up to : `--cpu-variant=lite+debug` all the time.
Strange behavior  was it seems to build the  `--cpu-variant=standard` once in _n_ times ( about 1:20)

Would it be preferable to set the cpu-variant to lite in linsn_rv901t.py as default so it would work out of the box with the steps as described in the readme ? ( just `./linsn_rv901t.py --build`)

Thanks

pauluzs, 2021-09-06:
Just found [https://github.com/q3k/chubby75/issues/86 ](https://github.com/q3k/chubby75/issues/86) 
Same problem, closing this one

closed 2021-09-06 by pauluzs