litex-hub/litex-boards #263: DRAM does not work on Arty with 70MHz system clock

danc86, 2021-09-08:
With 100MHz system clock (the default) or 80MHz, everything works fine. But when I reduced the system clock to 70MHz, Vivado can synthesise the design and the CPU appears to work, but the BIOS fails in memory testing.

Using these commands:

```
python3 -m litex_boards.targets.digilent_arty --build --sys-clk-freq 70000000
xc3sprog -c nexys4 build/digilent_arty/gateware/digilent_arty.bit
python3 -m litex.tools.litex_term /dev/ttyUSB1
```

BIOS output:

```
--========== Initialization ============--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Write latency calibration:
m0:0 m1:0
Read leveling:
  m0, b00: |00000000000000000000000000000000| delays: -
  m0, b01: |00000000000000000000000000000000| delays: -
  m0, b02: |00000000000000000000000000000000| delays: -
  m0, b03: |00000000000000000000000000000000| delays: -
  m0, b04: |00000000000000000000000000000000| delays: -
  m0, b05: |00000000000000000000000000000000| delays: -
  m0, b06: |00000000000000000000000000000000| delays: -
  m0, b07: |00000000000000000000000000000000| delays: -
  best: m0, b00 delays: -
  m1, b00: |00000000000000000000000000000000| delays: -
  m1, b01: |00000000000000000000000000000000| delays: -
  m1, b02: |00000000000000000000000000000000| delays: -
  m1, b03: |00000000000000000000000000000000| delays: -
  m1, b04: |00000000000000000000000000000000| delays: -
  m1, b05: |00000000000000000000000000000000| delays: -
  m1, b06: |00000000000000000000000000000000| delays: -
  m1, b07: |00000000000000000000000000000000| delays: -
  best: m1, b01 delays: -
Switching SDRAM to hardware control.
Memtest at 0x40000000 (2.0MiB)...
  Write: 0x40000000-0x40200000 2.0MiB
   Read: 0x40000000-0x40200000 2.0MiB
  bus errors:  256/256
  addr errors: 8191/8192
  data errors: 524288/524288
Memtest KO
Memory initialization failed
```

danc86, 2021-09-08:
I haven't studied how the DRAM is clocked on Arty yet. I guess 70MHz system clock is producing some incorrect or invalid clock configuration for the DRAM. But if 70MHz is not a valid configuration, I was hoping Litex would tell me that so I could change it.

enjoy-digital, 2021-09-09:
@danc86: Thanks for the feedback, this is fact probably similar to https://github.com/litex-hub/litex-boards/issues/166 and https://github.com/enjoy-digital/litex/issues/933. I'll try to look at it soon.

danc86, 2021-09-09:
Oh yes, it looks like the same as those other issues. Sorry, I didn't find those before I filed this.

enjoy-digital, 2021-09-14:
This has been fixed with https://github.com/enjoy-digital/litedram/commit/6f323f6a7aa032c8342b5931f4a562d2e28969e9.

closed 2021-09-14 by enjoy-digital