litex-hub/litex-boards #274: orangecrab doesn't build

mikey, 2021-10-01:
If I do:
```
cd litex_boards/targets
./gsd_orangecrab.py --build
```
It runs for a while but eventually dies in yosys with:
```
4.35.9. Executing OPT_CLEAN pass (remove unused cells and wires).
Finding unused cells or wires in module \gsd_orangecrab..
ERROR: Conflicting init values for signal 1'0 (\basesoc_cdcusb_cdcusbphy_inhandler_dtbs [4] = 1'0 != 1'x).
```
I assume this is an issue with valenty usb, but building valenty outside of litex isn't something I understand.

tcal-x, 2021-10-01:
I think this is an issue in Yosys that was resolved a few weeks ago.   Can you try a more recent Yosys?

https://github.com/YosysHQ/yosys/issues/2926

mikey, 2021-10-04:
Yep, that was the problem. Updating yosys fixed it. Thanks. @tcal-x

closed 2021-10-04 by mikey