litex-hub/litex-boards #276: symbiflow working? which boards do work correcty for Xilinx 7 series?

cacona666, 2021-10-04:
Hi there,

Thanks a lot for your work!

I tried to use symbiflow toolchain and I couldn't make it work.
I am using current master branch revision for litex-boards and linux-on-litex-vexriscv.

When using --board=arty like this:
`./make.py --board=arty --toolchain=symbiflow --build
`

I end up getting the following error:
```
symbiflow_synth -t arty -v /home/juan/Development/platforms/riscv/vexriscv/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/Ram_1w_1rs_Generic.v /home/juan/Development/platforms/riscv/vexriscv/pythondata-cpu-vexriscv-smp/pythondata_cpu_vexriscv_smp/verilog/VexRiscvLitexSmpCluster_Cc1_Iw32Is4096Iy1_Dw32Ds4096Dy1_ITs4DTs4_Ldw128_Ood.v /home/juan/Development/platforms/riscv/vexriscv/linux-on-litex-vexriscv/build/arty/gateware/arty.v -d artix7 -p xc7a35tcsg324-1 -x arty.xdc > /dev/null
**ERROR: Cell $iopadmap$arty.sdcard_cd of type \IBUF doesn't support the \SLEW attribute**

```

When using --board=arty_a7 I get a similar one:
`**ERROR: Cell $iopadmap$arty_a7.sdcard_cd of type \IBUF doesn't support the \SLEW attribute**
`
And when using the nexys4ddr it just tries to use Vivado toolchain:
```
OSError: Unable to find or source Vivado toolchain, please either:
- Source Vivado's settings manually.
- Or set LITEX_ENV_VIVADO environment variant to Vivado's settings path.
- Or add Vivado toolchain to your $PATH.

```

Symbiflow Xilinx 7 series examples include a 'Linux LiteX demo' example with "arty_100" target that builds correctly (haven't run them).

Question: Is this a known problem or I am doing something wrong?
Question: I plan to use the "nexy4ddr" target. Will it be possible/easy to make it work with current board configuration?
Question: Is it ok to use symbiflow or the recommendation is still to use Vivado for Litex?

Thanks!

mithro, 2021-10-04:
PTAL - @kgugala

mithro, 2021-10-04:
FYI - https://symbiflow-examples.readthedocs.io/en/latest/building-examples.html#linux-litex-demo

enjoy-digital, 2022-05-27:
The boards well supported by Symbiflow are probably listed in Symbiflow project. For designs with LiteX, the Digilent Arty is the one tested more regularly and recommended.

closed 2022-05-27 by enjoy-digital